AllskyTeam/allsky #758: Make error in capture.cpp Ubuntu 20.04

hoetzgit, 2021-11-03:
The following error occurs when compiling the actual release under Ubuntu 20.04. Can this be corrected? I use a powerful virtual machine to create startrail, keogram and video. Many thanks

pi@allsky:~/allsky01$ make -C src all
make: Entering directory '/home/pi/allsky01/src'
2021-11-03 09:49:13 Building capture program...
capture.cpp: In function ‘void* SaveImgThd(void*)’:
capture.cpp:298:18: error: ‘cvGetTickCount’ was not declared in this scope
  298 |             st = cvGetTickCount();
      |                  ^~~~~~~~~~~~~~
make: *** [Makefile:122: capture] Error 1
make: Leaving directory '/home/pi/allsky01/src'

linuxkidd, 2021-11-04:
Hi @hoetzgit ... Sorry for the issue... Would you be able to test the following tweak and see if the compile succeeds:
```
cd ~/allsky
sed -i -e 's/cvGetTickCount/cv::getTickCount/g' src/capture.cpp
make -C src all
```

Thanks!

hoetzgit, 2021-11-04:
Thanks very much! With this adjustment, it works.

pi@allsky:~/allsky01$ sed -i -e 's/cvGetTickCount/cv::getTickCount/g' src/capture.cpp
pi@allsky:~/allsky01$ make -C src all
make: Entering directory '/home/pi/allsky01/src'
2021-11-04 10:04:02 Building capture program...
2021-11-04 10:04:33 Done.
2021-11-04 10:04:33 Building capture_RPiHQ program...
2021-11-04 10:04:40 Done.
2021-11-04 10:04:40 Building startrails program...
2021-11-04 10:04:54 Done.
2021-11-04 10:04:54 Building keogram program...
2021-11-04 10:05:07 Done.
make: Leaving directory '/home/pi/allsky01/src'

closed 2021-11-04 by linuxkidd

linuxkidd, 2021-11-05:
Great to hear this resolved the compile issue.  We've modified our code to include this fix.  Thanks for helping us make Allsky better!

EricClaeys, 2021-11-05:
@hoetzgit, you can grab the updated src/capture.cpp which includes the fix.  Or not, since you already patched your file.

PR #762 (2021-11-04): capture.cpp: getTickCount
Fixes #758
commit e63a2314 (2021-11-04): capture.cpp: getTickCount

Fixes #758