AllskyTeam/allsky #760: Night capture failed in middle of night

dbainbridge, 2021-11-04:
Last night right at the peak of aurora activity the image captures stopped :(.  Am using version from about week ago (can get commit later).  Any ideas what may have happened?  Did not run out of disk space.

EricClaeys, 2021-11-04:
@dbainbridge please attach your log file `/var/log/allsky.log` and let me know the date and time capture stopped so I know where in the file to look.
Have you checked the ~/allsky/images/DATE directory to see if the files are there?  That would help determine if capture stopped versus the web page stopped.

dbainbridge, 2021-11-04:
yes, checked the image directory and they were not there.  Will check for logs later.

dbainbridge, 2021-11-04:
Last logs (the integer expression did not cause the issue since they were in all the previous log statements as well):

Nov  4 02:07:21 allsky allsky.sh[5652]: STARTING EXPOSURE at: 2021-11-04 02:07:21
Nov  4 02:07:23 allsky allsky.sh[5652]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 02:07:53 allsky allsky.sh[5652]:   > Saving image 'image.jpg' that started at 2021-11-04 02:07:21  (0 us)
Nov  4 02:07:53 allsky allsky.sh[5652]:   > Sleeping: 1960 ms
Nov  4 02:07:55 allsky allsky.sh[5652]: STARTING EXPOSURE at: 2021-11-04 02:07:55
Nov  4 02:07:58 allsky allsky.sh[5652]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 02:08:25 allsky allsky.sh[5652]:   > Saving image 'image.jpg' that started at 2021-11-04 02:07:55  (0 us)
Nov  4 02:08:25 allsky allsky.sh[5652]:   > Sleeping: 3292 ms
Nov  4 02:08:28 allsky allsky.sh[5652]: STARTING EXPOSURE at: 2021-11-04 02:08:28
Nov  4 02:08:29 allsky allsky.sh[5652]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 02:08:32 allsky allsky.sh[5652]: copy_notification_image.sh: Uploading NotRunning.jpg
Nov  4 02:08:32 allsky allsky.sh[5652]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 02:08:35 allsky allsky.sh[5652]:      ***** Stopping AllSky *****
Nov  4 08:48:36 allsky allsky.sh[5840]:      ***** Starting AllSky *****

Last one is when I realized the service was no longer running and I manually restarted it.

dbainbridge, 2021-11-05:
Seems to be continuing tonight but got some actual errors in logs:

Nov  4 19:37:08 allsky allsky.sh[19805]: STARTING EXPOSURE at: 2021-11-04 19:37:08
Nov  4 19:37:12 allsky allsky.sh[19805]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 19:38:13 allsky allsky.sh[19805]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT #1 (with 0.8 exposure = YES))
Nov  4 19:38:15 allsky allsky.sh[19805]: STARTING EXPOSURE at: 2021-11-04 19:38:15
Nov  4 19:39:20 allsky allsky.sh[19805]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT #2 (with 0.8 exposure = YES))
Nov  4 19:39:22 allsky allsky.sh[19805]: STARTING EXPOSURE at: 2021-11-04 19:39:22
Nov  4 19:40:27 allsky allsky.sh[19805]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT #3 (with 0.8 exposure = YES))
Nov  4 19:40:29 allsky allsky.sh[19805]: STARTING EXPOSURE at: 2021-11-04 19:40:29
Nov  4 19:41:34 allsky allsky.sh[19805]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT #4 (with 0.8 exposure = YES))
Nov  4 19:41:36 allsky allsky.sh[19805]: STARTING EXPOSURE at: 2021-11-04 19:41:36
Nov  4 19:42:41 allsky allsky.sh[19805]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT #5 (with 0.8 exposure = YES))
Nov  4 19:42:43 allsky allsky.sh[19805]: Maximum number of consecutive errors of 5 reached; exiting...
Nov  4 19:42:44 allsky allsky.sh[19805]: copy_notification_image.sh: Uploading NotRunning.jpg
Nov  4 19:42:44 allsky allsky.sh[19805]: /home/pi/allsky/scripts/upload.sh: line 90: [: l: integer expression expected
Nov  4 19:42:47 allsky allsky.sh[19805]:      ***** Stopping AllSky *****
Nov  4 19:42:47 allsky allsky.sh[19805]: 'capture' exited with RETCODE=2
Nov  4 19:42:52 allsky allsky.sh[20066]:      ***** Starting AllSky *****

dbainbridge, 2021-11-05:
Appears like there is some USB issue going on?

dbainbridge, 2021-11-05:
I think I figured it out:
`vcgencmd get_throttled`
is returning` throttled=0x50005`

People should be able to install that command with `pip install vcgencmd`
My value is indicating under voltage and throttling has occurred and is currently happening.

Might by helpful to install that python package as well so would be easier to rule out if users are having this issue.

dbainbridge, 2021-11-05:
More concerning is why am I getting this under voltage to begin with.  Have a DC to DC converter which is stepping down a 56V power supply to the 5V for the Pi.  The converter is no more than 2ft away from the Pi and have been using this setup for a couple years.. Will have to do further checks tomorrow to see.

EricClaeys, 2021-11-05:
@dbainbridge, what are the last few lines of your allsky/variables.sh file?
If the last file is not this, please upgrade the software:
`ALLSKY_DEBUG_LEVEL=${ALLSKY_DEBUG_LEVEL:-0}`
I believe the "integer expression expected" errors are because the ALLSKY_DEBUG_LEVEL variable isn't defined so upload.sh is comparing a null string to a number.

A fair number of people get `ASI_ERROR_TIMEOUT ` errors.  For some it's so bad that they need to turn off the "Version 0.8 Exposure" mode.  Why they occur is unknown, but some people resolved it by installing a powered USB hub.  I think it's likely a camera or Pi hardware problem, not an AllSky software problem, since it doesn't happen to everyone.

vcgencmd is installed by default, and allsky.sh uses it to detect RPiHQ cameras.

What is throttled?  The USB bus or something else?

dbainbridge, 2021-11-05:
I believe the CPU frequency is being throttled in order to reduce amp draw to help increase the voltage back to a safe level.  I also wonder if this is more prevalent with the 0.8 exposure mode because if I recall that version is now turning the camera on and off in order to "reduce camera heating".  Maybe this is causing unintended side affects from the the surge of camera going on and off constantly.

My variables.sh file doesn't have reference to ALLSKY_DEBUG_LEVEL.. I'll update since there is no aurora tonight at least :)

EricClaeys, 2021-11-05:
@dbainbridge, clearly the version 0.8 exposure mode is having side effects, like causing ASI_ERROR_TIMEOUTs for some people.  Why it's doing that is anyone's guess.  I've been communicating with someone at ZWO who says they are working on the ASI_ERROR_TIMEOUTs but it's unclear if they'll every do anything.

Funny, when we were working on 0.8, all 4 of us were able to resolve the ASI_ERROR_TIMEOUTs by changing our USB settings.  The only "side effect" we say was reduced temperature in the camera.

My ASI290 very rarely gives an ASI_ERROR_TIMEOUT, and when it does, it's just one every now and then.  I've never had the program restart like yours did a few posts ago after 5 consecutive errors.

Would you be willing to try the attached file?  It _may_ reduce the timeouts.
If so, do the following:
~~~
cd ~/allsky/src
mv capture.cpp capture-ORIGINAL.cpp   # save current file
# copy the attached file and call it "capture.cpp"
make capture
sudo service allsky stop
cp capture ..
sudo truncate -s 0 /var/log/allsky.log   # clears out log file
sudo service allsky start
~~~

Let it go a few minutes and then attach /var/log/allsky.log

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7480239/capture.cpp.txt)

linuxkidd, 2021-11-05:
The stop and start commands should be:
```
sudo systemctl stop allsky
```

And
```
sudo systemctl start allsky
```

dbainbridge, 2021-11-05:
I may have addressed my power issue by attaching a https://lifepo4wered.com/lifepo4wered-pi.html and have 12V attached to it.  It no longer is showing active under voltage or throttling.  Also provides other benefits to make the entire system more reliable.  I'll update if I run into anything related to this but the attached file and instructions you provided would probably be helpful  for others who are still having similar issues.

EricClaeys, 2021-11-05:
@linuxkidd, what's the difference in `sudo systemctl start allsky` and `sudo service allsky start`?  I use the latter, and it works fine.

linuxkidd, 2021-11-06:
The `service` command is the old way... and is being deprecated.  Currently, it's mapped through and ultimately uses the `systemctl` commands.  So, at some point in the future, the `service` command won't work at all.  ( honestly, I thought it was the case already.. which is why I commented above ).

EricClaeys, 2021-11-07:
@dbainbridge, David, if your problem is resolved would you please close the Issue?  Thanks

dbainbridge, 2021-11-07:
Yes.. Was just making sure after few nights the hardware change fixed the issue which it so far appears it has.  Would be good if there was way to detect the low voltage issue which I'm betting people are having but blaming the software.

EricClaeys, 2021-11-07:
@dbainbridge, does running the `vcgencmd get_throttled` command tell you that?

dbainbridge, 2021-11-08:
Yes, sorry.. Was thinking would be useful to dump to logs when there is camera error especially that ASI_ERROR_TIMEOUT one which is what I was getting and no longer am getting once I fixed my power issue.

EricClaeys, 2022-06-03:
Closing Issue - no response from use in almost 4 months, so assume problem is resolved.

closed 2022-06-03 by EricClaeys