AllskyTeam/allsky #766: stops taking pictures

rchadgray, 2021-11-05:
Hello!  I have been getting what i think it a lock up or failure of the software.  I have not found a reason.  Some nights it works fine and takes all the images.  Some nights it locks up at 10pm. I restart the raspberry pi if i realize it locked up.  Then 3am.

Let me know what else you need to debug.

Rasperrypi 4 with 8gig ram
32gig sd fast SD card
HQ Camera

30 second exposures with 5 seconds in-between them.

The log file just ends in the last Capture & Saving image...

The timelapse is not generated.  In the morning i can go and run it manually.

`pi@allsky:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G   18G   11G  63% /
devtmpfs        3.8G     0  3.8G   0% /dev
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           3.9G   58M  3.8G   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           3.9G     0  3.9G   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   48M  205M  19% /boot
tmpfs           783M  8.0K  783M   1% /run/user/1000`

**Log file:**

`Nov  4 22:42:02 allsky allsky.sh[1172]: CAMERA: RPiHQ
Nov  4 22:42:02 allsky allsky.sh[1172]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Nov  4 22:42:03 allsky allsky.sh[1172]: Starting allsky camera...
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[32m ******************************************
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[32m *** Allsky Camera Software v0.8 | 2021 ***
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[32m ******************************************
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[32mCapture images of the sky with a Raspberry Pi and an ZWO ASI or RPi HQ camera
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[33mAdd -h or -help for available options
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[0mContributors:
Nov  4 22:48:38 allsky allsky.sh[1172]: -Knut Olav Klo
Nov  4 22:48:38 allsky allsky.sh[1172]: -Daniel Johnsen
Nov  4 22:48:38 allsky allsky.sh[1172]: -Yang and Sam from ZWO
Nov  4 22:48:38 allsky allsky.sh[1172]: -Robert Wagner
Nov  4 22:48:38 allsky allsky.sh[1172]: -Michael J. Kidd - <linuxkidd@gmail.com>
Nov  4 22:48:38 allsky allsky.sh[1172]: -Rob Musquetier
Nov  4 22:48:38 allsky allsky.sh[1172]: -Eric Claeys
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[0m- Resolution: 4096x3040
Nov  4 22:48:38 allsky allsky.sh[1172]: - Pixel Size: 1.55microns
Nov  4 22:48:38 allsky allsky.sh[1172]: - Supported Bin: 1x, 2x and 3x
Nov  4 22:48:38 allsky allsky.sh[1172]: #033[32m
Nov  4 22:48:38 allsky allsky.sh[1172]: Capture Settings:
Nov  4 22:48:38 allsky allsky.sh[1172]:  Resolution (before any binning): 2068x1550
Nov  4 22:48:38 allsky allsky.sh[1172]:  Quality: 95
Nov  4 22:48:38 allsky allsky.sh[1172]:  Exposure (night): 30000ms
Nov  4 22:48:38 allsky allsky.sh[1172]:  Auto Exposure (night): 0 (no)
Nov  4 22:48:38 allsky allsky.sh[1172]:  Auto Focus: 0 (no)
Nov  4 22:48:38 allsky allsky.sh[1172]:  Gain (night): 4.00
Nov  4 22:48:38 allsky allsky.sh[1172]:  Auto Gain (night): 0 (no)
Nov  4 22:48:38 allsky allsky.sh[1172]:  Brightness (day): 50
Nov  4 22:48:38 allsky allsky.sh[1172]:  Brightness (night): 50
Nov  4 22:48:38 allsky allsky.sh[1172]:  Gamma: 50
Nov  4 22:48:38 allsky allsky.sh[1172]:  Auto White Balance: 1 (yes)
Nov  4 22:48:38 allsky allsky.sh[1172]:  WB Red: 2.80
Nov  4 22:48:38 allsky allsky.sh[1172]:  WB Blue: 2.00
Nov  4 22:48:38 allsky allsky.sh[1172]:  Binning (day): 1
Nov  4 22:48:38 allsky allsky.sh[1172]:  Binning (night): 1
Nov  4 22:48:38 allsky allsky.sh[1172]:  Delay (day): 30000ms
Nov  4 22:48:38 allsky allsky.sh[1172]:  Delay (night): 5000ms
Nov  4 22:48:38 allsky allsky.sh[1172]:  Text Overlay:
Nov  4 22:48:38 allsky allsky.sh[1172]:  Font Color: 255
Nov  4 22:48:38 allsky allsky.sh[1172]:  Font Background Color: 0
Nov  4 22:48:38 allsky allsky.sh[1172]:  Font Size: 64.0
[... 199 lines cut ...]
	echo "${RED}ERROR: Please set CAMERA in config.sh${NC}"
	exit 1
fi

# !!! The next few lines are for backwards compatibility and will be removed in the future - do not change !!!
UPLOAD_IMG="${IMG_UPLOAD}"	# UPLOAD_IMG is th eold name for IMG_UPLOAD
if [ "${DAYTIME_CAPTURE}" = "true" ]; then
	DAYTIME=1		# DAYTIME is the old name for DAYTIME_CAPTURE
else
	DAYTIME=0
fi
CAPTURE_24HR=${DAYTIME_SAVE}	# CAPTURE_24HR is the old name for DAYTIME_SAVE

if [[ ${CAMERA} == "auto" ]]; then
	# Restore currently saved autodiscovered camera type, if any.
	source "${ALLSKY_CONFIG}/autocam.sh"	# May reset ${CAMERA}
fi
if [[ ${CAMERA} != "auto" ]]; then
	CAMERA_SETTINGS="${CAMERA_SETTINGS_DIR}/settings_${CAMERA}.json"
 
	# So scripts can conditionally ouput messages; DO NOT CHANGE NEXT LINES.
	ALLSKY_DEBUG_LEVEL=$(jq -r '.debuglevel' "${CAMERA_SETTINGS}")
fi
ALLSKY_DEBUG_LEVEL=${ALLSKY_DEBUG_LEVEL:-0}
`

linuxkidd, 2021-11-05:
Hi @rchadgray ... Can you provide the following output:

```
ps awwuxf | grep -v grep | grep -A10 allsky
```

Thanks!

fwiw, I have a lock up of RPiHQ captures as well...  It seems the `raspistill` call is hung for some reason.
```
  679 ?        SLl    2:52  \_ /home/pi/allsky/capture_RPiHQ -nightexposure 60000 -nightgain 4 -nightautogain 0 -gamma 50 -brightness 50 -awb 0 -wbr 2.8 -wbb 2 -nightbin 1 -nightdelay 10 -daydelay 500 -quality 95 -width 0 -height 0 -filename image.jpg -rotation 0 -flip 0 -text  -showTime 1 -showDetails 1 -fontsize 32 -fontcolor 255 -background 0 -notificationimages 0 -latitude 31.930402N -longitude 109.039083W -angle 0 -autofocus 0 -debuglevel 0 -darkframe 0 -alwaysshowadvanced 1 -tty 0 -daytime 1
 7665 ?        S      0:00      \_ sh -c nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 4 --awb off --awbgains 2.8,2 --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000 
 7666 ?        SNl    0:00          \_ raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 4 --awb off --awbgains 2.8,2 --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000
```

NOTE: This is locked from Oct 30th.  Last `allsky.log` lines:
```
Oct 30 06:18:55 allsky-dev allsky.py[620]: Capturing & saving image done, now wait 0 seconds...
Oct 30 06:18:55 allsky-dev allsky.py[620]: Capturing & saving image...
```

@EricClaeys -- Maybe we should wrap the `raspistill` call with a timeout.  Say double exposure time, or exposure time + 30 seconds, whichever is longer?  That way, it may recover from this type of event without missing too many exposures?

linuxkidd, 2021-11-05:
Oh, another theory -- maybe interference in the camera CSI cable?  ( mine is only being used for testing, so it's laying across a power strip -- 60hz noise is definitely being injected. )

rchadgray, 2021-11-05:
I attached the output of the ps command.

As far as cabling and interference my setup is in a 3D printed weather station i am building.  You can see the routing of my camera cable. 
https://photos.app.goo.gl/PP7afRsiqYc3Ch549

I should say i do have POE Hat on my Raspberry pi.  That may have some interference on the camera cable??  Just a guess.

[new 8.txt](https://github.com/thomasjacquin/allsky/files/7487228/new.8.txt)
.

rchadgray, 2021-11-05:
Oh wow i am swapping my hardware over to a new enclosure and looked at my camera cable.  I am going to replace this with a new one and test again tonight.

https://photos.app.goo.gl/3xGc2W2RYE9tRsxc7

linuxkidd, 2021-11-06:
Interesting!  I'd love to know if the condition clears after replacing the cable.  I too am using a POE hat, so I'm hope'n that the cable is the fix... not the POE hat. :)

rchadgray, 2021-11-06:
Yeah i think the cable was affected by some rain on the camera.  I never checked it for crustys and damage after i left the camera and raspberry pi out in the rain.

It is taking pictures now.  I will report back if it is successful over the weekend.

rchadgray, 2021-11-06:
Locked up again at 1am.

I ran the ps command while in it locked up state.  The OS is responsive, the allsky website is responsive.  It seems it is just the picturing taking part that has stopped.

I will leave it in it's current state if there are other debugging things i can try.

[new 52.txt](https://github.com/thomasjacquin/allsky/files/7490431/new.52.txt)

EricClaeys, 2021-11-07:
@rchadgray Chad, please check the Wiki.  This behavior is usually resolved by upgrading the camera firmware.

rchadgray, 2021-11-07:
Thanks Eric..  i will check it out.  Looks like it is updating the packages in APT and installing them.

I am doing that now and will report back.

https://github.com/thomasjacquin/allsky/wiki/Troubleshooting:-other

EricClaeys, 2021-11-07:
@rchadgray, yea, I think I should have said "by upgrading the Pi firmware".  Either way, that's fixed the problem for everyone else.

rchadgray, 2021-11-07:
Thanks Eric!  My camera ran the entire night.   Before updating it would stop working about 3 out of 7 nights.  We will see if this improves.

I will update this thread this week if i run into it again.

Chad

EricClaeys, 2021-11-07:
@rchadgray, Chad, that's exactly what other people say.  Glad the firmware update appears to have helped.  Let's hope it continues.  If it does, would you please close the Issue?  Thanks.

rchadgray, 2021-11-08:
Another successful night.  I will close the issue in a few days if this continues.

Thanks again!

rchadgray, 2021-11-10:
No hicups so far!  I will close this out.

closed 2021-11-10 by rchadgray