AllskyTeam/allsky #774: New error in allsky

maserowik, 2021-11-09:
getting this in my allsky log.  any ideas?

Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]: === Starting nighttime capture ===
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]: ==========
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]:   xxxx Adjusting NIGHT gain by 0 on next picture to 50; will be gain change # 1 of 90.
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]: Buffer size: 954528
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]: STARTING EXPOSURE at: 2021-11-08 17:39:57
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]:   > Exposure set to 15000000 us (15000.00 ms), timeout: 35000 ms
Nov  8 17:39:57 Pi-AllSky allsky.sh[4864]:   > Cleared 1 frame
Nov  8 17:40:12 Pi-AllSky allsky.sh[4864]: /home/pi/allsky/allsky.sh: line 189:  4978 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  8 17:40:12 Pi-AllSky allsky.sh[4864]: 'capture' exited with RETCODE=139
Nov  8 17:40:12 Pi-AllSky allsky.sh[4864]: copy_notification_image.sh: Uploading Error.jpg
Nov  8 17:40:14 Pi-AllSky allsky.sh[4864]: *** After fixing, restart the allsky service. ***
pi@Pi-AllSky:/var/log $

EricClaeys, 2021-11-09:
@maserowik, what version of capture.cpp do you have?  It'll be in the log file.
Does it do this every time?
Try the attached version and see if it does it.

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7506658/capture.cpp.txt)

maserowik, 2021-11-09:
@EricClaeys 
So I did a sudo service allsky restart and the day light appears to be working.  Looking at the message it appear to have started when the allsky switches from daytime to nighttime.

I also reenable remote access to my pi.  same url and password as before if you like to jump on

Mike

maserowik, 2021-11-09:
also getting the lines in the text

[image]

EricClaeys, 2021-11-09:
@maserowik, did you get the Segmentation fault just once?
What did you do last time to get rid of the lines?  Or did they magically disappear on their own?
Do they go away if you don't display the date, or don't display the sensor temperature, or don't display the extratext ?

maserowik, 2021-11-09:
It starts as soon as the night capture starts
During the day it works fine

Michael A. Serowik

EricClaeys, 2021-11-09:
@maserowik, can you try the attached file?  It's been working for me.  Set Debug Level to 4.

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/7508722/capture.cpp.txt)

EricClaeys, 2021-11-10:
@maserowik, for the lines, in [port issue 75](https://github.com/thomasjacquin/allsky-portal/issues/75) you said you restarted the service and the lines went away and "not quite sure what that has todo with anything but I do have the Use Outline Font turned on not sure what I should be seeing."
If you restart the service (you can simply click on the "Save changes" button in the WebUI without making any changes), do the lines go away again?

Are the lines on EVERY picture, or just some of them?

maserowik, 2021-11-10:
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: ==========
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: === Starting nighttime capture ===
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: ==========
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: STARTING EXPOSURE at: 2021-11-09 19:35:21
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]:   > Exposure set to 10000000 us (10000.00 ms)
Nov  9 19:35:32 Pi-AllSky allsky.sh[14485]: /home/pi/allsky/allsky.sh: line 189: 14599 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  9 19:35:32 Pi-AllSky allsky.sh[14485]: 'capture' exited with RETCODE=139
Nov  9 19:35:33 Pi-AllSky allsky.sh[14485]: copy_notification_image.sh: Uploading Error.jpg
Nov  9 19:35:34 Pi-AllSky allsky.sh[14485]: *** After fixing, restart the allsky service. ***

still the same with you new capture.cpp.

EricClaeys, 2021-11-10:
@maserowik Would you please set Debug Level to 4?
If you start the service after nighttime (rather than letting it transition), do you still get the Segmentation fault?  I don't get it either way.

maserowik, 2021-11-10:
@EricClaeys 

[image]

Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: ==========
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: === Starting nighttime capture ===
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: ==========
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]: STARTING EXPOSURE at: 2021-11-09 19:35:21
Nov  9 19:35:21 Pi-AllSky allsky.sh[14485]:   > Exposure set to 10000000 us (10000.00 ms)
Nov  9 19:35:32 Pi-AllSky allsky.sh[14485]: /home/pi/allsky/allsky.sh: line 189: 14599 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  9 19:35:32 Pi-AllSky allsky.sh[14485]: 'capture' exited with RETCODE=139
Nov  9 19:35:33 Pi-AllSky allsky.sh[14485]: copy_notification_image.sh: Uploading Error.jpg
Nov  9 19:35:34 Pi-AllSky allsky.sh[14485]: *** After fixing, restart the allsky service. ***
Nov  9 20:02:07 Pi-AllSky allsky.sh[14731]:      ***** Starting AllSky *****
Nov  9 20:02:07 Pi-AllSky allsky.sh[14731]: CAMERA: ZWO
Nov  9 20:02:07 Pi-AllSky allsky.sh[14731]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Nov  9 20:02:08 Pi-AllSky allsky.sh[14731]: copy_notification_image.sh: Uploading StartingUp.jpg
Nov  9 20:02:09 Pi-AllSky allsky.sh[14731]: Starting allsky camera...
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[32m **********************************************
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[32m *** Allsky Camera Software v0.8.1b |  2021 ***
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[32m **********************************************
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[32mCapture images of the sky with a Raspberry Pi and an ASI Camera
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[33mAdd -h or --help for available options
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[0mContributors:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Knut Olav Klo
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Daniel Johnsen
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Yang and Sam from ZWO
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Robert Wagner
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Michael J. Kidd - <linuxkidd@gmail.com>
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Chris Kuethe
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: -Eric Claeys
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: #033[0m
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: ZWO ASI224MC Information:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Resolution:1304x976
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Pixel Size: 3.8microns
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Supported Bin: 1 2
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Color Camera: bayer pattern:RG
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Camera ID: [none]
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   - Camera Serial Number: 0x2f19060622090900
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: Control Caps:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: - Gain:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MinValue = 0
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MaxValue = 600
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - DefaultValue = 200
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsAutoSupported = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsWritable = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - ControlType = 0
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: - Exposure:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MinValue = 32
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MaxValue = 2000000000
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - DefaultValue = 10000
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsAutoSupported = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsWritable = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - ControlType = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: - WB_R:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MinValue = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MaxValue = 99
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - DefaultValue = 52
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsAutoSupported = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsWritable = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - ControlType = 3
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: - WB_B:
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MinValue = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - MaxValue = 99
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - DefaultValue = 95
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsAutoSupported = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - IsWritable = 1
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:    - ControlType = 4
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]: - Offset:
[... 130 lines cut ...]
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Temperature Type: B
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Show Exposure: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Show Gain: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Show Brightness: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Preview: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Taking Dark Frames: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Debug Level: 4
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  TTY: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  Non-continuous Capture Method: No
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:  ZWO SDK version 1, 20
[... cut ...]
Nov  9 20:02:14 Pi-AllSky allsky.sh[14731]:   > Exposure set to 10000000 us (10000.00 ms), timeout: 25000 ms
Nov  9 20:02:24 Pi-AllSky allsky.sh[14731]: /home/pi/allsky/allsky.sh: line 189: 14845 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Nov  9 20:02:24 Pi-AllSky allsky.sh[14731]: 'capture' exited with RETCODE=139
Nov  9 20:02:25 Pi-AllSky allsky.sh[14731]: copy_notification_image.sh: Uploading Error.jpg
Nov  9 20:02:27 Pi-AllSky allsky.sh[14731]: *** After fixing, restart the allsky service. ***

maserowik, 2021-11-10:
> @maserowik, for the lines, in [port issue 75](https://github.com/thomasjacquin/allsky-portal/issues/75) you said you restarted the service and the lines went away and "not quite sure what that has [...]

They are in every photo.  but as soon as I drop to a ssh session and issue the stop and start the lines go away.

EricClaeys, 2021-11-10:
@maserowik, can you try the version of capture.cpp I posted a few hours ago?
~~~
cd ~/allsky/src
mv capture.cpp capture-ORIGINAL.cpp   # save current file
# copy the attached file and call it "capture.cpp"
make capture
sudo service allsky stop
cp capture ..
sudo truncate -s 0 /var/log/allsky.log   # clears out log file
sudo service allsky start
~~~

EricClaeys, 2021-11-10:
@maserowik, someone else just reported the "Segmentation fault" problem, but he has the old version 0.7 capture.cpp file that knows about version 0.8 arguments.  That's likely a latent bug that's appearing now for some reason.

maserowik, 2021-11-10:
@EricClaeys 

so the software stopped again on the night capture.  I just did a restart on the service and it is no in daylight capture mode and working fine.  
 but I now have the lined on the txt messages.
[image]

i did a restart on the services and the lines went away
[image]

mike

maserowik, 2021-11-10:
also is there a way to have allsky be ticked into it being night time?

EricClaeys, 2021-11-10:
@maserowik, sure, you just have to set the Angle to (I believe) a high value like 89.9.

EricClaeys, 2021-11-10:
Were you able to try the version of capture.cpp from last night?

maserowik, 2021-11-10:
i set my angle to 89.9
[image]

as soon as the night time capture started i got the following
[image]

yes I am using the capture.cpp from lastnight.

EricClaeys, 2021-11-10:
@maserowik, can you set Debug Level to 4 and try nighttime again, please?

maserowik, 2021-11-10:
Level 4
[image]

error in log
[image]

maserowik, 2021-11-10:
[image]
the lines came back.

maserowik, 2021-11-12:
@EricClaeys 
the fix you provided solved my issue with the software stopping.
mike

maserowik, 2021-11-12:
@EricClaeys 
I know the issue is currently fix on the Pi I am using.  When I do a new git pull will the fix be there or have they yet been merged into the code?

Sorry to reopen the ticket.
Mike

EricClaeys, 2021-11-12:
@maserowik, the fix is NOT yet in Git.  It only impacts the src/capture.cpp file so you're safe to grab any other updates.
Keep bugging me until I put the fix in and have it merged.

maserowik, 2021-11-17:
@EricClaeys 
Just checking to see if you were able to get the "FIX" added to the git yet?.  I like to do a new git pull but do not want to break what you fixed.  not sure how to just grab the updates?  Is there something in the wiki i can follow?

Mike

EricClaeys, 2021-11-18:
@maserowik, not yet, although I did create a [Pull Request](https://github.com/thomasjacquin/allsky/pull/800) with the change, but it's waiting to be approved.

maserowik, 2021-11-18:
@EricClaeys 
Thanks  I will close the issue once the Pull Request gets approved. 

Mike

maserowik, 2021-11-24:
@EricClaeys 
I see you fix still has not been merged into the files  
[https://github.com/thomasjacquin/allsky/pull/800](url)

Not to rush but any idea when it will be merged?

Miek

maserowik, 2021-12-03:
@EricClaeys 
I did a fresh pull today and will check on the allsky tonight.

Mike

EricClaeys, 2021-12-03:
Sorry it took a while.  Other team members have been busy.

Eric

maserowik, 2021-12-06:
the update solved my issue closing the ticket

closed 2021-12-06 by maserowik