AllskyTeam/allsky #778: Error: dayOrNight isn't DAY or NIGHT, it's ''

Stef682, 2021-11-11:
Hi,

I installed AllSky a long time ago and everything worked perfectly.

Since I had an old version I decided to delete the program and install the latest one.

Unfortunately now it no longer works. On the live view it is indicated that there is an error and that I must see the log file.

Here is the log file:

Nov 11 00:46:45 AllSky allsky.sh[1682]:      ***** Starting AllSky *****
Nov 11 00:46:45 AllSky allsky.sh[1682]: CAMERA: ZWO
Nov 11 00:46:45 AllSky allsky.sh[1682]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Nov 11 00:46:45 AllSky allsky.sh[1682]: Starting allsky camera...
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[32m **********************************************
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[32m *** Allsky Camera Software v0.8.1b |  2021 ***
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[32m **********************************************
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[32mCapture images of the sky with a Raspberry Pi and an ASI Camera
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[33mAdd -h or --help for available options
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[0mContributors:
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Knut Olav Klo
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Daniel Johnsen
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Yang and Sam from ZWO
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Robert Wagner
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Michael J. Kidd - <linuxkidd@gmail.com>
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Chris Kuethe
Nov 11 00:46:49 AllSky allsky.sh[1682]: -Eric Claeys
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[0m
Nov 11 00:46:49 AllSky allsky.sh[1682]: ZWO ASI224MC Information:
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Resolution:1304x976
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Pixel Size: 3.8microns
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Supported Bin: 1 2
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Color Camera: bayer pattern:RG
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Camera ID: [none]
Nov 11 00:46:49 AllSky allsky.sh[1682]:   - Camera Serial Number: 0x0a18420611080900
Nov 11 00:46:49 AllSky allsky.sh[1682]: - Sensor temperature:33.70
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[32m
Nov 11 00:46:49 AllSky allsky.sh[1682]: Capture Settings:
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Image Type: ASI_IMG_RGB24
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Resolution (before any binning): 1304x976
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Quality: 95
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Daytime capture: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Exposure (day): 0.500ms
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Exposure (night): 10000ms
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Auto Exposure (day): Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Auto Exposure (night): Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Max Exposure (day): 15000ms (15.0s)
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Max Exposure (night): 20000ms (20.0s)
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Delay (day): 5000ms
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Delay (night): 10ms
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Gain (night only): 50
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Auto Gain (night only): No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Max Gain (night only): 200
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Gain Transition Time: 15.0 minutes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Brightness (day): 50
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Brightness (night): 50
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Cooler Enabled: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Target Temperature: 0C
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Gamma: 50
Nov 11 00:46:49 AllSky allsky.sh[1682]:  WB Red: 53
Nov 11 00:46:49 AllSky allsky.sh[1682]:  WB Blue: 90
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Auto WB: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Binning (day): 1
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Binning (night): 1
Nov 11 00:46:49 AllSky allsky.sh[1682]:  USB Speed: 80
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Auto USB Speed: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Text Overlay: [none]
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Text Extra Filename: [none]
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Text Extra Filename Age: 600
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Text Line Height 60px
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Text Position: 15px left, 30px top
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Font Name:  0 (SIMPLEX)
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Font Color: 255 , 255, 255
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Small Font Color: 0 , 0, 255
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Font Line Type: 16
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Font Size: 7.0
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Font Line: 1
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Outline Font : No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Flip Image: 0
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Filename: image.jpg
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Latitude: 47.34N
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Longitude: 7.30E
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Sun Elevation: -6
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Locale: en_US.UTF-8
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Notification Images: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Histogram Box: 500 500 50 50
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Histogram Box: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Histogram Mean: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Time: Yes (format: %Y%m%d %H:%M:%S)
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Details: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Temperature: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Temperature Type: C
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Exposure: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Gain: Yes
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Show Brightness: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Preview: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Taking Dark Frames: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Debug Level: 1
Nov 11 00:46:49 AllSky allsky.sh[1682]:  TTY: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  Non-continuous Capture Method: No
Nov 11 00:46:49 AllSky allsky.sh[1682]:  ZWO SDK version 1, 20
Nov 11 00:46:49 AllSky allsky.sh[1682]: #033[0mWill adjust gain at transitions
Nov 11 00:46:49 AllSky allsky.sh[1682]: Stop the allsky service to end this process.
Nov 11 00:46:49 AllSky allsky.sh[1682]: **********
Nov 11 00:46:49 AllSky allsky.sh[1682]: *** ERROR: dayOrNight isn't DAY or NIGHT, it's ''
Nov 11 00:46:49 AllSky allsky.sh[1682]: *** After fixing, restart the allsky service.
Nov 11 00:46:54 AllSky allsky.sh[1682]: *** Sleeping until you fix the problem.
Nov 11 00:46:54 AllSky allsky.sh[1682]: **********

Apparently there is a problem with "dayfOrNight". I searched through the different files but don't know how to fix this.

Please could you help me?

Thank you in advance.

EricClaeys, 2021-11-11:
@Stef682, sorry you are having a problem.  Would you please attach (don't copy/paste) your settings_ZWO.json file?  If you installed the WebUI it will be in /etc/raspap, otherwise it's in ~/allsky or allsky/config.

Eric

maserowik, 2021-11-11:
@EricClaeys 
Could this be the same issue I have?

log file look very similar.

Mike

Stef682, 2021-11-11:
> @Stef682, sorry you are having a problem. Would you please attach (don't copy/paste) your settings_ZWO.json file? If you installed the WebUI it will be in /etc/raspap, otherwise it's in ~/allsky or allsky/config.
> 
> Eric

Thank you for your reply.

I am using the WebUI, so the file is in /etc/raspap.

Please find attached the requested file. When I tried to attach it, it told me that this type of file is not supported, so I had to put it into a zip file.
[settings_ZWO.zip](https://github.com/thomasjacquin/allsky/files/7520385/settings_ZWO.zip)

EricClaeys, 2021-11-11:
@maserowik, Mike, I'm not aware of you every having the `ERROR: dayOrNight isn't DAY or NIGHT, it's ''` error, have you?

@Stef682, I used to zip files for the same reason you did, but recently have simply been appending ".txt" to the file name.  It's quicker than zip / unzip.
I'll look at the log.  Thanks for attaching it (much easier to read than long copy/paste).

EricClaeys, 2021-11-11:
@Stef682, please run the following command and let me know what it returns:
`sunwait poll angle -6 47.34N 7.30E`

Eric

Stef682, 2021-11-11:
Sorry but nothing happens when I enter this command.

Did you think the problem could be with latitude and longitude? In the camera settings I clicked on "reset to default values", but even with the original coordinates (60.7N and 135.05W) the problem persists. Personally I don't think it comes from there.

Perhaps the easiest would be to uninstall the AllSky program and reinstall it. I had moreover tried several times but without success. But maybe I didn't do it right, I had just deleted the allsky directory and re-installed the program by typing the command lines. Could it be possible that there are files left from my old installation and that these conflict with the latest version of the program?

Could you tell me the procedure to erase everything and redo a clean installation?

Thank you for your help.

linuxkidd, 2021-11-11:
@Stef682 -- to be clear, when you run the command, does it just return to the prompt with no output?  Or does it say something like 'command not found' or.. ??

When I run it on my allsky box, I get:
```
#  sunwait poll angle -6 47.34N 7.30E
NIGHT
```

Thanks!

EricClaeys, 2021-11-11:
@Stef682 I also got results when I tried with tour settings.
Try executing `type sunwait`.  It should return the full path name.  You can also ececute the command and then immediately execute `echo $?`.  That will give the exit code of the command.

Lastly you can try `strace sunwait poll angle -6 47.34N 7.30E` and share the output with us.

Stef682, 2021-11-11:
When I run sunwait poll angle -6 47.34N 7.30E I have no answer, not even an error message (it just returns to the prompt).

When I run only sunwait it shows me explanations (apparently how to use this command).

When I run type sunwait it returns sunwait is /usr/local/bin/sunwait.

I have to execute echo $? after which command? When I run it after sunwait poll angle -6 47.34N 7.30E I get 3, and after type sunwait I get 0.

When I run strace sunwait poll angle -6 47.34N 7.30E I get the following output:

execve("/usr/local/bin/sunwait", ["sunwait", "poll", "angle", "-6", "47.34N", "7.30E"], 0x7eab23e4 /* 42 vars */) = 0
brk(NULL)                               = 0xbeb000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fb3000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76fb2000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/local/bin/sunwait", 4096) = 22
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f70000
mprotect(0x76f74000, 61440, PROT_NONE)  = 0
mmap2(0x76f83000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x76f83000
close(3)                                = 0
munmap(0x76fb2000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=164321, ...}) = 0
mmap2(NULL, 164321, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76f47000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libm.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240r\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=464392, ...}) = 0
mmap2(NULL, 528504, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76ec5000
mprotect(0x76f36000, 61440, PROT_NONE)  = 0
mmap2(0x76f45000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x70000) = 0x76f45000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0@\343\6\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=1265528, ...}) = 0
mmap2(NULL, 1337068, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76d7e000
mprotect(0x76eac000, 65536, PROT_NONE)  = 0
mmap2(0x76ebc000, 28672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12e000) = 0x76ebc000
mmap2(0x76ec3000, 5868, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76ec3000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\270\321\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=116324, ...}) = 0
mmap2(NULL, 180532, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76d51000
mprotect(0x76d6d000, 61440, PROT_NONE)  = 0
mmap2(0x76d7c000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b000) = 0x76d7c000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\274x\1\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1296004, ...}) = 0
mmap2(NULL, 1364764, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76c03000
mprotect(0x76d3b000, 65536, PROT_NONE)  = 0
mmap2(0x76d4b000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x138000) = 0x76d4b000
mmap2(0x76d4e000, 8988, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76d4e000
close(3)                                = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fb1000
set_tls(0x76fb1500)                     = 0
mprotect(0x76d4b000, 8192, PROT_READ)   = 0
mprotect(0x76d7c000, 4096, PROT_READ)   = 0
mprotect(0x76f45000, 4096, PROT_READ)   = 0
mprotect(0x76ebc000, 20480, PROT_READ)  = 0
mprotect(0x76f83000, 4096, PROT_READ)   = 0
mprotect(0x76fb5000, 4096, PROT_READ)   = 0
munmap(0x76f47000, 164321)              = 0
brk(NULL)                               = 0xbeb000
brk(0xc0c000)                           = 0xc0c000
openat(AT_FDCWD, "/etc/localtime", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=2971, ...}) = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=2971, ...}) = 0
read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\r\0\0\0\r\0\0\0\0"..., 4096) = 2971
_llseek(3, -1872, [1099], SEEK_CUR)     = 0
read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\r\0\0\0\r\0\0\0\0"..., 4096) = 1872
close(3)                                = 0
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=2971, ...}) = 0
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=2971, ...}) = 0
exit_group(3)                           = ?
+++ exited with 3 +++

linuxkidd, 2021-11-11:
It looks like `sunwait` is having trouble getting the current time.  Can you try the following process:

```
date
dpkg-reconfigure tzdata
date
```

The `dpkg-reconfigure` command should prompt you to select your timezone.  Technically, this shouldn't cause the problem you're experiencing, but a complete lack of actual current time on the box may.

The two date commands ( one above, one below ) should show us more about what's going on, and if the reconfigure fixed it.

Assuming the 2nd `date` command shows the correct time, retry the `sunwait poll angle -6 47.34N 7.30E` command and see if it works now.

Thanks for your patience!

Stef682, 2021-11-11:
I followed your procedure, but the date and time are correct:

pi@allsky:~ $ date
jeudi 11 novembre 2021, 19:53:25 (UTC+0100)
pi@allsky:~ $ sudo dpkg-reconfigure tzdata

Current default time zone: 'Europe/Paris'
Local time is now:      Thu Nov 11 19:53:50 CET 2021.
Universal Time is now:  Thu Nov 11 18:53:50 UTC 2021.

pi@allsky:~ $ date
jeudi 11 novembre 2021, 19:53:54 (UTC+0100)

linuxkidd, 2021-11-11:
Huh...  It's interesting too, that `sunwait` exit code 3 represents `NIGHT` when I reviewed the code.  Lemme dig through it a bit more.

linuxkidd, 2021-11-11:
Ok, is there any chance you can attach the `~pi/allsky/src/sunwait-src/sunwait.cpp` file for us to review?  It looks like it's skipping the part where it prints 'NIGHT', but the exit code is reported properly.   Looking at the upstream code, I don't see a reason this would happen, so I'm hoping the local code on your Pi holds the key.

Thanks!

Stef682, 2021-11-11:
Please find attached the requested file:
[sunwait.cpp.txt.txt](https://github.com/thomasjacquin/allsky/files/7522819/sunwait.cpp.txt.txt)

Stef682, 2021-11-11:
Please find attached the requested file:
[sunwait.cpp.txt](https://github.com/thomasjacquin/allsky/files/7522825/sunwait.cpp.txt)

linuxkidd, 2021-11-11:
Ok, the source is identical.  In comparing the `strace` output of your execution to mine locally, there is a missing `gettimeofday` call in yours, that mine does perform.

Can you try running `sunwait` with sudo?  
```
sudo sunwait poll angle -6 47.34N 7.30E
```

Stef682, 2021-11-11:
Even putting sudo in front of sunwait poll angle -6 47.34N 7.30E nothing happens, it returns to the prompt.

linuxkidd, 2021-11-12:
Hi @Stef682 ... this is a real head scratcher... I have to think it's some library / compile anomaly.  In hope that is the case, can I ask that you try the following process:
```
sudo apt update && sudo -y full-upgrade
cd ~pi/allsky/src/sunwait-src
make clean
make all
./sunwait poll angle -6 47.34N 7.30E
```

If that works, you can install this newly compiled version with:
```
sudo cp sunwait /usr/local/bin
```

Let us know the outcome.

EricClaeys, 2021-11-12:
@Stef682, you can also try:
`sunwait poll angle -6 47.34N 7.30E > /tmp/xzy; cat /tmp/xyz`
This will run the command and put the output (which so far there hasn't been any) into a temporary file, then display the contents of that file.  This is for the unlikely chance that sunwait doesn't like the device you're on so isn't printing output.

Stef682, 2021-11-12:
I followed your procedure and got this:

pi@allsky:~ $ cd ~pi/allsky/src/sunwait-src
pi@allsky:~/allsky/src/sunwait-src $ make clean
rm -f *.o sunwait
pi@allsky:~/allsky/src/sunwait-src $ make all
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function ‘int main(int, char**)’:
sunwait.cpp:665:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function ‘void sun_RA_dec(double, double*, double*, double*)’:
sunriset.cpp:151:18: warning: variable ‘zs’ set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~
gcc -c -Wall  print.cpp -o print.o
gcc sunwait.o sunriset.o print.o -o sunwait -lm -lstdc++
pi@allsky:~/allsky/src/sunwait-src $ ./sunwait poll angle -6 47.34N 7.30E
NIGHT
pi@allsky:~/allsky/src/sunwait-src $ sudo cp sunwait /usr/local/bin
pi@allsky:~/allsky/src/sunwait-src $

The first time it didn't work, but I decided to give it a second try. And there to my surprise it worked!

Thank you very much for your help!

linuxkidd, 2021-11-12:
Woo hoo!  glad to hear it.  I'll go ahead and close this issue since it seems like just an odd library / compile issue that was resolved with the the update/recompile.  Thanks for sticking in there while we worked through it!

closed 2021-11-12 by linuxkidd

Stef682, 2021-11-12:
Thank you to you and the other members for your help.

EricClaeys, 2021-11-12:
@linuxkidd, should we update the code to get rid of the compiler warnings (which I could have sworn I did...)?

linuxkidd, 2021-11-12:
These warnings are from upstream code ( sunwait )... the best option would be to submit a PR to the sunwait repo to fix them.

EricClaeys, 2021-11-12:
Ah, I thought it looked familiar.  I submitted a PR weeks ago.  Looking at the Issues and PRs against sunwait, I don't think it's being actively maintained.