AllskyTeam/allsky #780: make error - mode_RPiHQ_mean.cpp - Ubuntu 20.04

hoetzgit, 2021-11-11:
The following error occurs when compiling the actual release under Ubuntu 20.04.

...
pi@allsky:~/allsky01$ make -C src all
make: Entering directory '/home/pi/allsky01/src'
2021-11-11 09:30:31 Building capture_RPiHQ program...
mode_RPiHQ_mean.cpp: In function ‘void RPiHQmask(const char*)’:
mode_RPiHQ_mean.cpp:59:35: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
   59 |      compression_params.push_back(CV_IMWRITE_PNG_COMPRESSION);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp:61:35: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
   61 |      compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp: In function ‘void RPiHQcalcMean(const char*, int, double, raspistillSetting&, modeMeanSetting&)’:
mode_RPiHQ_mean.cpp:197:34: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
  197 |     compression_params.push_back(CV_IMWRITE_PNG_COMPRESSION);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp:199:34: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
  199 |     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:127: capture_RPiHQ] Error 1
make: Leaving directory '/home/pi/allsky01/src'
...

I fixed the error with these adjustments:
sed -i -e 's/CV_IMWRITE_PNG_COMPRESSION/cv::IMWRITE_PNG_COMPRESSION/g' src/mode_RPiHQ_mean.cpp
sed -i -e 's/CV_IMWRITE_JPEG_QUALITY/cv::IMWRITE_JPEG_QUALITY/g' src/mode_RPiHQ_mean.cpp

...
pi@allsky:~/allsky01$ make -C src all
make: Entering directory '/home/pi/allsky01/src'
2021-11-11 09:31:38 Building capture_RPiHQ program...
2021-11-11 09:31:42 Done.
make: Leaving directory '/home/pi/allsky01/src'
...

AndreasLMeg, 2021-11-11:
Maybe a opencv version problem? I will check this.

greg-the-maker, 2021-11-11:
I'm getting a similar error on Rasbian Lite with an RPi Zero 2.

`pi@allskyz:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make[1]: Entering directory '/home/pi/allsky/src'
2021-11-11 12:52:58 Installing build dependencies...
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gawk is already the newest version (1:5.1.0-1).
imagemagick is already the newest version (8:6.9.11.60+dfsg-1.3).
jq is already the newest version (1.6-2.1).
lftp is already the newest version (4.8.4-2+b1).
libopencv-dev is already the newest version (4.5.1+dfsg-5).
libusb-1.0-0-dev is already the newest version (2:1.0.24-3).
libusb-dev is already the newest version (2:0.1.12-32).
ffmpeg is already the newest version (7:4.3.3-0+rpt1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
make[1]: Leaving directory '/home/pi/allsky/src'

* Compile allsky software

make[1]: Entering directory '/home/pi/allsky/src'
2021-11-11 12:53:10 Building capture_RPiHQ program...
mode_RPiHQ_mean.cpp: In function ‘void RPiHQmask(const char*)’:
mode_RPiHQ_mean.cpp:59:35: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
   59 |      compression_params.push_back(CV_IMWRITE_PNG_COMPRESSION);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp:61:35: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
   61 |      compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp: In function ‘void RPiHQcalcMean(const char*, int, double, raspistillSetting&, modeMeanSetting&)’:
mode_RPiHQ_mean.cpp:197:34: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
  197 |     compression_params.push_back(CV_IMWRITE_PNG_COMPRESSION);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
mode_RPiHQ_mean.cpp:199:34: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
  199 |     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:127: capture_RPiHQ] Error 1
make[1]: Leaving directory '/home/pi/allsky/src'
make: *** [Makefile:24: all] Error 2
Compile failed!
`

linuxkidd, 2021-11-11:
Hello all,
  Yes, as @AndreasLMeg theorized, this is due to a change in api when moving from OpenCV 3 to  OpenCV 4.  We'll get this patched up and posted to the repo shortly.

In the interim, the commands provided by @hoetzgit will allow compilation until the changes are made and merged into the repo.

```
sed -i -e 's/CV_IMWRITE_PNG_COMPRESSION/cv::IMWRITE_PNG_COMPRESSION/g' src/mode_RPiHQ_mean.cpp
sed -i -e 's/CV_IMWRITE_JPEG_QUALITY/cv::IMWRITE_JPEG_QUALITY/g' src/mode_RPiHQ_mean.cpp
```

Thanks!

closed 2021-11-11 by linuxkidd

AndreasLMeg, 2021-11-11:
https://github.com/thomasjacquin/allsky/pull/783#issue-1051174612