AllskyTeam/allsky #789: RPiHQ camera not found error

adibu, 2021-11-13:
Had an installation as of thee weeks ago was running stable. Decided to reinstall it from scratch (new sd card, new git clone, new install).

Executing /allsky.sh now I am getting:  "***** Starting AllSky ***** RPiHQ Camera not found. Exiting."

Executing libcamera-hello -t 0 works nicely and outputs a video stream so it cannot be the camera hardware itself.
Is this an error or how can I debug further?

Regards,
Adrian

EricClaeys, 2021-11-13:
@adibu do you enable the camera port using raspi-config (or is it config-raspi?).
Look in allsky.sh for how it's looking for rhe camera.  It uses the vcgencmd command, and try thoe commands manually.   Sorry, I am not on my PC right now.

adibu, 2021-11-14:
Hi Eric - many thanks also for the guidance on where to start investigating! And I think I found what is wrong ...

a) yes it is enabled
b) libcamera is able to capture (libcamera-hello -t 0) without any issues
c) vcgencmd get_camera shows supported=0 detected=0 though
d) faking the condition on camera detection to true (1) I can run allsky.sh and it then log what it is using to capture as:
"nice raspistill --nopreview --thumb none --output allskycam.jpg --burst -st --mode 3 --exposure auto --analoggain 1 --awb auto --saturation 50 --quality 95 -a 1104 -a 1036 -a "WALA shutter:1000000 gain:1 (li-Sat Nov 13 13:53:13 2021) br:50 WBR:2.5 WBB:2" -ae 32,0xff,0x808000 
nice: ‘raspistill’: No such file or directory"

Coming back to what changed to my installation before is that the Raspian build used by the PI Image Manager is the latest one with all updates/upgrades.

Searching for changes I found:
"Raspberry Pi is transitioning from a legacy camera software stack based on proprietary Broadcom GPU code to an open-source stack based on libcamera. Raspberry Pi OS images from Bullseye onwards will contain only the libcamera-based stack. Raspberry Pi OS images up to and including Buster will contain the legacy Raspicam stack, though the libcamera stack and applications can be installed using apt, or built by following the normal build instructions."

see https://www.raspberrypi.com/documentation/accessories/camera.html

On my Raspbi:
cat /etc/os-release > PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"

-> There we go I think I found the issue and will restage with Buster, if that runs this is the root cause.

Regards,
Adi

adibu, 2021-11-14:
https://downloads.raspberrypi.org/raspios_armhf/release_notes.txt
  * New camera demo applications (libcamera-still and libcamera-vid) have replaced raspistill and raspivid

adibu, 2021-11-14:
@EricClaeys After re-installation to the last Buster release I can confirm:

- raspistill is depreciated with the latest Raspberry OS release Bullseye as of end of October - the new command is libcamera
- it works like a charm once switched back to Raspberry OS release Buster which still has raspistill included

Many thanks for your input the hint what allsky.sh does (I am not a developer but can figure out when pointed to the right spot) - that saved my weekend. 

Eventually this should be noted in the installation guide for anyone installing it now - I guess the switch to libcamera is not a quick one to implement.

Regards,
Adi

EricClaeys, 2021-11-14:
@adibu is raspstill on the Bullseye release or did they remove it?

adibu, 2021-11-15:
Yes that is yet an unsolved mystery for me - fact is it now works under Buster while id didnt under Bullseye.

The change note mentions for the latest Bullseye release:
  * New default camera subsystem based on libcamera
  * New camera demo applications (libcamera-still and libcamera-vid) have replaced raspistill and raspivid
  * Legacy camera subsystem removed from 64-bit RPi OS (still available on 32-bit)

vcgencmd remains part of the command set but eventually the get_camera method has yet an issue due to the change or it is a one off issue with my installation. I have overriden the detection and then it failed at raspistill which made me search for build differences in the end.

EricClaeys, 2021-11-15:
@adibu looks like our development team has some work ahead of us, to get AllSky to work on Bullseye.  Thanks for bringing this to our attention.

EricClaeys, 2021-11-18:
@adibu, I added a [Discussion item](https://github.com/thomasjacquin/allsky/discussions/802) about this, so will close this Issue.
I have a version of the capture program that works with libcamera, but need people to test on Bullseye.

closed 2021-11-18 by EricClaeys