AllskyTeam/allsky #790: allsky install: capture.cpp:1:28: fatal error: opencv2/core.hpp

AndreasLMeg, 2021-11-13:
Hello erveryone !

Would make a 2nd testsystem, took an old RP3 and installed allsky. 
Got this message: capture.cpp:1:28: fatal error: opencv2/core.hpp: Datei oder Verzeichnis nicht gefunden

Pi Model 3B incompatible / wrong libopencv version ?

Apparently there is a difference between Raspi3 and Raspi4.
Has this been observed or corrected before?
Before I look around for a long time, I'd rather ask you for a solution.

Thanks very much.

Install logging
```
pi@raspberrypi:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make[1]: Verzeichnis „/home/pi/allsky/src“ wird betreten
2021-11-13 14:34:44 Installing build dependencies...
OK:1 http://archive.raspberrypi.org/debian stretch InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Aktualisierung für 56 Pakete verfügbar. Führen Sie »apt list --upgradable« aus, um sie anzuzeigen.
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
ffmpeg ist schon die neueste Version (7:3.2.15-0+deb9u4).
gawk ist schon die neueste Version (1:4.1.4+dfsg-1).
imagemagick ist schon die neueste Version (8:6.9.7.4+dfsg-11+deb9u13).
jq ist schon die neueste Version (1.5+dfsg-1.3).
lftp ist schon die neueste Version (4.7.4-1).
libopencv-dev ist schon die neueste Version (2.4.9.1+dfsg1-2+deb9u1).
libusb-1.0-0-dev ist schon die neueste Version (2:1.0.21-1).
libusb-dev ist schon die neueste Version (2:0.1.12-30).
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  coinor-libipopt1v5 libmumps-seq-4.10.0 libraw15 lxkeymap python-cairo python-gobject python-gobject-2 python-gtk2 python-xklavier realpath
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 56 nicht aktualisiert.
make[1]: Verzeichnis „/home/pi/allsky/src“ wird verlassen

* Compile allsky software

make[1]: Verzeichnis „/home/pi/allsky/src“ wird betreten
2021-11-13 14:34:53 Building capture program...
capture.cpp:1:28: fatal error: opencv2/core.hpp: Datei oder Verzeichnis nicht gefunden
 #include <opencv2/core.hpp>
                            ^
compilation terminated.
Makefile:121: die Regel für Ziel „capture“ scheiterte
make[1]: *** [capture] Fehler 1
make[1]: Verzeichnis „/home/pi/allsky/src“ wird verlassen
Makefile:24: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
Compile failed!
```
some system infos:
```
Linux raspberrypi 5.10.74-v7+ #1471 SMP Mon Oct 18 17:06:24 BST 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Oct 22 15:56:20 2021 from 10.0.0.21
pi@raspberrypi:~ $ pinout
,--------------------------------.
| oooooooooooooooooooo J8     +====
| 1ooooooooooooooooooo        | USB
|                             +====
|      Pi Model 3B  V1.2         |
|      +----+                 +====
| |D|  |SoC |                 | USB
| |S|  |    |                 +====
| |I|  +----+                    |
|                   |C|     +======
|                   |S|     |   Net
| pwr        |HDMI| |I||A|  +======
`-| |--------|    |----|V|-------'

Revision           : a02082
SoC                : BCM2837
RAM                : 1024Mb
Storage            : MicroSD
USB ports          : 4 (excluding power)
Ethernet ports     : 1
Wi-fi              : True
Bluetooth          : True
Camera ports (CSI) : 1
Display ports (DSI): 1

```

linuxkidd, 2021-11-15:
```
libopencv-dev ist schon die neueste Version (2.4.9.1+dfsg1-2+deb9u1).
```
^^ That's a very old version of OpenCV.. running on Debian 9 ( Stretch ).  I don't think there's any plan to support that old of a release.. best to upgrade to newer Raspberry Pi OS version.

Thanks!

EricClaeys, 2021-11-15:
@linuxkidd, @ckuethe,
this brings up an important topic.  What versions of the operating system will we support?
Stretch?  Debian version 11 (bullseye) that came out a couple weeks ago (and that doesn't work with the RPiHQ camera)?
We should probably state this in the README.md file.

AndreasLMeg, 2021-11-15:
Thanks for the information. 
Yes, this is a very old Raspi that I found there ;-)
I will take a look at the update on the weekend and report my findings if necessary

EricClaeys, 2021-11-16:
@AndreasLMeg please note that if you install the latest Raspberry OS then raspistill isn't there and the RPiHQ camera won't work.
We're going to need to use libcamera-still instead, and make sure it supports all our options.

Longer term we should use the libcamera library instead of their command, at which point the ZWO and RPiHQ code can be a lot closer to the same.

AndreasLMeg, 2021-11-16:
@EricClaeys 
I close the issue, because of the very old Raspberry Pi OS....

closed 2021-11-16 by AndreasLMeg