AllskyTeam/allsky #791: Raspberry camera always in autoexposure

Fredooo51, 2021-11-13:
Hi.

Following upgrade to 0.8.1 from 0.8.0, night image were all captured with autoexposure ON. 

I find a woarkaround by commenting out capture_extra_parameters from allsky.sh. Before that, these two parameters were always added to arguments list (looking at capture_args.txt): --mean-value 0.3 -autoexposure 1

EricClaeys, 2021-11-13:
@Fredooo51 I assume you meant config.sh, which is where the variable is defined?
Where you having a problem with the autoexposure?  @AndreasLMeg improved the algorithm a couple days ago.

AndreasLMeg, 2021-11-14:
Sorry, for the parameter in config.sh. It's a BUG to set these default values. Only for RPiHQ users it should be the way to activate the the new  improved algorithm.

I will fix this.
@Fredooo51 : thx for the response.

AndreasLMeg, 2021-11-14:
made pull request  https://github.com/thomasjacquin/allsky/pull/792 to fix this bug,

AndreasLMeg, 2021-11-14:
@Fredooo51,
could you give a feedback ? - with activated parameter the "mean" algorithm for RPiHQ is active - do you have keogram and timelapse - this would be very important for me.

rodrigoluk, 2021-11-14:
Hi,
So, I also discovered that capture_RPiHQ would end up taking very short exposures so I changed the variable to "". If mode mean is available now, how do I turn it on? What files do I need to change? 
Thanks 😊

AndreasLMeg, 2021-11-14:
@rodrigoluk 

1. config.sh: CAPTURE EXTRA PARAMETERS="-mean-value 0.3 -autoexposure 1"
2. GUI: Activate "Auto-Gain"

please give me feedback (allsky.log, image, url, keogram).

AndreasLMeg, 2021-11-14:
some background infos: #794

Fredooo51, 2021-11-14:
Hi Andreas.

Yes, it was in config.so. I have Keogram and Timelapse active. Autoexposure issue occurred in night mode where I set manually exposure to 25s, it was not taken into account since autoexposure was on whatever I set.

Thanks for your support.

rodrigoluk, 2021-11-14:
This is daytime -after enabling mean mode (with debug level 1):

CAPTURE_EXTRA_PARAMETERS="--mean-value 0.3 -autoexposure 1"
Auto-Gain = yes

Nov 14 14:26:37 allsky allsky.sh[23230]: Capturing & saving image...
Nov 14 14:26:37 allsky allsky.sh[23230]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 3.24,1.47 --hflip --vflip --saturation 50 --quality 95 -a "Brambles Observatory `date +\"%T %d-%m-%Y\"` shutter:1000000 gain:1" -ae 32,0xff,0x808000
Nov 14 14:26:45 allsky allsky.sh[23230]: Capturing & saving image done, now wait 30 seconds...

This is night time (started 16:49 GMT):

Nov 14 17:03:49 allsky allsky.sh[23230]: Capturing & saving image done, now wait 3 seconds...
Nov 14 17:03:52 allsky allsky.sh[23230]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: XXXXXN, longitude: XXXXXXE
Nov 14 17:03:52 allsky allsky.sh[23230]: Capturing & saving image...
Nov 14 17:03:52 allsky allsky.sh[23230]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 3.24,1.47 --hflip --vflip --saturation 50 --quality 95 -a "Brambles Observatory `date +\"%T %d-%m-%Y\"` shutter:1000000 gain:1" -ae 32,0xff,0x808000
Nov 14 17:04:00 allsky allsky.sh[23230]: Capturing & saving image done, now wait 3 seconds...

Images are dark (exposure does not change).

If I change to: 

CAPTURE_EXTRA_PARAMETERS=""

And leave Auto-Gain = yes

And restart allsky I get well/better exposed images (60s).

Nov 14 17:11:12 allsky allsky.sh[18631]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: XXXXXXN, longitude: XXXXXXE
Nov 14 17:11:12 allsky allsky.sh[18631]: Capturing & saving image...
Nov 14 17:11:13 allsky allsky.sh[18631]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 1 --awb off --awbgains 3.24,1.47 --hflip --vflip --saturation 50 --quality 95 -a "Brambles Observatory `date +\"%T %d-%m-%Y\"` shutter:1000000 gain:1" -ae 32,0xff,0x808000

Thanks for your support :-)

R.

ps: note that the shutter and gain values do not change in the title string...but that is another issue...

AndreasLMeg, 2021-11-14:
@rodrigoluk 

With 
1. **debug level 3** and
2. **CAPTURE_EXTRA_PARAMETERS="-mean-value 0.3 -autoexposure 1"**
I got this messages (only the important)

```
Nov 14 18:34:00 allsky allsky.sh[20690]: Processing argument: -mean-value
Nov 14 18:34:00 allsky allsky.sh[20690]: Processing argument: -autoexposure

Nov 14 18:34:00 allsky allsky.sh[20690]:  Mode Mean: 1 (yes)
Nov 14 18:34:00 allsky allsky.sh[20690]:  Mode Mean Value: 0.300
Nov 14 18:34:00 allsky allsky.sh[20690]:  Camera: Raspberry Pi HQ camera

Nov 14 18:34:00 allsky allsky.sh[20690]: ExposureLevel: -718.53646850 ... 354.29612288

Nov 14 18:34:00 allsky allsky.sh[20690]: MeanCnt: 0
Nov 14 18:34:00 allsky allsky.sh[20690]: mean_historySize: 3
Nov 14 18:34:00 allsky allsky.sh[20690]: i=1: idx=1 mean=1.0000 exp=0
Nov 14 18:34:00 allsky allsky.sh[20690]: i=2: idx=2 mean=0.0000 exp=0
Nov 14 18:34:00 allsky allsky.sh[20690]: i=3: idx=0 mean=0.1796 exp=0
Nov 14 18:34:00 allsky allsky.sh[20690]: mean_forecast: 0.3592
Nov 14 18:34:00 allsky allsky.sh[20690]: values: 9
Nov 14 18:34:00 allsky allsky.sh[20690]: mean_diff: 0.0093
Nov 14 18:34:00 allsky allsky.sh[20690]: ExposureChange: 3 (3)
Nov 14 18:34:00 allsky allsky.sh[20690]: Mean: 0.2907 (0.0093) Exposure level:1 (1) Exposure time:1.01944064 analoggain:1.00

Nov 14 18:34:10 allsky allsky.sh[20690]: MeanCnt: 1
Nov 14 18:34:10 allsky allsky.sh[20690]: mean_historySize: 3
Nov 14 18:34:10 allsky allsky.sh[20690]: i=1: idx=2 mean=0.0000 exp=0
Nov 14 18:34:10 allsky allsky.sh[20690]: i=2: idx=0 mean=0.1796 exp=0
Nov 14 18:34:10 allsky allsky.sh[20690]: i=3: idx=1 mean=0.0022 exp=1
Nov 14 18:34:10 allsky allsky.sh[20690]: mean_forecast: 0.0000
Nov 14 18:34:10 allsky allsky.sh[20690]: values: 9
Nov 14 18:34:10 allsky allsky.sh[20690]: mean_diff: 0.2593
Nov 14 18:34:10 allsky allsky.sh[20690]: ExposureChange: 48 (45)
Nov 14 18:34:10 allsky allsky.sh[20690]: Mean: 0.0407 (0.2593) Exposure level:49 (48) Exposure time:2.56882945 analoggain:1.00

Nov 14 18:34:23 allsky allsky.sh[20690]: MeanCnt: 2
Nov 14 18:34:23 allsky allsky.sh[20690]: mean_historySize: 3
Nov 14 18:34:23 allsky allsky.sh[20690]: i=1: idx=0 mean=0.1796 exp=0
Nov 14 18:34:23 allsky allsky.sh[20690]: i=2: idx=1 mean=0.0022 exp=1
Nov 14 18:34:23 allsky allsky.sh[20690]: i=3: idx=2 mean=0.0015 exp=49
Nov 14 18:34:23 allsky allsky.sh[20690]: mean_forecast: 0.0007
Nov 14 18:34:23 allsky allsky.sh[20690]: values: 9
Nov 14 18:34:23 allsky allsky.sh[20690]: mean_diff: 0.2788
Nov 14 18:34:23 allsky allsky.sh[20690]: ExposureChange: 55 (7)
Nov 14 18:34:23 allsky allsky.sh[20690]: Mean: 0.0212 (0.2788) Exposure level:104 (55) Exposure time:7.40699770 analoggain:1.00
```

Can you send me you allsky.log file for checking ?

rodrigoluk, 2021-11-14:
Oh! Actually, the images coming are quite dark so, I disabled Auto-Gain. Below you can see the difference:
[image]
[image]

AndreasLMeg, 2021-11-14:
first i want to check the logging - to be sure the "mean mode" is running, then we can check the images and the mean value.

AndreasLMeg, 2021-11-14:
for me it only makes sense to work with autogain, so I did not consider this situation. I have already found the cause of the wrong texts (only in this situation), but I still have to come up with a solution ...

Autogain active: please change the mean value to 0.5 or more, if the image is to dark for you.

rodrigoluk, 2021-11-15:
Ok. Andreas - I'm very grateful youare looking into this. Please see a compress log from yesterday and today.
Much appreciated.
[allsky.log.gz](https://github.com/thomasjacquin/allsky/files/7536985/allsky.log.gz)

AndreasLMeg, 2021-11-15:
@rodrigoluk 
I have now made an adjustment (not merged) so that no incorrect values ​​are displayed.

But I have to consider: mean control works DAY and night with the same parameters. 
1. autoGain active: (1 ... Gain), otherwise always with Gain
2. autoExposure active: (1µs ... Exposure) otherwise always with Exposure

That means it only makes sense to set both values ​​to AUTO.

AndreasLMeg, 2021-11-15:
> 
> 
> Ok. Andreas - I'm very grateful youare looking into this. Please see a compress log from yesterday and today. Much appreciated. [allsky.log.gz](https://github.com/thomasjacquin/allsky/files/7536985/allsky.log.gz)

I need a logging with activated mean control. (+autoExposre +autoGain)

rodrigoluk, 2021-11-15:
Somewhere after 17:04 I changed to the parameters you recommended and then
at 18:15 I switched them off as I was getting short dark images anyway.
Exposure was 60s but the gain seems to be stuck at 1. I switch back on at
22:43 for the rest of the night but now all exposures where too fast (like
10s). I've always saved the settings to ensure the software restarts.

Tonight we have a clear night here. And I'm running with mean =1 auto gain
=1  from about 22:40 2but all exposures so far are ~10s. I'll send you
another log tomorrow.

I just noticed something weird in the image string. With debug level 3
there is a weird date and time showing:

Nov 15 22:53:38 allsky allsky.sh[9214]: Capture command: nice raspistill
--nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure
auto --analoggain 1 --awb off --awbgains 3.24,1.47 --hflip --vflip
--saturation 50 --quality 95 -a 1104 -a "Brambles Observatory `date +\"%T
%d-%m-%Y\"` shutter:1000000 gain:1 (li-Fri Nov 12 13:39:58 2021) br:50
WBR:3.24 WBB:1.47" -ae 32,0xff,0x808000
Nov 15 22:53:39 allsky allsky.sh[9214]: upload.sh: Copying image.jpg to
/var/www/html/allsky//image.jpg

Very grateful for your help and support.

rodrigoluk, 2021-11-15:
Hi,

I'm confused here. There is no autoexposure setting for the RPiHQ camera in
the software. That seems to be missing.

Kind regards and thanks for your help again.

R.

AndreasLMeg, 2021-11-16:
```
Nov 15 22:53:38 allsky allsky.sh[9214]: Capture command: nice raspistill
--nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure
auto --analoggain 1 --awb off --awbgains 3.24,1.47 --hflip --vflip
--saturation 50 --quality 95 -a 1104 -a "Brambles Observatory `date +\"%T
%d-%m-%Y\"` shutter:1000000 gain:1 (li-Fri Nov 12 13:39:58 2021) br:50
WBR:3.24 WBB:1.47" -ae 32,0xff,0x808000

```

`(li-Fri Nov 12 13:39:58 2021)` : this is the compile timestamp, very useful for me and my tests. 
`--exposure auto`: looks like, the the parameters are not correct. Is "-autoexposure 1" set? You have to use CAPTURE_EXTRA_PARAMETERS, setting is not available in GUI 

**Please check parameters and GUI setting (autoGain active) und please send me allsky.log** 
  CAPTURE_EXTRA_PARAMETERS="-mean-value 0.3 -autoexposure 1"

rodrigoluk, 2021-11-16:
Hi Andreas, Please see attached the log from last night. I had a power failure at one point (close to midnight). 
CAPTURE_EXTRA_PARAMETERS="--mean-value 0.5 -autoexposure 1" (config.sh) and Auto Gain =yes are both set.
Note changes at 17:02, 22:39 and 23:26 (CAPTURE_EXTRA_PARAMETERS=""). The image below shows the thumbs from when the night exposures start at 16:48 and to 19:22. 

When I change to CAPTURE_EXTRA_PARAMETERS="--mean-value 0.5 -autoexposure 1" I see --exposure auto in the log. I was under the impression that the value here should change as the mean changes? 

[image]

[allsky.log.gz](https://github.com/thomasjacquin/allsky/files/7546166/allsky.log.gz)

Again, many thanks for your support.

AndreasLMeg, 2021-11-16:
quick review:
```
Nov 14 22:48:28 allsky allsky.sh[1387]: Processing argument: --mean-value
Nov 14 22:48:28 allsky allsky.sh[1387]: Processing argument: 0.5
Nov 14 22:48:28 allsky allsky.sh[1387]: Processing argument: -autoexposure

```
I think  --mean-value is not supported, -autoexposure is set

maybe that is the reason:  use "-mean-value" instead of "--mean-value"   
`
"-mean-value 0.3 -autoexposure 1" 
`

rodrigoluk, 2021-11-16:
I will give that a try, Makes more sense now. Thanks! I will report as soon as it gets dark.

Manythanks!

EricClaeys, 2021-11-16:
@AndreasLMeg FYI, the example in config.sh has "--mean-value" (two minus signs).
I suggest we add that setting to the WebUI and config.repo/settings_RPiHQ.json.repo, then remove the comment from config.sh.repo.

EricClaeys, 2021-11-16:
@AndreasLMeg I'm running with auto-exposure on with the libcamera changes.  So far it's running fine although I haven't tested it much.

Attached is the .cpp file I used.  You'll see a bunch of comments with "xx" in them - those are lines I commented out that don't work in libcamera.  We'll want to remove them going forward.

libcamera doesn't have a digital gain setting.  Instead, if the gain is set higher than what the camera supports (and I don't know what value that is), the remainder is the digital gain.
Line 394 will need to change as will other places that mention digital gain.

Please check the "if" block on line 451.  I don't think the check for myModeMeanSetting.mode_mean is needed since this "if" statement inside it is the same as the "if" statement in the "else" clause.

[capture_RPiHQ.cpp-libcamera.txt](https://github.com/thomasjacquin/allsky/files/7549652/capture_RPiHQ.cpp-libcamera.txt)

rodrigoluk, 2021-11-16:
Changing the "--mean-value" to "-mean-value" has done the trick. Now I'm
getting exposure values as expected 😁

Yes, it would make sense to restore this setting to the WebUI.

Thank you!

R.

EricClaeys, 2021-11-16:
@Fredooo51 if this fixes the problem for you, would you please close the Issue?
Thanks.

rodrigoluk, 2021-11-16:
Done.

AndreasLMeg, 2021-11-17:
> @AndreasLMeg I'm running with auto-exposure on with the libcamera changes. So far it's running fine although I haven't tested it much.
>
> Attached is the .cpp file I used. You'll see a bunch of [...]

@EricClaeys 
I'll check the changes over the weekend. Isn't it a good time for a new version BEFORE we make the libcamera adjustments?

EricClaeys, 2021-12-04:
Closing - no response from original poster in almost 3 weeks.

closed 2021-12-04 by EricClaeys