AllskyTeam/allsky #805: allsky-website installation

Amenophis54, 2021-11-17:
Hello,

I can't clearly find the installation steps to get the allsky-website.

Indeed, all the steps for the allsky-portal are very well described but I cannot understand how to install the website on my RPI 4.

I do not have a website and I access my allsky via the IP address of the Raspberry Pi .

Thank you very much.

EricClaeys, 2021-11-17:
@Amenophis54 On [the main Git page](https://github.com/thomasjacquin/allsky) search  for the heading **Show your sky on your own website**.  There's not much to do to install allsky-website.
Once you've installed it, look at the descriptions of the settings on the [allsky-website page](https://github.com/thomasjacquin/allsky-website).

If this answers you question, please close this Issue.  Thanks.

Amenophis54, 2021-11-17:
Thank you very much for your answer.

I saw this section, but when it says: 
"download the source files from this repository: https://github.com/thomasjacquin/allsky-website.git."

How to save the requested files?
Do I have to download them one by one?
Are there any commands to run in the terminal? 

Then, as I want to save the website on my raspberry, I have to type the command:
website / install.sh

Where should I note this order?

EricClaeys, 2021-11-17:
@Amenophis54, the documentation isn't as clear as it could be.
You can host the website on a different machine or on the Pi.
If on a different machine you need to "download the source files..."
If on the Pi, you just need to execute `website/install.sh` then set up the variables as shown so it copies the startrails and other files to the website.  Then update the settings per the allsky-website page.

I'll update the documentation to (hopefully) be more clear.

Amenophis54, 2021-11-17:
Thank you very much for your reply.
I will try to feel that way with your details.

closed 2021-11-17 by Amenophis54

EricClaeys, 2021-11-17:
@Amenophis54 My pleasure.  If you have any other questions, please add them to the Discussions link, not an Issue.  Thanks much.