AllskyTeam/allsky #810: Issue using Outline Font

maserowik, 2021-11-19:
@EricClaeys 
I added the outline font option here is a snippet from the allsky.log
[image]
here is a snippet from the webui
[image]
here is what showing on my website
[image]

as soon as i remove the show outline the lines go away
[image]

Thanks Mike

EricClaeys, 2021-12-24:
@maserowik.  Are you still seeing the aurora forecast being older than the current time?  If so, try deleting your browser cache by pressing F12, then in the "Network" tab, click on the checkbox to disable the cache. The refresh the window.

EricClaeys, 2022-06-03:
Closing Issue - no response in 6 months.

closed 2022-06-03 by EricClaeys