AllskyTeam/allsky #811: raspistill & libcamera-still

greg-the-maker, 2021-11-19:
I have been trying to get AllSky to work with the raspberry pi zero 2 and HQ camera.

I have the latest Lite (31/10/21) and this distribution doesnt have raspistill, it seems that everything has / is in the process of, swapping to libcamera (https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps). As a consequence I get errors with allsky.

`Nov 19 13:29:26 allskyz allsky.sh[709]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure auto --analoggain 1 --awb auto --vflip --saturation 50 --quality 95 -a 1104 -a>
Nov 19 13:29:26 allskyz allsky.sh[740]: nice: ‘raspistill’: No such file or directory`

I have tried to fix the problem by doing a raspi-update (this does install raspistill etc, but the paths dont work?), but the errors persist.  I am well outside of my skills set and I have no clue as to how one might fix the issue.

Any help greatly welcome, thanks.

P.S. The camera wouldnt actually detect on the first install, I had to manually enable the camera in config.txt and then install libcamera-apps, vcgencmd get_camera now both supports and detects 1 camera.

AndreasLMeg, 2021-11-19:
@greg-the-maker bullseye not supported, but they try to solve it. Please read this: "RPiHQ cameras on Bullseye" #802

AndreasLMeg, 2021-11-19:
I understand you. I also had to learn it a few days before ;-)

EricClaeys, 2021-11-20:
@greg-the-maker, do you have the most recent `allsky/install.sh`?  It should have given you a message saying the RPiHQ camera isn’t supported on Bullseye, and pointed you to a GitHub page.
I have a version of the program that works with Bullseye.

greg-the-maker, 2021-11-20:
I did a git clone as per the instructions. I can't recall seeing a message about it, but I could have glossed over it. Although I did install 4 or 5 times during my learnings....

Shand359, 2021-11-25:
> @greg-the-maker, do you have the most recent `allsky/install.sh`? It should have given you a message saying the RPiHQ camera isn’t supported on Bullseye, and pointed you to a GitHub page. I have a version of the program that works with Bullseye.

Eric do u have a link to this updated file for the HQ camera in Bullseye OS...?

EricClaeys, 2021-11-26:
@Shand359, others,
For a TEST version of capture_RPiHQ that works on Bullseye, see my post around the middle of the page [here](https://github.com/thomasjacquin/allsky/discussions/802).
This version also supports most of the overlay options that are in the ZWO version (time, exposure, brightness, etc.).

Feedback, both good and bad, is appreciated.

EricClaeys, 2021-11-26:
@greg-the-maker, the message about "RPiHQ not supported on Bullseye" would have been in /var/log/allsky.log.  When Allsky runs it outputs all messages there.

greg-the-maker, 2021-11-26:
@EricClaeys I'll give the test version a try over the weekend.

I didn't think to look in the logs, I'd have had no idea where to look anyway. For people who know these things stuff like this is obvious, However, as this is all unknown and confusing nothing is obvious.

EricClaeys, 2021-12-03:
@greg-the-maker, any luck?

greg-the-maker, 2021-12-03:
@EricClaeys Sort of, I found the basic issue with the camera is Bullseye itself and specific bugs related to the Zero 2 and cameras. I found a thread here on the subject | https://github.com/thomasjacquin/allsky/discussions/802#discussioncomment-1727420 | I did follow the instructions as suggested with the new cpp files, but run into problems again.

`* Compile allsky software

make[1]: Entering directory '/home/pi/allsky/src'
2021-12-03 15:56:05 Building capture_RPiHQ program...
/usr/bin/ld: /tmp/cc7knhw2.o: in function `get_focus_measure(cv::Mat, modeMeanSe                                                                           tting&)':
/home/pi/allsky/src/mode_RPiHQ_mean.cpp:50: undefined reference to `Log(int, cha                                                                           r const*, ...)'
/usr/bin/ld: /tmp/cc7knhw2.o: in function `RPiHQcalcMean(char const*, int, doubl                                                                           e, raspistillSetting&, modeMeanSetting&)':
/home/pi/allsky/src/mode_RPiHQ_mean.cpp:127: undefined reference to `Log(int, ch                                                                           ar const*, ...)'
/usr/bin/ld: /home/pi/allsky/src/mode_RPiHQ_mean.cpp:130: undefined reference to                                                                            `Log(int, char const*, ...)'
/usr/bin/ld: /home/pi/allsky/src/mode_RPiHQ_mean.cpp:131: undefined reference to                                                                            `Log(int, char const*, ...)'
/usr/bin/ld: /home/pi/allsky/src/mode_RPiHQ_mean.cpp:138: undefined reference to                                                                            `Log(int, char const*, ...)'
/usr/bin/ld: /tmp/cc7knhw2.o:/home/pi/allsky/src/mode_RPiHQ_mean.cpp:152: more u                                                                           ndefined references to `Log(int, char const*, ...)' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: capture_RPiHQ] Error 1
make[1]: Leaving directory '/home/pi/allsky/src'
make: *** [Makefile:24: all] Error 2
Compile failed!
`

I had a quick look around for the issue and found something about the camera not detecting, but alas ran out of time to take things any further. However, it's Friday again and I'll see what I can do in the coming days :)

EricClaeys, 2021-12-03:
@greg-the-maker, We JUST posted an updated Allsky that _should_ work with Bullseye.  Please grab it and let me know how it goes.
Thanks - Eric

greg-the-maker, 2021-12-03:
@EricClaeys works a treat, fantastic job!  Thank you everyone :)

closed 2021-12-03 by greg-the-maker