AllskyTeam/allsky #813: updated allsky software, now cand access pi

Skippyinspace, 2021-11-20:
The usual if it can go wrong it will go wrong has happened. Ive updated the allsky software through VNC using the instructions on the wiki.
I  was given the allsky.local or IP of 169.254.167.220 and rebooted the pi which seemed to take a long time and I didn't get back in through VNC.
Ive tried allsky.local and the IP address and my PC browser just times out.
Ive tried VNC and it also times out.
If run IP Scanner and it shows the original PI at its old address of 192.168.1.250 (or 192.168.1.6) and is the only Pi that shows on Ip Scanner
I have also un plugged and replugged in the pi to reboot it.

Pi is in Observatory and the internet can be a bit dodgy but Ive plugged an ethernet cable from the pi into a BT Mesh disc (Which was all working fine before I try updating)

Im very tempted to reformat or get a new SD card as the menu structure was a bit of a mess before trying the update as I couldn't change settings in the GUI and had to manually change them in a config file which did seem to be in a different directory that it should have been in, Which is why I decided to do the update.

Any thoughts
ian
www.skippyinspace.com/allsky

EricClaeys, 2021-11-20:
@Skippyinspace An IP address that begins with 169.254 is what computers give themselves when they are not connected to a network.  For example, if you turn off Wifi on your laptop and reboot it will have an address like that.

What gave you the 169.254.167.220 address?
Have you tried usuing the 192.168.1.250 or 192.168.1.6 addresses?

The Allsky software doesn't change your network settings as far as I know.  You will need to log in to the Pi using a keyboard, mouse, and monitor in order to get the network working.

Skippyinspace, 2021-11-20:
Thanks for the response. my answers below

What gave you the 169.254.167.220 address? - The pi  when the GUI was
updated.
Have you tried usuing the 192.168.1.250 or 192.168.1.6 addresses? yes I
did.and it just timed out

I think Ill take the pi from the Observatory tomorrow and give it a try in
the radio room computer monitors etc and see what happens with that

Thanks again

Ian

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

EricClaeys, 2021-11-21:
@Skippyinspace Interesting.  When you got the 169.254.167.220 address were you on the Pi's keyboard/monitor, or connected via LAN or WiFi ?

Skippyinspace, 2021-11-21:
connected by wifi via vnc

Skippyinspace, 2021-11-21:
Moved pi inside to radio room and all works fine, moved back to the
observatory and again wont connect to IP address so its clearly a wifi
issue.

Will see what happens as I have a meteor camera to install later this week
early next week in the same location.

Quite annoying as I have a BT Mesh disk in the Observatory saying its
connected to the wifi net work.

Thanks for the help.

Ian

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Skippyinspace, 2021-11-21:
After some investigation I discovered the wifi at the back of the house was
on a different channel to that at the front of the house. have now managed
to do the update etc,
So now to get the setting right for the ZWO178 and update the FTP settings.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

EricClaeys, 2021-11-26:
@Skippyinspace, if your problem has been resolved, and it sounds like it is, would you please close the Issue?
Thanks.

linuxkidd, 2021-12-11:
Appears to be resolved, closing issue.  Please re-open if it's not working still.

Thanks!

closed 2021-12-11 by linuxkidd