AllskyTeam/allsky #818: Feature request: Set camera cooler on when 'nighttime' starts, off when 'daytime' starts

Jonk2, 2021-11-21:
I have a ZWO ASI 178MC-Cool and I've noticed something that I think would be a wise thing to do.

During the day, short exposures are used and the sensor temperature doesn't get that high with respect to the ambient temperature. unless of course the sun's up in which case the camera sensor can get very high, and above the operating range of the cmos sensor.

The noise and any hot pixels aren't apparent on daytime exposures, only nighttime exposures.

Trying to cool the sensor with high ambient or high camera temperatures can cause the TEC to saturate, which can actually make the problem worse - it ends up heating the sensor.

I've tested this at work with a bench setup and I believe trying to cool during the day / sunlight is a bad idea.

I would like a feature to only have the camera cooler come on when nighttime starts, in other words have it on at the same time as whatever is set as sunset.

I've heard of others doing this manually using the sunwait module, so I guess it's fairly easy to do?

For daytime high temperatures in the camera dome, I'm in the process of writing python code to run cooling fans when the internal temperature is above a setpoint and the dew heater when the external temperature is below a setpoint. Advanced code will follow to use humidity and dew points.

The dew heater works perfectly well so far based on external temperature, but I'll have to wait for next spring / summer to try the cooler.

Anyway, sensor cooling only at 'nighttime' please!

EricClaeys, 2022-06-03:
@Jonk2   Is this still needed, or did your fan cool the temperature enough during the day that you can keep the camera cooler on all the time?
Assuming you still need it, wouldn't it be better to have different temperatures for day and night, rather than turn the cooler off completely during the day?
I'd think ultimately you'd want the camera cooler set temperature to somehow be related to the ambient.  For example, the cooler would be set to 20 degrees below ambient.  That's how most astronomy cameras/software does it.  It's always a relative number, and the camera manufacture specifies the max delta.

Jonk2, 2022-06-07:
Hi Eric - yes I believe this will be a great feature.

I have bench tested a TEC within a dome at work, and subjected it to ambient and forced raised temperatures.
The problem with TECs is if they cannot dump the heat on the hot side, the cool side starts to warm up and it basically can go into thermal runaway, albeit very slowly. Ultimately, this will lead to component failure.

With this in mind, my original plan of cooling the dome and therefore camera enclosure with a TEC during warm / sunny days is not going to work, and I would suggest that cooling the CMOS sensor with the camera's TEC during the day isn't necessary and can actually be counterintuitive, and possibly dangerous to the hardware.

My current script measures the internal and external temperatures and a setpoint is used which is 4 Deg C above external. If the internal is above the setpoint, a relay switches 2 small cooling fans on, and if internal is below the setpoint, a relay turns the heater on.

I'm yet to have any significant internal temperature rises during a hot sunny day and likewise at night, I don't have any dew or internal condensation - it's important to understand that my camera dome isn't sealed and is designed to breathe.

Regarding the sensor cooling, at a sensor temperature of around 15 Deg C and higher, I start to see hot pixels. I would like to aim for a sensor temperature of 5, or as low as 0 Deg C.
My suggestion would be start cooling to a user defined setpoint as soon as the 'nighttime' is triggered with sunwait, then on sunrise, warm the TEC up to ambient - this should be done slowly and consistently and I'm not sure how other astro software does this, I suspect it's taken care of by the camera driver.

I'm looking at a separate script to do this myself but haven't made much progress. I also prefer to only have anything drawing power only when it needs it, so no camera TEC use during daylight - yes a PI power consumption is insignificant in the grand scheme of things, but it's good practice nonetheless.

EricClaeys, 2022-06-23:
@Jonk2    Jon, Turning a cooler off without first gradually increasing the temperature (which Allsky doesn't do, and I likely won't implement due to the complexity of it) isn't normally a good thing to do.  I think it would be better to have separate daytime and nighttime target temperatures.  You could set the daytime target higher than the nighttime.

Would you please set **Debug Level** to 4, then restart Allsky and attach `/var/log/allsky.log`?  You can then reset the **Debug Level** to its prior number.  I'd like to see the camera capabilities for a cooled camera.   Thanks.

Jonk2, 2022-06-27:
Yes agreed, just switching the cooler off isn't normally a good thing to do as the temperature needs to slowly rise, but I though that was taken care of in the camers driver? Maybe ZWo do it differently.
i stand by my findings, a TEC running during the day isn't needed, and is potentially dangerous if it saturates.

Anyway, here's what you asked for:

Jun 25 20:46:22 allsky allsky.sh[27526]: **********************************************
Jun 25 20:46:22 allsky allsky.sh[27526]: *** Allsky Camera Software v0.8.3.3 |  2022 ***
Jun 25 20:46:22 allsky allsky.sh[27526]: **********************************************
Jun 25 20:46:22 allsky allsky.sh[27526]: Capture images of the sky with a Raspberry Pi and an ASI Camera
Jun 25 20:46:22 allsky allsky.sh[27526]: Add -h or --help for available options
Jun 25 20:46:22 allsky allsky.sh[27526]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Jun 25 20:46:22 allsky allsky.sh[27526]: Contributors:
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Knut Olav Klo
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Daniel Johnsen
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Yang and Sam from ZWO
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Robert Wagner
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Michael J. Kidd - <linuxkidd@gmail.com>
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Chris Kuethe
Jun 25 20:46:22 allsky allsky.sh[27526]:  -Eric Claeys
Jun 25 20:46:22 allsky allsky.sh[27526]: ZWO ASI178MC-Cool Information:
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Native Resolution: 3096x2080
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Pixel Size: 2.4microns
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Supported Bins: 1 2 3 4
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Color Camera: bayer pattern: RG
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Camera with cooling capabilities
Jun 25 20:46:22 allsky allsky.sh[27526]: Camera does not support serialNumber
Jun 25 20:46:22 allsky allsky.sh[27526]: - Sensor temperature: 20.70
Jun 25 20:46:22 allsky allsky.sh[27526]:   - Current camera mode: NORMAL (0)
Jun 25 20:46:23 allsky allsky.sh[27526]: Control Caps:
Jun 25 20:46:23 allsky allsky.sh[27526]: - Gain:
Jun 25 20:46:23 allsky allsky.sh[27526]:    - Description = Gain
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MinValue = 0
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MaxValue = 510
Jun 25 20:46:23 allsky allsky.sh[27526]:    - DefaultValue = 210
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsAutoSupported = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsWritable = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - ControlType = 0
Jun 25 20:46:23 allsky allsky.sh[27526]: - Exposure:
Jun 25 20:46:23 allsky allsky.sh[27526]:    - Description = Exposure Time(us)
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MinValue = 32
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MaxValue = 2,000,000,000
Jun 25 20:46:23 allsky allsky.sh[27526]:    - DefaultValue = 10,000
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsAutoSupported = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsWritable = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - ControlType = 1
Jun 25 20:46:23 allsky allsky.sh[27526]: - WB_R:
Jun 25 20:46:23 allsky allsky.sh[27526]:    - Description = White balance: Red component
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MinValue = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MaxValue = 99
Jun 25 20:46:23 allsky allsky.sh[27526]:    - DefaultValue = 70
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsAutoSupported = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsWritable = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - ControlType = 3
Jun 25 20:46:23 allsky allsky.sh[27526]: - WB_B:
Jun 25 20:46:23 allsky allsky.sh[27526]:    - Description = White balance: Blue component
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MinValue = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MaxValue = 99
Jun 25 20:46:23 allsky allsky.sh[27526]:    - DefaultValue = 90
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsAutoSupported = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsWritable = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - ControlType = 4
Jun 25 20:46:23 allsky allsky.sh[27526]: - Offset:
Jun 25 20:46:23 allsky allsky.sh[27526]:    - Description = offset
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MinValue = 0
Jun 25 20:46:23 allsky allsky.sh[27526]:    - MaxValue = 600
Jun 25 20:46:23 allsky allsky.sh[27526]:    - DefaultValue = 10
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsAutoSupported = 0
Jun 25 20:46:23 allsky allsky.sh[27526]:    - IsWritable = 1
Jun 25 20:46:23 allsky allsky.sh[27526]:    - ControlType = 5
Jun 25 20:46:23 allsky allsky.sh[27526]: - BandWidth:
[... 138 lines cut ...]
Jun 25 20:46:23 allsky allsky.sh[27526]:  Filename: image.jpg
Jun 25 20:46:23 allsky allsky.sh[27526]:  Filename Save Directory: /home/pi/allsky/tmp
Jun 25 20:46:23 allsky allsky.sh[27526]:  Latitude: 50.9N, Longitude: 1.3W
Jun 25 20:46:23 allsky allsky.sh[27526]:  Sun Elevation: 0
Jun 25 20:46:23 allsky allsky.sh[27526]:  Locale: en_GB.UTF-8
Jun 25 20:46:23 allsky allsky.sh[27526]:  Notification Images: Yes
Jun 25 20:46:23 allsky allsky.sh[27526]:  Histogram Box: 900 900 52 48, center: 1609x998, upper left: 1159x548, lower right: 2059x1448
Jun 25 20:46:23 allsky allsky.sh[27526]:  Show Histogram Box: No
Jun 25 20:46:23 allsky allsky.sh[27526]:  Show Histogram Mean: No
Jun 25 20:46:23 allsky allsky.sh[27526]:  Show Time: Yes (format: %A, %d/%m/%Y %H:%M:%S)
[... cut ...]
Jun 25 20:46:23 allsky allsky.sh[27526]:  ZWO SDK version 1, 20
Jun 25 20:46:23 allsky allsky.sh[27526]: NOTICE: Camera does not support ControlCap # 2; not setting to 50.
Jun 25 20:46:23 allsky allsky.sh[27526]: Will adjust gain at transitions
Jun 25 20:46:23 allsky allsky.sh[27526]: Extra Text File Age Disabled So Displaying Anyway
Jun 25 20:46:23 allsky allsky.sh[27526]: *** Stop the allsky service to end this process. ***
Jun 25 20:46:23 allsky allsky.sh[27526]: xxx resetGainTransitionVariables(0, 220) called at DAY

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Separate "cooler on" settings for daytime and nighttime #818**

closed 2022-06-29 by EricClaeys