AllskyTeam/allsky #820: White image from the start of the day

Amenophis54, 2021-11-24:
Hello,

I am currently having a problem because at the end of the night / start of the day I get a saturated image.

The exposure time is blocked on 15s whereas normally I am in automatic exposure.
The gain is 0 and the histogram is 254.

To solve the problem, I go back to the camera settings tab. I don't make any changes, I just click on the "Save change" button

Thank you in advance for your help.

EricClaeys, 2021-11-24:
@Amenophis54, I'll need a copy of your log to be able to help.  See [this page](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues) for instructions.

Amenophis54, 2021-11-24:
Thanks for your feedback. Here is the .log file 
https://we.tl/t-R9iT1jbiAU

EricClaeys, 2021-11-24:
@Amenophis54, please make sure you follow the instructions [here](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues).  Your Debug Level isn't 4 so there's no useful information in the log file.  Also, please attach the log file to your reply instead of using a 3rd party service.

A couple other things to check:
* Make sure you have the latest capture.cpp file from GitHub.  It should be version v0.8.1b.
* In the WebUI make sure your "Locale" is set correctly.  That should get rid of the errors on line 11 in darkCapture.sh.

Thanks - Eric

ptson92, 2021-12-07:
I have the same problem as you. If you update to the new version, it will fix it

EricClaeys, 2021-12-07:
Closing issue. Possible resolution given, plus no user response in 2 weeks.

closed 2021-12-07 by EricClaeys