AllskyTeam/allsky #821: Camera Image blank

Shand359, 2021-11-24:
I am once again struggling with the setup....

I updated the software to the newest version but when i reboot the RPi the liveview screen is totally blank...

EricClaeys, 2021-11-24:
@Shand359, can you attach (don't copy/paste) your config/config.sh and config/ftp-settings.sh files?  You may need to append ".txt" to their names.
Make sure to blank out any sensitive data in ftp-settings.sh.

You can also try pressing F12 in the web browser, then click on the "Network" link and refresh the browser window.  It should show what the name of the file is that it's trying to display.

Shand359, 2021-11-24:
Thank you for your quick reply..... this is so annoying as i had it working with a previous version.... both files you asked about have been attached...

[ftp-settings.sh.txt](https://github.com/thomasjacquin/allsky/files/7595216/ftp-settings.sh.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/7595217/config.sh.txt)

Shand359, 2021-11-24:
ok, from what i can see and figure out, it is trying and failing to load the liveview-image.jpg file....

Shand359, 2021-11-24:
Nov 24 16:29:41 raspberrypi allsky.sh[688]:      ***** Starting AllSky *****
Nov 24 16:29:42 raspberrypi allsky.sh[688]: RPiHQ Camera not found. Exiting.
Nov 24 16:34:05 raspberrypi allsky.sh[746]:      ***** Starting AllSky *****
Nov 24 16:34:05 raspberrypi allsky.sh[746]: RPiHQ Camera not found. Exiting.
Nov 24 16:37:50 allsky allsky.sh[781]:      ***** Starting AllSky *****
Nov 24 16:37:51 allsky allsky.sh[781]: RPiHQ Camera not found. Exiting.
Nov 24 16:39:55 allsky allsky.sh[1533]:      ***** Starting AllSky *****
Nov 24 16:39:55 allsky allsky.sh[1533]: RPiHQ Camera not found. Exiting.
Nov 24 16:56:17 allsky allsky.sh[725]:      ***** Starting AllSky *****
Nov 24 16:56:17 allsky allsky.sh[725]: RPiHQ Camera not found. Exiting.

I suppose that explains why there is no image being produced...

EricClaeys, 2021-11-24:
@Shand359 That'll do it.
What do you get when you run `vcgencmd get_camera`  and `grep -i "VERSION_CODENAME=bullseye" /etc/os-release` ?

EricClaeys, 2021-11-24:
@Shand359, Did you keep restarting the service, or did it do that automatically?
After printing the "not found" message, it's supposed to stop the service.

Shand359, 2021-11-24:
it was probably me resetting and restarting the rpi

EricClaeys, 2021-11-25:
@Shand359 
What do you get when you run `vcgencmd get_camera`  and `grep -i "VERSION_CODENAME=bullseye" /etc/os-release` ?

Shand359, 2021-11-25:
pi@allsky:~ $ vcgencmd get_camera
supported=1 detected=0
pi@allsky:~ $ grep -i "VERSION_CODENAME=bullseye" /etc/os-release

samuelvoltarius, 2021-11-25:
Look vor 2020-02-13-raspbian-buster-full  
on this it will run.

Shand359, 2021-11-25:
> Look vor 2020-02-13-raspbian-buster-full on this it will run.

I found a copy of buster and got it running like before on it. Thank you for that....

Shand359, 2021-11-25:
I had to go back to an older RPi3 as the 4 kept crashing or freezing.... One day things will just work as intended.... lol

EricClaeys, 2021-11-26:
@Shand359 A possible hardware or power issue?  Were you running Allsky when the Pi 4 crashed and froze?

If your original problem has been resolved would you please close the iseue?

Pedritoprint, 2021-11-27:
I have the same Problem with my "**Raspberry Pi 4**", newest OS since days. After complete installation and configuration of OS with Windows "Raspberry Pi Imager" and the "allsky"-SW with your instructions here, the Liveview-Screen is also totally blank; only a small Icon in the top-left corner. I totally formatted the SD-Card 3-times without success, always same issue.

_**vcgencmd get_camera**_ --> supported=0 detected=0
_**grep -i "VERSION_CODENAME=bullseye" /etc/os-release**_- -> VERSION_CODENAME=bullseye
_**./allsky.sh**_
     ***** Starting AllSky *****
RPiHQ Camera not found. Exiting.

At first, i thought that my camera is defective, but after reading above discussion, i think, this is not a camera problem. isn'it?
greetings, Peter from CH

Shand359, 2021-11-27:
almost identical to my issue after a fresh clean install... I dont know enough about Rasbian/Linux or RPi software to play around with settings.... I can only follow the instructions that other people have typed....

EricClaeys, 2021-11-27:
@Shand359, @Pedritoprint, the command we use to determine if a RPiHQ camera is attached works differently in Bullseye so unfortunately the Allsky software thinks there is no camera when there actually is one.  So the software stops and you don't get any pictures.

We will fix, but in the meantime I posted a workaround in another discussion.   In allsky/allsky.sk there is a line that uses "vcgencmd get_camera" to determine if a camera exists.  If it does it sets a variable (called something like RPiHQisPresent??) to 1, else it sets it to 0.  Simply change the 0 to 1 and restart the service.

Look in /var/log/allsky.log for any error messages.
And let me know how it goes.

Shand359, 2021-11-27:
Nov 27 23:35:52 allsky allsky.sh[28556]: [0:28:22.924995929] [28558] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0901 but control is not exposed by device /dev/video0
Nov 27 23:35:52 allsky allsky.sh[28556]: [0:28:22.926533469] [28558] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x00980911 but control is not exposed by device /dev/video0
Nov 27 23:35:52 allsky allsky.sh[28556]: [0:28:22.926886169] [28558] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0903 but control is not exposed by device /dev/video0
Nov 27 23:35:52 allsky allsky.sh[28556]: Stream configuration adjusted
Nov 27 23:35:52 allsky allsky.sh[28556]: [0:28:22.930974461] [28558] FATAL Serializer control_serializer.cpp:598 A list of V4L2 controls requires an ControlInfoMap
Nov 27 23:35:52 allsky allsky.sh[28556]: Backtrace:
Nov 27 23:35:52 allsky allsky.sh[28555]: Aborted
Nov 27 23:35:52 allsky allsky.sh[19202]:  >>> Unable to take picture, return code=134
Nov 27 23:35:52 allsky allsky.sh[19202]:   > Sleeping from failed exposure: 0.0 seconds
Nov 27 23:35:52 allsky allsky.sh[19202]: STARTING EXPOSURE at: 2021-11-27 23:35:52   @ 1.0 sec

Pedritoprint, 2021-11-28:
> @Shand359, @Pedritoprint, the command we use to determine if a RPiHQ camera is attached works differently in Bullseye so unfortunately the Allsky software thinks there is no camera when there [...]

Thanks a lot, here my logfile after changing value to "1":
Nov 28 10:42:21 PedroSky allsky.sh[941]:      ***** Starting AllSky *****
Nov 28 10:42:22 PedroSky allsky.sh[941]: CAMERA: RPiHQ
Nov 28 10:42:22 PedroSky allsky.sh[941]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Nov 28 10:42:22 PedroSky allsky.sh[941]: Starting allsky camera...
Nov 28 10:46:46 PedroSky allsky.sh[941]: ***
Nov 28 10:46:46 PedroSky allsky.sh[941]: Sorry, AllSky with RPiHQ cameras on the Bullseye operating system does not yet work.
Nov 28 10:46:46 PedroSky allsky.sh[941]: See https://github.com/thomasjacquin/allsky/discussions/802 for more information.
Nov 28 10:46:46 PedroSky allsky.sh[941]: ***
BTW, now in "Liveview" i could see a Picture with the hint to see in "var/log/allsky.log" about the "Error"

Pedritoprint, 2021-11-28:
> @Shand359, @Pedritoprint, the command we use to determine if a RPiHQ camera is attached works differently in Bullseye so unfortunately the Allsky software thinks there is no camera when there [...]

Shand359, 2021-11-28:
I re-opened this simply to state that the allsky camera with the RPiHQ camera is now fully operational, running Bullseye OS on an RPi4.,....

Thank you to everyone who gave ideas and pointers...

closed 2021-11-28 by Shand359