AllskyTeam/allsky #828: "nightmaxexposure" value ignored

Apennine2, 2021-11-27:
Hello,

I have a new issue where no matter what value I enter for "nightmaxexposure" (30s), the camera ramps up to a greater value (~60s).

I recently after changing multiple settings and configs in allsky reverted to the stock 0.8.1 version via a clean install, except with the update capture.sh v0.8.1g passed along by @EricClaeys .

Now, no matter what I set "nightmaxexposure" to, the camera goes rogue with the exposure time. Any advice on why this is happening would be great. Log attached, which shows 30s max night exposure, and the camera going quickly to 60s.

Thank you,

Chris
[allsky.log](https://github.com/thomasjacquin/allsky/files/7611122/allsky.log)

Apennine2, 2021-12-08:
Hello,

I have updated to 0.8.2c and it has fixed the issue, thank you!

-Chris

EricClaeys, 2021-12-09:
Issue resolved with newest release.

closed 2021-12-09 by EricClaeys