AllskyTeam/allsky #840: buster / RPiHQ / raspistill: Invalid command line option (--info-text)

AndreasLMeg, 2021-12-04:
My raspistill version doesn't support any option (--info-text). This is probably a bug that came in through the libcamera/overlay text conversion.

```
Dec  4 13:22:59 allsky allsky.sh[929]: STARTING EXPOSURE at: 2021-12-04 13:22:59   @ 32 us
Dec  4 13:22:59 allsky allsky.sh[929]:   > Capture command: raspistill --output 'image.jpg' --thumb none --burst -st --timeout 1000 --nopreview --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 2.5,1.7 --hflip --vflip --saturation 0 --brightness 50 --quality 100 --info-text " -a 1036 -a " RPiHQ Test"" -ae 64,0xff,0x808000
Dec  4 13:22:59 allsky allsky.sh[929]: Invalid command line option (--info-text)
Dec  4 13:22:59 allsky allsky.sh[929]:  >>> Unable to take picture, return code=64
```

For me works this source code change
```
                        if (! libcamera)
-                               command += " --info-text \"" + info_text + "\"";
+                               command += " " + info_text;

```
[image]

Pedritoprint, 2021-12-04:
and in my case, after update to newest version from yesterday (2021-12-03) following error in "allsky.log":
 Capture command: LIBCAMERA_LOG_LEVELS='ERROR,FATAL' libcamera-still --output 'image.jpg'

AndreasLMeg, 2021-12-04:
@Pedritoprint  can you send me the entire command line AND the following 5 lines.
Then I can verify if that is the same error.

btw: Schöne Grüße in die Schweiz

AndreasLMeg, 2021-12-04:
works for me.
after:  capture_RPiHQ.cpp: fix info_text #843

closed 2021-12-04 by AndreasLMeg

Pedritoprint, 2021-12-05:
After completly new istallation, i have no Errors in log. Sorry couldn't reproduce what's happened.
Grüsse nach Österrreich, Peter