AllskyTeam/allsky #848: My allsky is overexposed during the day

ptson92, 2021-12-07:
Hello
My allsky during the day it will have a few normally exposed photos, then one will be overexposed.
I checked the allsky log and it got the following error:
Dec  6 11:14:42 allsky allsky.sh[11232]: WARNING: ASISetControlCaps() for control 1, value=18605 failed: ASI_ERROR_GENERAL_ERROR (16)
Dec  6 11:14:47 allsky allsky.sh[11232]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #1 (with 0.8 exposure = NO) (11)
Dec  6 11:14:59 allsky allsky.sh[11232]: STARTING EXPOSURE at: 2021-12-06 11:14:59   @ 18.61 ms (0.02 sec)
I don't get this error at night. It only happens during the day.
Sorry my english is not good.

ptson92, 2021-12-07:
Dec  6 11:12:25 allsky allsky.sh[10884]: STARTING EXPOSURE at: 2021-12-06 11:12:25   @ 120 us
Dec  6 11:12:25 allsky allsky.sh[10884]: === next exposure: old way=215, new way=206 (multiply by 1.721) [last_exposure_us=120, minAcceptable=122, mean=78]
Dec  6 11:12:26 allsky allsky.sh[10884]: === next exposure: old way=370, new way=352 (multiply by 1.721) [last_exposure_us=205, minAcceptable=122, mean=78]
Dec  6 11:12:28 allsky allsky.sh[10884]: === next exposure: old way=245, new way=197 (multiply by 0.564) [last_exposure_us=350, maxAcceptable=134, mean=216]
Dec  6 11:12:28 allsky allsky.sh[10884]:  >xxx mean was 78 and went from 44 below min of 122 to 82 above max of 134, is now at 216; OK to set temp max to current_exposure_us of 350
Dec  6 11:12:29 allsky allsky.sh[10884]: === next exposure: old way=370, new way=331 (multiply by 1.617) [last_exposure_us=205, minAcceptable=122, mean=83]
Dec  6 11:12:29 allsky allsky.sh[10884]:  >xxx mean was 216 and went from 82 above max of 134 to 39 below min of 122, is now at 83; should NOT set temp min to current_exposure_us of 205
Dec  6 11:12:30 allsky allsky.sh[10884]: === next exposure: old way=595, new way=533 (multiply by 1.617) [last_exposure_us=330, minAcceptable=122, mean=83]
Dec  6 11:12:30 allsky allsky.sh[10884]: WARNING: ASISetControlCaps() for control 1, value=350 failed: ASI_ERROR_GENERAL_ERROR (16)
Dec  6 11:12:35 allsky allsky.sh[10884]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #1 (with 0.8 exposure = NO) (11)
Dec  6 11:12:47 allsky allsky.sh[10884]: STARTING EXPOSURE at: 2021-12-06 11:12:47   @ 350 us
Dec  6 11:12:52 allsky allsky.sh[10884]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #2 (with 0.8 exposure = NO) (11)
Dec  6 11:12:54 allsky allsky.sh[10884]: STARTING EXPOSURE at: 2021-12-06 11:12:54   @ 350 us
Dec  6 11:12:59 allsky allsky.sh[10884]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #3 (with 0.8 exposure = NO) (11)
Dec  6 11:13:01 allsky allsky.sh[10884]: STARTING EXPOSURE at: 2021-12-06 11:13:01   @ 350 us
Dec  6 11:13:06 allsky allsky.sh[10884]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #4 (with 0.8 exposure = NO) (11)
Dec  6 11:13:08 allsky allsky.sh[10884]: STARTING EXPOSURE at: 2021-12-06 11:13:08   @ 350 us
Dec  6 11:13:13 allsky allsky.sh[10884]:   > ERROR: Failed getting image: ASI_ERROR_TIMEOUT #5 (with 0.8 exposure = NO) (11)
Dec  6 11:13:15 allsky allsky.sh[10884]: *** ERROR: Maximum number of consecutive errors of 5 reached; exiting...
Dec  6 11:13:16 allsky allsky.sh[10884]: copy_notification_image.sh: Uploading NotRunning.jpg
Dec  6 11:13:19 allsky allsky.sh[10884]:      ***** Stopping AllSky *****
Dec  6 11:13:19 allsky allsky.sh[10884]: 'capture' exited with RETCODE=99
Dec  6 11:13:19 allsky allsky.sh[10884]: copy_notification_image.sh: Uploading NotRunning.jpg
Dec  6 11:13:25 allsky allsky.sh[11057]:      ***** Starting AllSky *****
Dec  6 11:13:25 allsky allsky.sh[11057]: CAMERA: ZWO
Dec  6 11:13:25 allsky allsky.sh[11057]: Starting allsky camera...
Dec  6 11:13:25 allsky allsky.sh[11057]: copy_notification_image.sh: Uploading StartingUp.jpg
Dec  6 11:13:34 allsky allsky.sh[11057]: **********************************************
Dec  6 11:13:34 allsky allsky.sh[11057]: *** Allsky Camera Software v0.8.2c |  2021 ***
Dec  6 11:13:34 allsky allsky.sh[11057]: **********************************************
Dec  6 11:13:34 allsky allsky.sh[11057]: Capture images of the sky with a Raspberry Pi and an ASI Camera

EricClaeys, 2021-12-07:
@ptson92, please make sure you havr the lasted files from GitHub.  We just updated several a few days ago.
Also see the Wiki page for reporting Issues and follow those instructions.

Your English is great.  I would not have known it wasn’t your native language if you didn't mention it.

Thanks

ptson92, 2021-12-07:
Thank you very much.
I just updated yesterday.
Before that, I had a problem when switching from night to day the exposure time was locked at 15s.

Bra1nsen, 2021-12-11:
is there a command to check the allsky version?

EricClaeys, 2021-12-11:
@Bra1nsen look in /var/log/allsky.log for the word "Software".

Bra1nsen, 2021-12-11:
thanks Eric.

maybe not the right place, but youre definitely the right person to ask:

in addition to the skycam, i also connected a couple of weather sensors (temp, wind, radiation) to my raspi. I wrote a simple python script in which I send the data to my server via mqtt. apparently this interferes with the allsky software, since both apparently don't run at the same time. can that be so? And if so, what would you advise me to do?

linuxkidd, 2021-12-11:
There's really no reason your sensor python and the allsky code shouldn't coexist.  The allsky `capture` app doesn't use any ports itself, and the WebUI is on port 80 ( via lighttpd, could be changed in `/etc/lighttpd/lighttpd.conf` if desired ).

The only potential issue would be resource exhaustion... memory or processor.. but as long as your sensor python code isn't doing anything crazy.. you should be more than fine.  You could have both running, and then use `top` to monitor the CPU/Memory just in case.  You can also use `iostat` (part of the `sysstat` package) to check for disk IO issues -- but I highly doubt there will be any.

fwiw, you could modify your Python code to dump the sensor data to a text file, then add it as a text overlay for capture.  Then you'd have your sensor data under the date/time, sensor temp, gain, etc... at the top left of your allsky image.

Bra1nsen, 2021-12-11:
ok i will check that. in any case, after activating my python script, the allsky service no longer worked. but your answer calms me down. Thank you Sir

EricClaeys, 2021-12-12:
@Bra1nsen can you be more specific when you say "the allsky service no longer worked"?
Did you get any messages in /var/log/allsky.log?

Bra1nsen, 2021-12-12:
Yep, something is wrong. and it is definitely due to the prallel running python script.
the log show immediately after I run my script:  "***** Stopping AllSky *****"

Bra1nsen, 2021-12-12:
I mean I could share the script, but it's an absolutely basic program that publishes simple strings

linuxkidd, 2021-12-12:
@Bra1nsen let's create a '[new issue](https://github.com/thomasjacquin/allsky/issues/new/choose)' for this... then, please do share the name of your script, as well as the code, and the `/var/log/allsky.log` file.

Thanks!

EricClaeys, 2021-12-12:
@ptson92 Any update?

ptson92, 2021-12-12:
I have updated to the latest version. It still has the same error.  I tried turning the 0.8 exposure on and off, but still the same error. Now that I'm using manual exposure, I don't get the error anymore.

EricClaeys, 2021-12-12:
@ptson92 Please see the Wiki page for reporting Issues and follow those instructions.  I will take a look at your log file.

ptson92, 2021-12-12:
Thanks
This is my log file.
[allsky.log](https://github.com/thomasjacquin/allsky/files/7698394/allsky.log)

EricClaeys, 2021-12-15:
@ptson92 Try setting your USB Bandwith to 40 with auto exposure on and see if it helps.  If not, try 60 and 100.
Also see if your Pi is having undervoltage issues.  In the WebUI, click on the "System" link.

ptson92, 2021-12-17:
Thank you @EricClaeys 
I'm going to change the power supply to the Pi this weekend and try again.

EricClaeys, 2021-12-18:
@ptson92 Changing the USB bandwidth is super easy in the WebUI and is a g[d debugging step.

Bra1nsen, 2021-12-22:
Eric i did the update as suggested, but now iam not able to set exposure time during the day? What can i do?

Bra1nsen, 2021-12-22:
[image]
 
would it work if i adjust here?

Bra1nsen, 2021-12-22:
/etc/raspap/camera_options_RPiHQ.json

EricClaeys, 2021-12-23:
@Bra1nsen In the WebUI's "Daytime settings" section there should be an "Exposure" setting.  It is an "advanced" option since people usually have auto-exposure on during the day.  Go to the end of the page and click on the "Show advanced options" button.
Do NOT edit the "camera_options_RPiHQ.json" file manually, since the WebUI overwrites it.  That file is meant to be updated only via the WebUI.

EricClaeys, 2021-12-23:
@ptson92 Did changing the power supply and/or USB bandwidth help?

You log file showed that you got a couple pictures that were fairly dark, then a super bright one, then a couple fairly dark, then the ASI_ERROR_TIMEOUTs.  There wasn't anything with a 15-second lock.

Since the first picture was dark, the software increased the exposure time but the second picture was just as dark so the software increased the exposure time again, and the 3rd picture jumped significantly in brightness.
It's like there's stiction in the camera exposure - the exposure time keeps increasing but the brightness doesn't change, then all of a sudden another change in exposure time causes a big change in brightness.  I don't know why that is.

ptson92, 2021-12-24:
@EricClaeys 

I have tried changing the power supply, trying to change the usb bandwidth. Still nothing changed, I still get the same error. That problem only shows up during the day. Everything is fine at night.
I tried it on ASI 120MC and it doesn't have the same problem. Maybe the problem is because the 178MC is faulty or the firmware is not good.

Thanks

EricClaeys, 2021-12-24:
@ptson92 You are only getting the ASI_ERROR_TIMEOUT errors during the day?
Please follow the instructions [here](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues) and upload your `/var/log
As a test, set your daytime exposure to 10000 (same as night), your day brightness to 50 (which is the default), and night max auto-exposure to 60000 (which is the max amount the camera supports). and run during the day.
If that still fails, set the "Version 0.8 Exposure" option to "Yes" and try again.

EricClaeys, 2022-05-17:
Closing issue - no response from user in 5 months.

closed 2022-05-17 by EricClaeys