AllskyTeam/allsky #854: Allsky stops while running mqtt python script

Bra1nsen, 2021-12-12:
weather_mqtt.py
---------------------------------
#!/usr/bin/env python

imports (..)

def on_message(client, userdata, message):
	msg = str(message.payload.decode("utf-8"))
	#print("message received: ", msg)
	#print("message topic: ", message.topic)

def on_connect(client, userdata, flags, rc):
	client.subscribe('skycam/picam2021/metrics')

# Create sensor object, default I2C bus
bus = SMBus(1)
bme280 = BME280(i2c_dev=bus)
i2c = board.I2C()   
ltr = adafruit_ltr390.LTR390(i2c)
si1145 = SI1145.SI1145()
uv1 = adafruit_veml6070.VEML6070(i2c)

# open the file in the write mode
with open('/home/pi/weather/weather.csv', 'a', newline="") as log:

	writer = csv.writer(log)
	#writer.writerow(["Satellit", "", "BME280", "", "", "LTR390", "", "", "", "SI1145", "", "", "VEML6070"])
	#writer.writerow(["Datum", "Zeit", "Temperatur", "Druck", "rel_Luftfeuchtigkeit", "uv", "amblight", "UVI", "lux", "vis", "IR", "uvIndex", "uv_raw", "risk_level"])

	while True:
	
		today = date.today()
		now = datetime.datetime.now().time()
		now_format = now.strftime("%H:%M:%S")
		unix_timestamp = time.time()
		#BME280	
		temperature = round(bme280.get_temperature(),1)
		pressure = round(bme280.get_pressure(),1)
		humidity = round(bme280.get_humidity(),1)
		#LTR390
		uv = round(ltr.uvs,2)
		amblight = round(ltr.light,2)
		UVI = round(ltr.uvi,2)
		lux = round(ltr.lux,2)
		#SI1145
		vis = round(si1145.readVisible(),2)
		IR = round(si1145.readIR(),2)
		UV2 = round(si1145.readUV(),2)
		uvIndex = UV2 / 100.0
		#VEML6070
		uv_raw = round(uv1.uv_raw,2)
		risk_level = uv1.get_index(uv_raw)

		# Append data to the spreadsheet
		row = (today, now_format, temperature, pressure, humidity, uv, amblight, UVI, lux, vis, IR, uvIndex, uv_raw, risk_level) 
	
		# write a row to the csv file
		writer.writerow(row)
			
		print(f"Datum: {today}, Zeit: {now_format}, Temperatur:{temperature}°C, Druck:{pressure}hPa, rel_Luftfeuchtigkeit:{humidity}%, uv_LTR={uv}, lux={lux}, uv_VEML={uv_raw}")
		
		print("saved in weather.csv")
		
		BROKER_ADDRESS = ""
		PORT = 59444
		TOPIC = 'skycam/picam2021/metrics'
		USER = "skycam"
		PW = "unknown"
		
		jsonO1_Temp = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "°C", "source": "", "minInterval": "0", "status": "0"}
		jsonO2_Druck = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "hPa", "source": "", "minInterval": "0", "status": "0"}
		jsonO3_relLF = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "%", "source": "", "minInterval": "0", "status": "0"}
		jsonO4_uv = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO5_amblight = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO6_UVI = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "%", "source": "", "minInterval": "0", "status": "0"}
		jsonO7_lux = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "lux", "source": "", "minInterval": "0", "status": "0"}
		jsonO8_vis = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO9_IR = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO10_uvIndex = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO11_uvraw = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "count", "source": "", "minInterval": "0", "status": "0"}
		jsonO12_risklevel = { "timestamp": unix_timestamp, "urn": "SkyCam_BME280_Temp", "value": temperature, "unit": "", "source": "", "minInterval": "0", "status": "0"}
		json_list_sensors = [jsonO1_Temp, jsonO2_Druck, jsonO3_relLF, jsonO4_uv, jsonO5_amblight, jsonO6_UVI, jsonO7_lux, jsonO8_vis, jsonO9_IR, jsonO10_uvIndex, jsonO11_uvraw, jsonO12_risklevel]
		jsonSUMsensor = json.dumps(json_list_sensors) 
		
		mqtt_file = { "source": "SkyCam_21F19", "psk": "", "content": jsonSUMsensor, "type": "json", "sendTime": "0"}
		
		jsonFile = json.dumps(mqtt_file)
		
		rowStr = (str(temperature), str(pressure), str(humidity), str(uv), str(amblight), str(UVI), str(lux), str(vis), str(IR), str(uvIndex), str(uv_raw), risk_level)
		
		jsonStr = json.dumps(rowStr)
		
		client = mqtt.Client()
		client.username_pw_set(USER, PW)
		client.on_connect = on_connect
		client.on_message = on_message
		client.connect(BROKER_ADDRESS, PORT)
		client.loop_start()
		
		
		client.publish(TOPIC, jsonFile, 1)
		#msgs = [{'topic': "skycam/picam2021/metrics", 'payload': "multiple 1"}, ("skycam/picam2021/metrics", str(temperature), 1), ("skycam/picam2021/metrics", str(pressure), 1)]
		#publish.multiple(msgs, hostname="skycam/picam2021/metrics"

		print("Connected to MQTT Broker: " + BROKER_ADD
			
		# Wait for 30 seconds
		time.sleep(30)
--------------------------------------
[image]

Dec 12 00:58:14 allsky allsky.sh[3033]: Capturing & saving image...
Dec 12 00:58:22 allsky allsky.sh[3033]: Capturing & saving image done, now wait 30000 seconds...
Dec 12 02:04:59 allsky allsky.sh[3033]:      ***** Stopping AllSky *****

--------------------------------------------

Maybe the problem is not in the script at all. I remember reading something about the allsky service while installing mosquietto ..

you guys are awesome that you really make the effort to help me. that is insane

Bra1nsen, 2021-12-12:
[image]

Bra1nsen, 2021-12-12:
[image]

okay i guess heres the problem, but how can i fix that?

linuxkidd, 2021-12-12:
The mosquitto bit seems likely to be a misconfig in `/etc/mosquitto/mosquitto.conf`, but the logs don't give enough detail on that.  It's definitely unrelated to Allsky exiting.

And, the script definitely has nothing to do with Allsky exiting.

From the log line `now wait 30000 seconds`, it seems a night delay is set to 30000 seconds... instead of 30 seconds.  Please check in the webUI (assuming you have it installed) for this parameter and adjust it as you see fit.

EricClaeys, 2021-12-12:
@Bra1nsen please update Allsky.  The issue with waiting 1000 times longer than requested was fixed a long time ago.  Look in the Wiki for tips on how to upgrade.

EricClaeys, 2021-12-12:
@Bra1nsen thanks for the compliment. We enjoy using and improving AllSky and helping others enjoy it.

Bra1nsen, 2021-12-12:
Ok, I'll do the update, hoping that I won't break everything. My SkyCam is designed for the day, as I photograph clouds (solar industry), so I intentionally set the delay so high for the night.

There must be a connection between my py script and allsky, otherwise the allsky service wouldn't stop.

Bra1nsen, 2021-12-12:
GNU nano 3.2                                           mosquitto.conf                                                       

# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example

pid_file /var/run/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log

include_dir /etc/mosquitto/conf.d

Bra1nsen, 2021-12-12:
to autostart the script i use /etc/rc.local - could there be a connection?

Bra1nsen, 2021-12-12:
this is crazy!

linuxkidd, 2021-12-13:
The mosquitto MQTT broker should stay running unless a config parameter isn't set properly.  I would expect if you look at `journalctl -u mosquitto`, there may be more information about the crash there.

If what you posted above is the exact contents of your config file, I'd make sure that all the stuff before `pid_file ... ` is commented ( preceded by a # sign ).. otherwise, it wouldn't be valid and could certainly cause a crash.

linuxkidd, 2021-12-13:
Here's the stock mosquitto config:
```
# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example

pid_file /run/mosquitto/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log

include_dir /etc/mosquitto/conf.d
```

linuxkidd, 2021-12-14:
Hi @Bra1nsen, any luck with this?

Bra1nsen, 2021-12-22:
the problem only occurs when I run the script automatically when booting via /etc/rc.local.

If I boot the Pi and start afterwards manually the script on the console, everything works.

In any case, I made the update as recommended. However, I can no longer set the exposure time for the day ?! That would be catastrophic because I need to adjust for imaging clouds. Should I trick the camera into believing it was night?

Bra1nsen, 2021-12-22:
i just discovered the allsky map, if one like - add my cam to the map

EricClaeys, 2021-12-24:
@Bra1nsen In the WebUI, on the "Camera settings" page, go to the bottom and click the "Show advanced options" button.
In the **Daytime settings** section at the top of the page is the **Manual Exposure** field to set the daytime exposure.  If daytime auto-exposure is turned on, this is the starting exposure and the camera will the adjust the exposure based on the daytime brightness.  If daytime auto-exposure is off, all daytime exposures will be **Manual Exposure** in length.

Bra1nsen, 2021-12-28:
Great Eric - Thank You! Is there also an option to generate a keogramm from the day? That would be very helpful for analysis of radiation:)

EricClaeys, 2021-12-28:
@Bra1nsen yes, there is.  If you want a keogram generated every morning then look in the "keogram" section of the "config.sh" file.
If instead you want to create a keogram for a specific day, use GenerateForDay.sh-k <DA T E>.

Bra1nsen, 2021-12-28:
i only find the possibility to activate the keogram for the night pictures ...

EricClaeys, 2021-12-30:
@Bra1nsen Not sure what you mean.  When a keogram is created, it uses all the images in allsky/images/DATE.  It doesn't care if the images are day or night ones.

closed 2022-03-02 by Bra1nsen

EricClaeys, 2022-03-03:
@Bra1nsen   Did you close this because it got resolved?  If so, how?

Bra1nsen, 2022-03-04:
Sorry I couldn't identify the exact cause, but the script is currently running via cronlab and it works