AllskyTeam/allsky #856: Text file overlay issue

Jonk2, 2021-12-17:
I have a sky quality meter based on Rob Brown's open source MySQM Pro. It outputs a text file that can be displayed on the allsky image.

I asked Rob to do this for this particular allsky application and it works really well.

However, I've just added some ascii characters to the logging for the text file to be overlaid and they're not displayed on the allsky properly.

In the software on the RPI, the ascii characters are copied and pasted from Windows because I don't know how to do the equivalent ALT + xxx on linux. I'm working on the RPI via remote desktop.

They look ok on the software and the resulting text file, but they're being displayed as '??' on the RPI. The ascii characters are 'ALT + 248' = ° and 'ALT +253' = ².

These 2 screenshots show what I mean.

First, the RPI:

[image]

This is the allsky:

[image]

Not super urgent obviously, but it would be nice if this could be fixed - IF it's an Allsky issue, not sure how it could be a MySQM Pro issue.

EricClaeys, 2021-12-18:
@Jonk2 Jon, Allsky currently ony supports normal "printable" ASCII characters.  I will add this to the enhancement request list.

EricClaeys, 2021-12-18:
@Jonk2, this may not be easy to fix.  The library we use for all image manipulation (crop, add text, etc.) only supports ASCII.

Pedritoprint, 2021-12-19:
The same with "umlaut" ä,ö,ü Ä,Ö,Ü,  because my residence is Wängi, so i have to write Waengi

AndreasLMeg, 2021-12-25:
a simple test with freetext - maybe a solution....
```

						//freetext
						int fontHeight = 60;
						int thickness = -1;
						int linestyle = 8;
						cv::Point textOrg(iTextX, iTextY + (iYOffset / currentBin));

						cv::Ptr<cv::freetype::FreeType2> ft2;
						ft2 = cv::freetype::createFreeType2();
						ft2->loadFontData( "/usr/share/fonts/truetype/dejavu/DejaVuSerif.ttf", 0 );
						ft2->putText(pRgb, line, textOrg, fontHeight,
               cv::Scalar::all(255), thickness, linestyle, true );
```

[image]

EricClaeys, 2021-12-25:
@AndreasLMeg Does freetext require different #includes's?  Is it installed by default on Pi's ?
How did you know what font to load?

AndreasLMeg, 2021-12-25:
@EricClaeys 
1. ask google for Opencv unicode, ttf,,...
2. found the solution with freetext
3. ttf installed on raspi: https://blog.adafruit.com/2014/06/20/displaying-raspberry-pi-truetype-fonts-piday-raspberrypi-raspberry_pi/
4. made a test with DejaVuSerif.ttf - and it works

BUT, this was just for fun, tock the first solution, maybe better solution is possible....

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Improve Image Overlay Options**.

The next release of Allsky (after v2022.03.01) will support a GUI overlay editor with drag-and-drop text and images.  It will support additional fonts and non-ASCII characters.

closed 2022-06-29 by EricClaeys