AllskyTeam/allsky #859: Aurora forcast during day

Pedritoprint, 2021-12-19:
I'm working with Raspi 4 b "Bullseye" and newest Allsky SW,  `DAYTIME_CAPTURE="false"` in Raspi
 and then, although the parameter `auroraForecast: false,` in config.js of html-Page, during daytime, i see the Aurora forecast in case of "camera is of during daytime.

EricClaeys, 2021-12-19:
@Pedritoprint would you please look in /var/log/lighttpd/error.log to see what it contains?
If it has some entries, run `sudo truncate -s 0 /var/log/lighttpd/error.log` then refresh your browser page and see if there are any new entries in the log file.

Pedritoprint, 2021-12-20:
I found only this entry
`2021-12-19 00:00:24: server.c.1787) logfiles cycled UID = 0 PID = 27676`
and after "error-clean" and restart the HomePage, there is no entry but still the aurora forecast during the day.

EricClaeys, 2021-12-24:
@Pedritoprint In config.sh, what is your `POST_END_OF_NIGHT_DATA` set to?
During the day when it's showing the aurora forecast, what happens if you refresh your browser window?

Pedritoprint, 2021-12-24:
- config.sh: `POST_END_OF_NIGHT_DATA="true"`
- by refreshing the browser, there is always the aurora forecast visible. Also if i press SHIFT and refresh.

EricClaeys, 2021-12-25:
@Pedritoprint would you please attach your config.sh, /etc/raspap/settings_ZWO.json, and /var/www/html/allsky/config.js files?

Pedritoprint, 2021-12-25:
Here my files
[config.js.txt](https://github.com/thomasjacquin/allsky/files/7775789/config.js.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/7775791/config.sh.txt)
[settings_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/7775792/settings_ZWO.json.txt)

:

EricClaeys, 2021-12-25:
@Pedritoprint I asked for the incorrect settings file.  Would you please attach the /etc/raspap/settings_RPiHQ.json file?

In summary, the issue is that the aurora forecast shows only during the day if you started allsky at night?  Is that correct?
If you start allsky tonight, it won't show the aurora forecast until morning, and it will then show it until nighttime, when it will turn off?  And then it will come back on during at daytime?

Pedritoprint, 2021-12-26:
OK, here the correct file.
[settings_RPiHQ.json.txt](https://github.com/thomasjacquin/allsky/files/7776810/settings_RPiHQ.json.txt)
And defenitively, if i power-up the camara during daytime, the correct picture will be shown on website but as soon the camera is on during night, then after "Angel"-Value, the picture on website is korrect till sunrise (Camera is off during the day). Afterwards, on the web-UI, still correct picture but on website, the aurora forecast is visible.
regards, Peter [CH]

EricClaeys, 2021-12-27:
@Pedritoprint Thanks.  Would you please post the data.json file?  It should be in allsky/tmp.

Pedritoprint, 2021-12-27:
Here the next file.
[data.json.txt](https://github.com/thomasjacquin/allsky/files/7779302/data.json.txt)
regards Peter

EricClaeys, 2022-01-21:
@Pedritoprint   My bad for taking almost a month to reply.  Very sorry about that.

There's a newer "controller.js" file in [allsky-website](https://github.com/thomasjacquin/allsky-website).  It should improve the situation, but may not completely fix it.  I'm working on an improved version but need more testing.
Please also grab the updated "scripts/postData.sh" file in [allsky](https://github.com/thomasjacquin/allsky).  It adds the sunrise time to data.json so the website can better determine if it's daytime or nighttime.  You need both files.

If the new version does NOT fix the problem, please press F12 in the browser when it shows the aurora image, then on the right side click on "Console" (may need to first click on ">>") and let me know what the console output is.  There should be a line about reading the data.json file and some other info.

EricClaeys, 2022-02-10:
@Pedritoprint   Any update?  Did the newer controller.js file and scripts/postData.sh file help?

EricClaeys, 2022-02-25:
@Pedritoprint   Any update?

EricClaeys, 2022-06-03:
Closing Issue.  No response in over 3 months despite two attempts.

closed 2022-06-03 by EricClaeys