AllskyTeam/allsky #860: .png picture in case of .jpg and startrails/keogram

Pedritoprint, 2021-12-19:
an other issue i have; i configured the Raspi to upload .png-pictures in case of default -jpg. It works fine but the "startrail" and "keogram" does't work because the html-code is looking for .jpg pictures and couldn't find corresponding files. In config.js (HomePage) i changed the parameter `imageName: "xxxx.png` but this does't changes the geogram- & startrail call.
An other issue with startrail & keograms in my case is the fact, that if i upload .jpg pictures following Error is visible in the homepage:
[image]
how to change the Permission on my WebServer for these directories. The Timelaps is working fine without these permission-error.

EricClaeys, 2021-12-19:
@Pedritoprint there is currently no way to change the extension of the startrails or keogram files from the default of .jpg.  The "image" file is the only one whose extension can be changed.  Have you verified your srartrails and keogram files are being uploaded to the correct location on your web server, with the correct names?

For the permission error, first make sure the keogram/thumbnails directory exists.  If not, create it with permissions that allow the web server to write to it.  If you don't know how to do that, contact your web hosting provider.

Pedritoprint, 2021-12-20:
1. if i change the picture extension also the extension of keograms& startrails pictures change to .png
2. upload is correct ant to the correct directories. As you could see in my first post, a keogram-picture is visible and the link to see the keogram is ok. 
3. i'm my own web hosting provider on the Synology-NAS.

Pedritoprint, 2021-12-20:
Now, i found the "failure" in case of wrong permission on my web-server. It wasn't allowed to write to the directory from inside the HomePage. Sorry.

closed 2021-12-20 by Pedritoprint