AllskyTeam/allsky #862: Error when compiling sunwait.cpp

Hydrazine2501, 2021-12-20:
Decided to wipe my allsky cameras and had this pop up when installing:

2021-12-20 15:19:24 Building sunwait...
make[2]: Entering directory '/home/pi/allsky/src/sunwait-src'
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function ‘int main(int, char**)’:
sunwait.cpp:665:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function ‘void sun_RA_dec(double, double*, double*, double*)’:
sunriset.cpp:151:18: warning: variable ‘zs’ set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~

Not sure if that affects anything.

EricClaeys, 2021-12-21:
@Hydrazine2501 It doesn't impact anything.  Sunwait is a separate program that we don't have control over, although I submitted a request to change it to eliminate the messages.

@linuxkidd Michael, a couple have reported this.  Is it possible to do something in the Makefile to eliminate the messages?  A compiler flag or send output to dev null?

linuxkidd, 2021-12-21:
Unfortunately... if we redirect the sunwait make to /dev/null, we'd also not get any output if something was actually wrong.

iirc, you submitted a PR to fix this to the upstream 'sunwait' repo.  Maybe until that's merged, we include a patch file that's applied before make.

Thoughts?

EricClaeys, 2021-12-21:
@linuxkidd I submitted an issue (and maybe a PR, I don't remember) to sunwait, but it doesn't appear it's being maintained.  Isn't there a compiler option to tell it not to complain about unused options?

linuxkidd, 2021-12-21:
Sure, but that would require patching the sunwait make file.  I'd rather patch the cpp to fix the warning complaint, vs masking it.

EricClaeys, 2021-12-24:
@linuxkidd Fixing one of the messages is easy and what I did the PR for.  I couldn't figure out how to fix the other message so submitted an Issue for it.  If you can figure it out, I'm all for fixing the .cpp file.

AndreasLMeg, 2022-01-23:
@EricClaeys @linuxkidd 
with this hack we can suppress the warning

makefile:
```
sunwait:
	@echo `date +%F\ %R:%S` Initializing sunwait submodule...
	@git submodule init
	@git submodule update
	@echo `date +%F\ %R:%S` Building sunwait...
	@sed -i -e "s/CFLAGS=/CFLAGS=-w /" sunwait-src/makefile
	@$(MAKE) -C sunwait-src
	@cp sunwait-src/sunwait .
	@echo `date +%F\ %R:%S` Done.

```

EricClaeys, 2022-01-25:
@AndreasLMeg Are you goi g to create a PR for this?

AndreasLMeg, 2022-01-25:
No, I didn't intend to because I'm not well versed enough to assess whether this is the right and/or optimal solution.

In any case, the warning bothers me, or the discussions and issues in this regard.
If I can't influence the external component anyway, then I can ignore it.

AND this can also cause other more important warnings to go under.

AndreasLMeg, 2022-01-28:
@EricClaeys should I make a PR ?

EricClaeys, 2022-01-29:
@AndreasLMeg Yes, please.

EricClaeys, 2022-02-10:
Issue fixed, closing issue.  Grab the latest src/Makefile and src/sunwait.patch.

closed 2022-02-10 by EricClaeys