AllskyTeam/allsky #866: wrong usage timelapse.sh / TIMELAPSE_EXTRA_PARAMETERS

AndreasLMeg, 2021-12-24:
timelapse.sh has problems with defined TIMELAPSE_EXTRA_PARAMETERS

problem is in /generateForDay.sh:
CMD="'${ALLSKY_SCRIPTS}/timelapse.sh' ${DATE} ${TIMELAPSE_EXTRA_PARAMETERS}"

don't use TIMELAPSE_EXTRA_PARAMETERS as start parameter.

EricClaeys, 2021-12-25:
@AndreasLMeg what do you mean by "don't use TIMELAPSE_EXTRA_PARAMETERS as start parameter"?

AndreasLMeg, 2021-12-25:
1. https://github.com/thomasjacquin/allsky/blob/ca9708d261296d384965d9fb22284e73f48f810b/scripts/timelapse.sh#L18 -> TIMELAPSE_EXTRA_PARAMETERS now usable
2. https://github.com/thomasjacquin/allsky/blob/ca9708d261296d384965d9fb22284e73f48f810b/scripts/timelapse.sh#L29 -> more parameter will show the usage and exit the script
3. https://github.com/thomasjacquin/allsky/blob/ca9708d261296d384965d9fb22284e73f48f810b/scripts/timelapse.sh#L127 -> TIMELAPSE_EXTRA_PARAMETERS  is used and not the parameter ${3}
4. https://github.com/thomasjacquin/allsky/blob/ca9708d261296d384965d9fb22284e73f48f810b/scripts/generateForDay.sh#L207 -> so, it is not necessary but incorrect if TIMELAPSE_EXTRA_PARAMETERS is used as parameter in the CMD.

EricClaeys, 2021-12-25:
@AndreasLMeg Thanks, I understand now and just submitted a PR to fix it.  I wish all fixes were this easy.

PR #867 (2021-12-25): generateForDay.sh: bug in args to timelapse.sh
$TIMELAPSE_EXTRA_PARAMETERS should NOT be passed to timelapse.sh since it's in config.sh and is already on the command line for ffmpeg in timelapse.sh.

Fixes #866
commit c94670ab (2021-12-25): generateForDay.sh: bug in args to timelapse.sh

$TIMELAPSE_EXTRA_PARAMETERS should NOT be passed to timelapse.sh since it's in config.sh and is already on the command line for ffmpeg in timelapse.sh.

Fixes #866
closed 2021-12-25 by EricClaeys