AllskyTeam/allsky #879: Gain problem

Shand359, 2021-12-29:
Whenever i think i have things sorted out....

My daytime images are set at 1 sec exposure and gain at 1 with it set to switch to night settings at 0 degrees. I am typing this at 8pm and its pitch black outside but its still using the daytime settings

EricClaeys, 2021-12-30:
@Shand359 Please see [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues) and attach your log file.

To see what time Allsky thinks your sunrise and sunset are, execute: `./sunwait list angle 0 41.79N 88.12W` replacing my latitude and longitude with yours.

EricClaeys, 2022-01-07:
@Shand359    Any update?

EricClaeys, 2022-04-12:
Closing Issue - no response from user in 4 months.

closed 2022-04-12 by EricClaeys