AllskyTeam/allsky #880: "std::out_of_range" during day time exposure

jmodrako, 2021-12-31:
Hey, from time to time I have this exception:

```
Dec 31 08:01:34 allsky allsky.sh[20737]: STARTING EXPOSURE at: 2021-12-31 08:01:34   @ 2.00 ms (0.00 sec)
Dec 31 08:01:34 allsky allsky.sh[20737]:   > Capture command: raspistill --output 'image.jpg' --thumb none --burst -st --timeout 10 --nopreview --mode 3 --exposure off --shutter 2000 --analoggain 1 --awb off --awbgains 3.38,1.8 --saturation 20 --brightness 50 --quality 100  -a 1036 -ae 32,0xff,0x808000
Dec 31 08:01:38 allsky allsky.sh[20737]: terminate called after throwing an instance of 'std::out_of_range'
Dec 31 08:01:38 allsky allsky.sh[20737]:   what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
Dec 31 08:01:39 allsky allsky.sh[20737]: /home/pi/allsky/allsky.sh: line 210: 20829 Aborted                 "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Dec 31 08:01:39 allsky allsky.sh[20737]: 'capture_RPiHQ' exited with RETCODE=134
Dec 31 08:01:39 allsky allsky.sh[20737]: ***
Dec 31 08:01:39 allsky allsky.sh[20737]: *** After fixing, restart the allsky service. ***
Dec 31 08:01:39 allsky allsky.sh[20737]: ***
```

Version:

```
commit a6ab44efecb1a9af3e57dee93a278cd5fad581bb (HEAD -> master, origin/master, origin/HEAD)
Merge: f5c0c38 530f168
Author: AndreasLMeg <85451655+AndreasLMeg@users.noreply.github.com>
Date:   Sat Dec 4 20:07:19 2021 +0100

    Merge pull request #843 from thomasjacquin/info_text
    
    capture_RPiHQ.cpp: fix info_text
```

AndreasLMeg, 2022-01-02:
@jmodrako: Please try the commands and send me the output:

cat /var/log/messages | grep out_
cat /var/log/messages | grep out_ -A15 -B20

jmodrako, 2022-01-02:
No results at all:

```
/etc [astroberry@astroberry] [17:39]
> cat /var/log/messages | grep out_ -A15 -B20

/etc [astroberry@astroberry] [17:39]
> cat /var/log/messages | grep out_          

```

AndreasLMeg, 2022-01-02:
And with allsky.log ?
cat /var/log/allsky.log | grep out_
cat /var/log/allsky.log | grep out_ -A15 -B20

AndreasLMeg, 2022-01-02:
@jmodrako:  I think I can reproduce it. But I'll do another test tomorrow.

jmodrako, 2022-01-02:
That's awesome! Regarding logs there is nothing too in the allsky.log...

AndreasLMeg, 2022-01-02:
maybe allsky.log.1 contains some information

EricClaeys, 2022-01-02:
@AndreasLMeg I assume the issue is in raspistill, and not capture_RPiHQ?

AndreasLMeg, 2022-01-03:
@jmodrako 
if I use 9ms for Daytime settings: Delay I get the "same" problem.
probably not the same trigger, but the same impact.
Can you attach the allsky.log and allsky.log.1 file ?

```
Jan  3 08:10:55 allsky allsky.sh[17103]: [INFO   ]: STARTING EXPOSURE at: 2022-01-03 08:10:55   @ 8.44 ms (0.01 sec)
Jan  3 08:10:58 allsky allsky.sh[17103]: [INFO   ]:   > image.jpg: 0.0 sec, mean: 0.298169 0.001831
Jan  3 08:11:00 allsky allsky.sh[17103]: [INFO   ]: Sleeping 0.0 seconds...
Jan  3 08:11:00 allsky allsky.sh[17103]: [INFO   ]: ----------------------------
Jan  3 08:11:00 allsky allsky.sh[17103]: [INFO   ]: STARTING EXPOSURE at: 2022-01-03 08:11:00   @ 8.44 ms (0.01 sec)
Jan  3 08:11:01 allsky allsky.sh[17103]: Premature end of JPEG file
Jan  3 08:11:02 allsky allsky.sh[17103]: terminate called after throwing an instance of 'std::out_of_range'
Jan  3 08:11:02 allsky allsky.sh[17103]:   what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
Jan  3 08:11:02 allsky allsky.sh[17103]: /home/pi/allsky/allsky.sh: Zeile 210: 17178 Abgebrochen             "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Jan  3 08:11:02 allsky allsky.sh[17103]: 'capture_RPiHQ' exited with RETCODE=134
Jan  3 08:11:07 allsky allsky.sh[17614]:      ***** Starting AllSky *****

```

jmodrako, 2022-01-03:
Here are the files: [allsky_logs.zip](https://github.com/thomasjacquin/allsky/files/7801578/allsky_logs.zip)

AndreasLMeg, 2022-01-03:
@jmodrako thank you for the files.

as far as I can adjust the problem is a function that loads the current image into memory for further processing. 
cv :: imread (fileName, cv :: IMREAD_UNCHANGED);

We still have to come up with something, because the image is currently being read and manipulated too often. (capture, meanvalue, overlay text, resize, annotate, web-ui...)
The probability of errors decreases with the waiting time after the capture.
Is there a reason for your short waiting time (10s) during the day? (Monitoring function?)
Otherwise I would recommend a waiting time of 60s. (Protects the Raspi, the SD and the camera)

I would also like to recommend the following settings:
```
Daytime settings:
Delay: 60000

Nighttime settings:
Auto-Exposure: Yes
Exposure: 60000
Delay: 1000
Auto-Gain: Yes
Gain: 15

```

```
# Additional Capture parameters.  Run "capture_RPiHQ -h" to see the options.
# e.g. activate autoexposure algorithm "mean" for RPiHQ: "-mean-value 0.3 -autoexposure 1"
CAPTURE_EXTRA_PARAMETERS="-mean-value 0.3 -autoexposure 1"

```

jmodrako, 2022-01-03:
Thanks, I will change the settings to the recommended ones 👍🏻 And thank you for the quick investigation and provided recommendations ;)

jmodrako, 2022-01-04:
There are still some problems though:

```
Jan  4 15:57:15 allsky allsky.sh[17798]:      ***** Starting AllSky *****
Jan  4 15:57:15 allsky allsky.sh[17798]: CAMERA: RPiHQ
Jan  4 15:57:15 allsky allsky.sh[17798]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Jan  4 15:57:15 allsky allsky.sh[17798]: Starting allsky camera...
Jan  4 15:57:20 allsky allsky.sh[17798]:  ********************************************
Jan  4 15:57:20 allsky allsky.sh[17798]:  *** Allsky Camera Software v0.8.1 | 2021 ***
Jan  4 15:57:20 allsky allsky.sh[17798]:  ********************************************
Jan  4 15:57:20 allsky allsky.sh[17798]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Jan  4 15:57:20 allsky allsky.sh[17798]: Add -h or --help for available options
Jan  4 15:57:20 allsky allsky.sh[17798]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Jan  4 15:57:20 allsky allsky.sh[17798]: Contributors:
Jan  4 15:57:20 allsky allsky.sh[17798]: -Knut Olav Klo
Jan  4 15:57:20 allsky allsky.sh[17798]: -Daniel Johnsen
Jan  4 15:57:20 allsky allsky.sh[17798]: -Robert Wagner
Jan  4 15:57:20 allsky allsky.sh[17798]: -Michael J. Kidd - <linuxkidd@gmail.com>
Jan  4 15:57:20 allsky allsky.sh[17798]: -Rob Musquetier
Jan  4 15:57:20 allsky allsky.sh[17798]: -Eric Claeys
Jan  4 15:57:20 allsky allsky.sh[17798]:  Camera: Raspberry Pi HQ camera
Jan  4 15:57:20 allsky allsky.sh[17798]:   - Resolution: 4096x3040
Jan  4 15:57:20 allsky allsky.sh[17798]:   - Pixel Size: 1.55microns
Jan  4 15:57:20 allsky allsky.sh[17798]:   - Supported Bins: 1x, 2x and 3x
Jan  4 15:57:20 allsky allsky.sh[17798]: Capture Settings:
Jan  4 15:57:20 allsky allsky.sh[17798]:  Command: raspistill
Jan  4 15:57:20 allsky allsky.sh[17798]:  Resolution (before any binning): 4096x3040
Jan  4 15:57:20 allsky allsky.sh[17798]:  Quality: 100
Jan  4 15:57:20 allsky allsky.sh[17798]:  Daytime capture: 1 (yes)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Exposure (night): 60000ms
Jan  4 15:57:20 allsky allsky.sh[17798]:  Auto Exposure (night): 1 (yes)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Gain (night): 15.00
Jan  4 15:57:20 allsky allsky.sh[17798]:  Auto Gain (night): 1 (yes)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Brightness (day): 50
Jan  4 15:57:20 allsky allsky.sh[17798]:  Brightness (night): 50
Jan  4 15:57:20 allsky allsky.sh[17798]:  Saturation: 20.0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Auto White Balance: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  WB Red: 3.38
Jan  4 15:57:20 allsky allsky.sh[17798]:  WB Blue: 1.80
Jan  4 15:57:20 allsky allsky.sh[17798]:  Binning (day): 1
Jan  4 15:57:20 allsky allsky.sh[17798]:  Binning (night): 1
Jan  4 15:57:20 allsky allsky.sh[17798]:  Delay (day): 60000ms
Jan  4 15:57:20 allsky allsky.sh[17798]:  Delay (night): 1000ms
Jan  4 15:57:20 allsky allsky.sh[17798]:  Text Overlay:
Jan  4 15:57:20 allsky allsky.sh[17798]:  Text Position: 485px left, 75px top
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Name:  0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Color: 255, 255, 255
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Background Color: 0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Small Font Color: 0, 0, 0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Line Type: 16
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Size: 32.0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Font Line: 1
Jan  4 15:57:20 allsky allsky.sh[17798]:  Outline Font : 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Rotation: 0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Flip Image: 0
Jan  4 15:57:20 allsky allsky.sh[17798]:  Filename: image.jpg
Jan  4 15:57:20 allsky allsky.sh[17798]:  Latitude: 53.19N
Jan  4 15:57:20 allsky allsky.sh[17798]:  Longitude: 17.77E
Jan  4 15:57:20 allsky allsky.sh[17798]:  Sun Elevation: -5
Jan  4 15:57:20 allsky allsky.sh[17798]:  Locale: en_US.UTF-8
Jan  4 15:57:20 allsky allsky.sh[17798]:  Notification Images: 1 (yes)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Show Time: 1 (yes) (format: %Y%m%d %H:%M:%S)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Show Exposure: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Show Gain: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Show Brightness: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Show Focus Metric: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Preview: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Taking Dark Frames: 0 (no)
Jan  4 15:57:20 allsky allsky.sh[17798]:  Debug Level: 3
Jan  4 15:57:20 allsky allsky.sh[17798]:  On TTY: No
[... 43 lines cut ...]
Jan  4 15:57:32 allsky allsky.sh[17798]:   > Capture command: raspistill --output 'image.jpg' --thumb none --burst -st --timeout 10 --nopreview --mode 3 --exposure off --shutter 221 --analoggain 1 --exif IFD0.Artist=li_221_1 --awb off --awbgains 3.38,1.8 --saturation 20 --brightness 50 --quality 100  -a 1036 -ae 32,0xff,0x808000
Jan  4 15:57:34 allsky allsky.sh[17798]:   > image.jpg: 0.0 sec, mean: 0.000013 0.299987
Jan  4 15:57:34 allsky allsky.sh[17798]:   > MeanCnt: 2, mean_historySize: 3
Jan  4 15:57:34 allsky allsky.sh[17798]:   > i=1: idx=0 mean=0.8979 exp=-1
Jan  4 15:57:34 allsky allsky.sh[17798]:   > i=2: idx=1 mean=0.5020 exp=-393
Jan  4 15:57:34 allsky allsky.sh[17798]:   > i=3: idx=2 mean=0.0000 exp=-437
Jan  4 15:57:34 allsky allsky.sh[17798]:   > mean_forecast: 0.0000
Jan  4 15:57:34 allsky allsky.sh[17798]:   > values: 9
Jan  4 15:57:34 allsky allsky.sh[17798]:   > mean_diff: 0.0887
Jan  4 15:57:34 allsky allsky.sh[17798]:   > ExposureChange: 22 (-22)
[... cut ...]
Jan  4 15:57:39 allsky allsky.sh[17798]:   what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
Jan  4 15:57:39 allsky allsky.sh[17798]: /home/pi/allsky/allsky.sh: line 210: 17896 Aborted                 "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Jan  4 15:57:39 allsky allsky.sh[17798]: 'capture_RPiHQ' exited with RETCODE=134
Jan  4 15:57:39 allsky allsky.sh[17798]: ***
Jan  4 15:57:39 allsky allsky.sh[17798]: *** After fixing, restart the allsky service. ***
Jan  4 15:57:39 allsky allsky.sh[17798]: ***

```

AndreasLMeg, 2022-01-04:
@jmodrako some questions:
1. is you sd card slow or defect ?
```
Jan  4 15:57:30 allsky allsky.sh[17798]: convert-im6.q16: Corrupt JPEG data: 97377 extraneous bytes before marker 0xd9 `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Jan  4 15:57:31 allsky allsky.sh[17798]: convert-im6.q16: Corrupt JPEG data: 48422 extraneous bytes before marker 0xd9 `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.

```
2. we have short shutter times, but the images are overexposed - is this a test (indoor) or outdoor ?
3. can you send me some images and the log file
4. Raspberry Pi type ?

EricClaeys, 2022-01-04:
@AndreasLMeg this might be due to the 1 second sleep between images when "fast" mode is on (I forget the exact name, but it's when you are quickly going through the first few pictures).  If it takes the Pi longer than a second to process the image the next exposure will overwrite the first one.

This shouldn't be an issue with the saveImage.sh PR code though since every file from raspistill/libcamera has a unique name.  However, we may want to consider not even saving the "fast" frames whose exposure is way off, sonce they will only cause flickering in videos.  In the ZWO version I have the concept of "skip frames" which are the number of frames skipped (i.e. not saved) when first starting the program.  The defaults are 5 during the day and 1 at night.

AndreasLMeg, 2022-01-04:
@EricClaeys  Yes, I saw that. Unfortunately I did not take over the quickstart parameter from the PR, so it is always active at the start (10x1s).

But for me there was never a problem (500ms). In addition, 10s waiting time was still problematic for Jacek.
I'm afraid there is something wrong with the Raspi (old Raspi, memory, CPU load, SD, ...)
I have to think of a test for that ...

AndreasLMeg, 2022-01-04:
@jmodrako 
here is one test file: https://github.com/AndreasLMeg/allsky/blob/li_overlay_text/scripts/check.sh
or as file (please rename it to check.sh):
[check.sh.txt](https://github.com/thomasjacquin/allsky/files/7810532/check.sh.txt)
 
copy it to ~/allsky/scripts
chmod +x check.sh
execute it from ~/allsky/ directory: ./scripts/check.sh

and yes, I think I have to change this quickstart with 1s 
```
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 5s
Di Jän  4 22:35:54 CET 2022
Di Jän  4 22:36:27 CET 2022
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 2s
Di Jän  4 22:36:27 CET 2022
Di Jän  4 22:36:44 CET 2022
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 1s
Di Jän  4 22:36:44 CET 2022
/home/pi/allsky/scripts/saveImageNight.sh: Zeile 46: 15426 Abgebrochen             convert "$IMAGE_TO_USE" -resize "$IMG_WIDTH"x"$IMG_HEIGHT" "$IMAGE_TO_USE"
*** saveImageNight.sh: ERROR: IMG_RESIZE failed with RET=134{}
/home/pi/allsky/scripts/saveImageNight.sh: Zeile 91: 15514 Abgebrochen             convert "$IMAGE_TO_USE" -resize "${THUMBNAIL_SIZE_X}x${THUMBNAIL_SIZE_Y}" "$THUMBNAILS_DIR/$SAVED_FILE"
*** saveImageNight.sh: WARNING: THUMBNAIL resize failed with RET=134; continuing.
Di Jän  4 22:36:56 CET 2022
```

AndreasLMeg, 2022-01-05:
@jmodrako: I have 3 new files to test (scripts) 
https://github.com/AndreasLMeg/allsky/blob/c039b0f56465f98318ed631d44196b7de6a165fe/scripts/check.sh
https://github.com/AndreasLMeg/allsky/blob/c039b0f56465f98318ed631d44196b7de6a165fe/scripts/saveImageDay.sh
https://github.com/AndreasLMeg/allsky/blob/c039b0f56465f98318ed631d44196b7de6a165fe/scripts/saveImageNight.sh

AndreasLMeg, 2022-01-14:
@jmodrako 
thank you for the response. As you can see in log1.txt everything was OK for sleep 20,10 and 5s between the captures.
For 2s, 1s the convert, upload and maybe other things take too long and the next capture will destroy the old image

This is only a simple test, so I would use 20s for the delay. The next version of allsky has solved this problem, but wait 1 week to take this - we have still some other issues inside. 

Is this ok for you? Esle you have to contact me in a week or so.

```
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 20s
Fri 14 Jan 2022 04:05:58 PM CET
Fri 14 Jan 2022 04:07:46 PM CET
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 10s
Fri 14 Jan 2022 04:07:46 PM CET
Fri 14 Jan 2022 04:08:44 PM CET
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 5s
Fri 14 Jan 2022 04:08:44 PM CET
Fri 14 Jan 2022 04:09:17 PM CET
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 2s
Fri 14 Jan 2022 04:09:17 PM CET
[0;31m*** darkSubtract.sh: ERROR: Image 'image.jpg' is corrupt; ignoring.[0m
[0;31m*** darkSubtract.sh: ERROR: Image 'image.jpg' is corrupt; ignoring.[0m
Fri 14 Jan 2022 04:09:34 PM CET
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 1s
Fri 14 Jan 2022 04:09:34 PM CET
[0;31m*** darkSubtract.sh: ERROR: Image 'image.jpg' is corrupt; ignoring.[0m
[0;31m*** darkSubtract.sh: ERROR: IMG_RESIZE failed with RET=134{[0m}
[0;31m*** darkSubtract.sh: ERROR: IMG_RESIZE failed with RET=134{[0m}
Fri 14 Jan 2022 04:09:48 PM CET
------------------------------------------------
check 5x capture + SaveImageNight & + sleep 0.5s
Fri 14 Jan 2022 04:09:48 PM CET
[0;31m*** darkSubtract.sh: ERROR: Image 'image.jpg' is corrupt; ignoring.[0m
[0;31m*** darkSubtract.sh: ERROR: IMG_RESIZE failed with RET=134{[0m}
Fri 14 Jan 2022 04:09:58 PM CET
------------------------------------------------
```

AndreasLMeg, 2022-01-14:
I think "DARK_FRAME_SUBTRACTION" is active. Please deaktivate this.  (uses also the convert application)

# Set to "true" if you want to enable dark frame subtraction on your night-time images.
# You must first create dark frames - see the README.md file for instructions.
DARK_FRAME_SUBTRACTION="false"

EricClaeys, 2022-01-14:
@jmodrako As @AndreasLMeg said, this problem should be fixed in 0.8.3, which saves every image with a unique name so the images aren't overwritten.
I think it should be safe to upgrade as long as you follow directions on the second Wiki page exactly.  One of the first steps is to rename your "allsky" directory so if you have major problems you can easily revert to the prior version.

@AndreasLMeg. I assume DARK_FRAME_SUBTRACTION won't need to be disabled in 0.8.3?

AndreasLMeg, 2022-01-14:
> @AndreasLMeg. I assume DARK_FRAME_SUBTRACTION won't need to be disabled in 0.8.3?
Yes, I think so 👍

EricClaeys, 2022-02-25:
@jmodrako What version are you running?  You can search for "Software" in /var/log/allsky.log.  It's it's prior to 0.8.3, please upgrade as that version has significantly reduced the risk of one image overwriting another, which is what I think is happening in your case.

EricClaeys, 2022-06-03:
@jmordrako,   Any update?

EricClaeys, 2022-06-30:
Closing Issue - no user response in several months, and no one else has reported this problem.

closed 2022-06-30 by EricClaeys