AllskyTeam/allsky #881: Keogram image trimmed at the top + text

Jonk2, 2021-12-31:
By @EricClaeys request, I've started a new issue regarding the keogram.

It hasn't been trimmed by the software, that was my mistake sorry - my sensor pcb is not central to the image circle, I need to fix that:

[image]

However as also mentioned on the othe thread, is there the possbility of the date at the below of the keogram being turned off, moved or otherwise user editable?

I guess the same goes for the text showing the hours. Maybe a nice feature to have the user choose the font / colour / size etc?

EricClaeys, 2022-01-01:
@Jonk2 
I'll look at what it takes to not show the date.
You can already turn off the hours using the "--no-label" option.  Set it in the KEOGRAM_EXTRA_ARGUMENTS variable in config.sh.  You can also change the font color, linewidth, name, size, and type.  To see all the options:
`cd allsky; ./keogram --help`

EricClaeys, 2022-02-19:
Closing.  Just submitted update to keogram to add "-D" / "--no-date" option to disable displaying the date.  Turning off labels still disables the date.

commit 23ff18f5 (2022-02-19): Update keogram.cpp

Fixes #881 
* Allow users to turn off displaying date via new -D option.
* Added #define's for annotation arguments (date, hour, etc.)
* Replaced "prevHour" check with check if we already added the current hour to the annotation list.  The old method would miss hours when run in multi-processor mode.
* Date is always displayed - at the 0 hour if possible, otherwise right-justified.  It used to only be displayed if there was a 0 hour.
* Minor optimization - put column in new "column" variable rather than recalculate it multiple times.  This also makes the code easier to read [... cut]
PR #1043 (2022-03-06): Keogram.cpp misc changes
Fixes #881

Adds -D option to Disable displaying the date without disabling the hour labels.
Fixes issues with not always displaying the date.
Warns if an hour label or the date can't be displayed due to lack of space.
Displays the date even if there is no data for hour 0.
Other misc. changes - see the commits.
closed 2022-02-19 by EricClaeys