AllskyTeam/allsky #882: throttle status

pierrebilb, 2022-01-01:
Hi all and happy new year,
i just install yesterday allsky software on orange pi+2e; on the webUI system page, i've got this message:
Not able to get throttle status:
sudo: no tty present and no askpass program specified
Run 'sudo ~/allsky/gui/install.sh --update' to try and resolve

I run the last line without any effect. I don't know if there is a link, but timelapse and keogram creations failed..
Any help welcome
Pierre

EricClaeys, 2022-01-01:
@pierrebilb For the timelapse and keogram issues, please follow the directions on the Wiki page on reporting issues, then upload your log file.

Are you on Bullseye?  To find out, run `cat /etc/os-release`.

Please do a `tail -2 /etc/sudoers.d/allsky` and post the results.

pierrebilb, 2022-01-02:
Hi Eric,
i'm on armbian Buster
Here is the result of tail -2 /etc/sudoers.d/allsky:
www-data ALL=(ALL) NOPASSWD:/bin/mv
www-data ALL=(ALL) NOPASSWD:/usr/bin/vcgencmd

Last night, keogram and startrails worked but not the timelapse. I will report issue as on wiki after next night..
Thanks
Pierre

AndreasLMeg, 2022-01-02:
> Hi all and happy new year,
> i just install yesterday allsky software on orange pi+2e; on the webUI system page, i've got this message:
> Not able to get throttle status:

@pierrebilb: please try the 2 commands and send us your output.

`pi@allsky:~/allsky $ sudo vcgencmd get_throttled`
my result was: throttled=0xe0000
`pi@allsky:~/allsky $ echo $?`
my result was: 0

EricClaeys, 2022-01-02:
@AndreasLMeg I had this problem on my Pi as well.  `vcgencmd `worked fine from the command line but not from the webUI.  I'm not sure what I did to fix it, but it eventually worked.  Possibly a reboot.

@pierrebilb Would you please reboot your Pi and let us know if that fixes it?

FYI, `0xe0000` means:
- Throttling has occurred since last reboot.
- ARM frequency capped has occurred since last reboot.
- Soft temperature limit has occurred

pierrebilb, 2022-01-05:
Hi,
it seems there is a problem with vcgencmd: no vcgencmd found, but after that, my orangepi crashed and don't want to restart from remote. I will tell you more when i'm able to restart my allsky

EricClaeys, 2022-02-10:
@pierrebilb   Any update?
Can you run `cat /etc/os-release` and post the results?  I'm thinking you may have the "light" version of the operating system which may not include vcgencmd.

After that, run `sudo apt-get install libraspberrypi-bin` which should install the `vcgencmd` command, so run `vcgencmd get_throttled`.  If that works, try the "System" page in the WebUI again.

EricClaeys, 2022-04-12:
Closing issue - no response from user in 2 months.  Other people aren't having this issue so I assume it was a configuration issue on the user's Pi.

closed 2022-04-12 by EricClaeys