AllskyTeam/allsky #885: Question

maserowik, 2022-01-04:
I did a new git pull of the software and have a question.  I select ZWO since I am using a ZWO camera.  I see in the install screen it builds both the capture and the capture_RPIHQ program.  Is there a way not to have the system build both programs?

[image]

EricClaeys, 2022-01-04:
@maserowik Not easily.  At install time we don't know with certainty what camera you have, or in the case of people with multiple cameras, which one they will use.  What if a user changes camera types?  The software would need to be smart enough to realize that and then compile the new camera.
Plus, we are looking at having one executable file that works with both cameras.

maserowik, 2022-01-04:
sounds logical.  
Closing issue.

closed 2022-01-05 by maserowik