AllskyTeam/allsky #886: Question 2

maserowik, 2022-01-04:
There was a version of allsky that would show a map of the world that had the Aurora overlay.  Is there a way to get it back when the camera is off during the day?
Mike

EricClaeys, 2022-01-04:
@maserowik do you remember when this was? 6 months ago?  2 years ago?  Etc.
I assume this was the Allsky website and not the WebUI or allsky itself?

maserowik, 2022-01-04:
@EricClaeys 
I know it was in the version that Thomas made before all the changes were incorporated into this software.  How I remember it there was a gif of the world that show the Aurora overlay and stated the camera was offline during the day and there was a count down that show when to check back.

EricClaeys, 2022-01-05:
@maserowik Did you see it in the WebUI, or in the website (with the Casseopea icon thingy on the left side)?

maserowik, 2022-01-05:
@EricClaeys 
I think this should be moved to the webui group.  
here is what I am looking at 
https://github.com/thomasjacquin/allsky-website/issues/26

maserowik, 2022-01-05:
I am closing this and opened it up in the allsky-website forum

closed 2022-01-05 by maserowik