AllskyTeam/allsky #888: capture_RPiHQ.cpp: mean issues

EricClaeys, 2022-01-07:
The user-specified day and night values for auto-exposure, gain, and exposure should be used by the mode-mean code.
The initial exposure and gain values should be starting points when the program begins.
Need to add variable for the max values of gain and exposure for day and night, and have RPiHQcalcMean() use them.

Need to change ZWO's gain from 0 to max-supported-by-camera to 0 to 1 so mode-mean can use it.

AndreasLMeg, 2022-01-07:
@EricClaeys 
to make it camera independend we/I have to refacture the function - is this urgent ?
```
// Calculate new raspistillSettings (exposure, gain)
// Algorithm not perfect, but better than no exposure control at all
float RPiHQcalcMean(const char* fileName, int exposure_us, double gain, raspistillSetting &currentRaspistillSetting, modeMeanSetting &currentModeMeanSetting)

```

EricClaeys, 2022-01-07:
@AndreasLMeg Not urgent.  I created the Issue so we could track the things that we need to do to the mode mean code.

AndreasLMeg, 2022-01-14:
Before my change (mean algo), only fixed values ​​were possible with the RaspiHQ camera.
This led to overexposure in the twilight. There were also problems with the moon and extraneous light from cars.

I wanted to find a solution for this without having to constantly change the parameters. (#793)

And I think it has become an optimal solution for the night.
During the day the pictures are a bit too dark, but that didn't bother me as I'm not interested in day shots.
Older Allsky versions also had restrictions on the settings (Day/Night, AutoExposure, AutoGain,...)

Hence the recommendation: "-mean-value 0.3 -autoexposure 1"
This is also the reason why the settings in config.sh are overwritten.
If you don't want this delete "-autoexposure 1"

The feature currently requires autogain and autoexposure to be enabled, if not there are known side effects
If you don't want that you have to remove the extra parameters completely

My next development is to support autoGainOnly, AutoexposureOnly and NoAuto
If you want this, you have to wait a short time

After this I will look to make the day images better or to switch to "autoexposure" of raspistill/libcamera
If you want this, you have to wait a little bit longer

AndreasLMeg, 2022-01-14:
@EricClaeys please "review" my changes in https://github.com/AndreasLMeg/allsky/pull/9

tonight i will make some tests, hopefully i works now for the others....

```
// 2022-01-14  MEAN_AUTO_MODE, depending in autoGain and autoExposure different modes are in use  
//             new optional start parameter -mean-value-day
//             use can define different values for day and night (autoExposure, Exposure, mean-value,...)

```

AndreasLMeg, 2022-01-14:
I have now made some extraordinary day/night switches. A few abnormalities in the control behavior became visible.
I still have to check that. Otherwise it looks pretty good

AndreasLMeg, 2022-01-15:
Here is one test (exposure_day=20s,exposure_night=30s) 
look at EXPOSURE_US, MEAN, GAIN
```
18:41:25 ======================================================================
18:41:25 MOCK: s ./test_dayNightChange.sh: 
18:41:59 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184115.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.302 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:42:00 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:42:00 MOCK: < /usr/local/bin/sunwait: NIGHT
18:42:44 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184200.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.302 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:42:45 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:42:45 MOCK: < /usr/local/bin/sunwait: NIGHT
18:43:29 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184245.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.300 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:43:30 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:43:30 MOCK: < /usr/local/bin/sunwait: NIGHT
18:44:14 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184330.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.299 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:44:15 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:44:15 MOCK: < /usr/local/bin/sunwait: NIGHT
18:44:45 ----------------------------------------------------------------------
18:44:45 MOCK: i /home/pi/allsky/mock_variables: ALLSKY_MOCK_SUNWAIT=DAY
18:44:59 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184415.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.299 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:45:00 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:45:00 MOCK: < /usr/local/bin/sunwait: DAY
18:45:00 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:45:00 MOCK: < /usr/local/bin/sunwait: DAY
18:45:00 MOCK: > scripts/endOfNight.sh: 
18:45:10 MOCK: i scripts/endOfNight.sh: 10 s slept
18:45:36 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184500.jpg EXPOSURE_US=29999999 BRIGHTNESS=50 MEAN=0.299 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=4.52 GAINDB=131 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 35
18:45:37 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:45:37 MOCK: < /usr/local/bin/sunwait: DAY
18:46:03 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184537.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.499 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 57
18:46:04 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:46:04 MOCK: < /usr/local/bin/sunwait: DAY
18:46:33 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184604.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.497 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 57
18:46:34 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:46:34 MOCK: < /usr/local/bin/sunwait: DAY
18:47:03 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184634.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.500 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 57
18:47:04 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:47:04 MOCK: < /usr/local/bin/sunwait: DAY
18:47:30 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184704.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.499 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 58
18:47:31 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:47:31 MOCK: < /usr/local/bin/sunwait: DAY
18:47:58 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184731.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.497 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 57
18:47:59 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:47:59 MOCK: < /usr/local/bin/sunwait: DAY
18:48:05 ----------------------------------------------------------------------
18:48:05 MOCK: i /home/pi/allsky/mock_variables: ALLSKY_MOCK_SUNWAIT=NIGHT
18:48:28 MOCK: > scripts/saveImage.sh: DAY /home/pi/allsky/tmp/image-20220115184759.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.499 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 58
18:48:29 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:48:29 MOCK: < /usr/local/bin/sunwait: NIGHT
18:48:29 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:48:29 MOCK: < /usr/local/bin/sunwait: NIGHT
18:49:03 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115184829.jpg EXPOSURE_US=20000000 BRIGHTNESS=50 MEAN=0.500 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=16.00 GAINDB=241 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 56
18:49:03 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
[... cut ...]
18:50:31 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:50:31 MOCK: < /usr/local/bin/sunwait: NIGHT
18:51:15 MOCK: > scripts/saveImage.sh: NIGHT /home/pi/allsky/tmp/image-20220115185031.jpg EXPOSURE_US=30000000 BRIGHTNESS=50 MEAN=0.278 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.80 WBB=2.00 GAIN=3.95 GAINDB=119 BIN=1 FLIP=3 BIT_DEPTH=08 FOCUS= 36
18:51:15 MOCK: > /usr/local/bin/sunwait: poll angle -9 48.18N 13.80E
18:51:15 MOCK: < /usr/local/bin/sunwait: NIGHT
18:51:25 ======================================================================
```

closed 2022-02-16 by AndreasLMeg