AllskyTeam/allsky #891: Benutzen von Dar-Bildern

Torakin9366, 2022-01-07:
Hallo zusammen,
ich habe das Programm neu aufgesetzt und alle Eingestellt so das es mir gefällt. Ich nutze den Rpi 4 4gB und die Asi 290 MC.
Bevor ich die Asi Verbaut hab, habe ich noch Darkbilder erstelt. Im Dark Ordner sind jetzt Bilder von -7° bis +35° gespeichert. Die Darks hab ich am Tag mit abgedeckter Kamera aufgenommen. In der config habe ich Nutzung der Darks auf true eingestellt. Aber wenn ich mir dann Nachts die Bilder anschaue, sehen die sehr schlecht aus. Noch mehr verpixelt ls wenn ich ohne Darks bearbeite. 
Gibt es da eine Lösung?
Danke schön.

EricClaeys, 2022-01-07:
@Torakin9366 Try setting the "Debug Level" to 4 in the WebUI to see what dark file Allsky is applying to your images.  "Debug Level" is an advanced option so you'll need to show it - at the bottom of the "Camera Settings" page in the WebUI, click on the "Show advanced options" button.  The "Debug Level" is near the bottom.

It should apply a dark of the same, or very similar, temperature to what the sensor is.

Are the exposures of your images near the maximum exposure you set for nighttime shots?  The darks are taken at the maximum exposure (for example, 45 seconds), and if your light frames are taken at 5 seconds, the results won't be good.  Try setting the nighttime maximum exposure to slightly above the longest exposure you take at night.

If you have an image editing program, look at one of the darks and lighten it as much as possible.  It should look like salt-and-pepper.

Torakin9366, 2022-01-09:
[image]
[image]
[image]
Das mit den Einstellung ist mir auch eingefallen, also habe ich neue Dark`s gemacht mit den Einstellungen die ich auch Nachts benutze Exposure 50000. Als Vergleich habe ich mal die Bilder eingestellt die ich gemacht habe. Ich bin damit nicht zufrieden, Sollte ich mich vin der Kamera verabschieden?

EricClaeys, 2022-01-10:
@Torakin9366 No need to get rid of the camera.
What temperature was the first image (the dark)?  Did you lighten it before attaching it?  Is it an original dark, or a new one?
Is the 2nd picture with the old darks taken at your maximum nighttime exposure?
Is the 3rd picture with a new dark taken at 50 seconds?

I'd like to work with you to figure this out, as it likely impacts lots of people.
I think if we can get one dark and one normal night picture taken with known settings we can figure this out.  Please do this _at night_:
~~~
sudo systemctl stop allsky     # Stop the allsky service
sudo truncate -s 0 /var/log/allsky.log   # clear out log file
cd ~/allsky/darks
mkdir SAVED
mv *.jpg SAVED    # save your current darks
# Cover your camera
~~~
Make these changes in the WebUI:
* In the "Nighttime Settings" section:
   * If "Auto-Exposure" is on, turn it off.
   * Set "Max Auto-Exposure" to your "Manual Exposure" (probably 50 seconds).
   * Turn OFF "Auto-Gain".
* Change "Debug Level" to 4.
* Turn on "Dark Frame".
* Click on the "Save changes" button. This will restart the service.

Wait until you have a new dark in allsky/darks, then stop Allsky.
Uncover your camera.

Take a regular image with the same settings as above:
* In config/config.sh make sure `DARK_FRAME_SUBTRACTION="true".
* In the WebUI, turn off "Dark Frame" the click the "Save changes" button.

Wait until you get a normal nighttime image or two, then stop Allsky.
Attach /var/log/allsky.log (do not copy/paste it) to your reply.  Also, zip the dark and image file and attach them.

To restore your Pi:
~~~
cd ~/allsky/darks
# rename the new dark to XXdark.jpg-TEST where "XX" is the temperature
mv SAVED/* .    # restore your original darks
rmdir SAVED
# Turn "Auto-Gain" back on in the WebUI.
~~~

Torakin9366, 2022-01-10:
Ich habe am Samstag die Kamera abgedunkelt sie in den Kühlschrank gelegt und dann Darks mit manuell 50 sek gemacht. Habe somit dann Darks von -10°C bis +28°C erhalten. Die Bilder die ich gepostet habe sind alle mit 17° Bild 1=Dark 17°C, Bild 2= Sensor 17° ohne Dark, Bild 2= Sensor 17° mit Dark.
Ich werde dein Vorschlag demnächst, wenn wieder klare Nacht ist, einmal umsetzen.
Danke schonmal für deine Hilfe.

EricClaeys, 2022-01-10:
@Torakin9366 I enjoy helping people get Allsky to work.  And it increases my knowledge of the program.
Let me know how it goes.  We are working on a script to take darks at various exposures and gain settings, in addition to different temperatures, which should help improve dark subtraction.  If my instructions improve your dark subtraction it will give us more confidence that our script will help others.

I noticed the dark you attached above has a red tint to it.  I opened it in Photoshop and it's pretty obvious.  That why I thought taking the test dark frame at night might help, if your daytime darks were letting in a tiny amount of light.

Danke

Torakin9366, 2022-01-10:
Wie von Geisterhand, läuft die Software jetzt .-))

EricClaeys, 2022-01-12:
@Torakin9366  I think we're working on your other problems in [Issue 911](https://github.com/thomasjacquin/allsky/issues/911).

How does your comment that the software is now working relate to that Issue?

closed 2022-01-19 by Torakin9366