AllskyTeam/allsky #904: New v0.8.3 (07-01-22) compile failed

Jonk2, 2022-01-09:
RPI 4b, renamed preious allsky, removed raspap folder. Ran sudo apt-get update then sudo apt-get upgrade.

Cloned new allsky folder, ran ./install.sh and it failed. Tried twice, here's the console.

```
**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make[1]: Entering directory '/home/pi/allsky/src'
2022-01-09 21:31:41 Installing build dependencies...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                  
Get:3 http://giteduberger.fr rpimonitor/ InRelease [1,933 B]                                                                     
Hit:4 https://repos.influxdata.com/debian buster InRelease                                                                       
Hit:5 https://packages.grafana.com/oss/deb stable InRelease                                                                      
Hit:6 https://www.astroberry.io/repo buster InRelease                                                    
Fetched 1,933 B in 1s (1,564 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gawk is already the newest version (1:4.2.1+dfsg-1).
imagemagick is already the newest version (8:6.9.10.23+dfsg-2.1+deb10u1).
jq is already the newest version (1.5+dfsg-2+b1).
lftp is already the newest version (4.8.4-2).
libopencv-dev is already the newest version (3.2.0+dfsg-6).
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).
libusb-dev is already the newest version (2:0.1.12-32).
ffmpeg is already the newest version (7:4.1.8-0+deb10u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
make[1]: Leaving directory '/home/pi/allsky/src'

* Compile allsky software

make[1]: Entering directory '/home/pi/allsky/src'
2022-01-09 21:31:47 Building capture program...
/usr/bin/ld: lib/armv8/libASICamera2.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:122: capture] Error 1
make[1]: Leaving directory '/home/pi/allsky/src'
make: *** [Makefile:24: all] Error 2
Compile failed!
```

EricClaeys, 2022-01-10:
@linuxkidd Can you take a look at this?  You're the compiler guru.  Thanks.

linuxkidd, 2022-01-10:
This line
```
/usr/bin/ld: lib/armv8/libASICamera2.a: error adding symbols: file format not recognized
```
Indicates it's failing to link the binary libraries from ZWO  ( libASICamera2.a ).  It's possible this file is corrupted locally?

Can you run the following:
```
md5sum ~/allsky/src/lib/armv8/libASICamera2.a
```

And provide the output... Thanks!

EricClaeys, 2022-01-10:
@linuxkidd 
Would a `file ~/allsky/src/lib/armv8/libASICamera2.a` help?

@Jonk2, CC'd

EricClaeys, 2022-01-10:
@linuxkidd I thought on my Pi 4 it used "armv7".  What does yours use?  If it's also armv7, could that be the problem?

Jonk2, 2022-01-10:
> This line
>
> ```
> /usr/bin/ld: lib/armv8/libASICamera2.a: error adding symbols: file format not recognized
> ```
>
> Indicates it's failing to link the binary libraries from ZWO ( libASICamera2.a ). It's [...]

Here you go:

```
pi@allsky:~ $ md5sum ~/allsky/src/lib/armv8/libASICamera2.a
5161e951ed924dffe7b4131c5ea87a96  /home/pi/allsky/src/lib/armv8/libASICamera2.a
```

Jonk2, 2022-01-10:
> file ~/allsky/src/lib/armv8/libASICamera2.a

and this one shows:

```
pi@allsky:~ $ file ~/allsky/src/lib/armv8/libASICamera2.a
/home/pi/allsky/src/lib/armv8/libASICamera2.a: current ar archive
```

linuxkidd, 2022-01-10:
> @linuxkidd I thought on my Pi 4 it used "armv7". What does yours use? If it's also armv7, could that be the problem?

Very good point!  Mine too is armv7.

@Jonk2 thanks for that output.  Can you also provide the following output:

```
tail -n 4 /proc/cpuinfo
cat /etc/os-release
```

Thanks!

Jonk2, 2022-01-10:
> Very good point! Mine too is armv7.
> 
> @Jonk2 thanks for that output. Can you also provide the following output:
> 
> ```
> tail -n 4 /proc/cpuinfo
> cat /etc/os-release
> ```
> 
> Thanks!

```
pi@allsky:~ $ tail -n 4 /proc/cpuinfo
Hardware	: BCM2835
Revision	: d03114
Serial		: 100000008c23cbb1
Model		: Raspberry Pi 4 Model B Rev 1.4
pi@allsky:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Jonk2, 2022-01-10:
I've just ran these commands on the pi outside and I get a different cpuinfo result (hardware) - note I haven't updated outside yet, I tend to mess about with the one indoors before I commit.

```
pi@allsky2:~ $ tail -n 4 /proc/cpuinfo
Hardware	: BCM2711
Revision	: d03114
Serial		: 10000000824c9484
Model		: Raspberry Pi 4 Model B Rev 1.4
```

Jonk2, 2022-01-10:
Apparently it's being reported wrong (the 2835 cpu).

Using _cat /proc/device-tree/model_ shows `
Raspberry Pi 4 Model B Rev 1.4` on both models.

Does this help?

linuxkidd, 2022-01-10:
That is super odd.. maybe the chip is different rev due to chip shortages? 

My test pi is also the BCM2711.

On the problem Pi, can you provide:
```
grep -m1 'model name' /proc/cpuinfo
```

linuxkidd, 2022-01-10:
scratch that request.. I found a [full cpuinfo](https://forums.raspberrypi.com/viewtopic.php?t=245384) from a Pi with this chip.

It still reports the CPU is v7l... so, I'm not sure why the make `platform` variable is matching `aarch64`, not `armv7l`.

How old was the allsky code on this device before you just tried to update?  I recall the 'aarch64' bit being added in the past 3 months or so...

Jonk2, 2022-01-10:
On the test pi that won't compile (the problem pi, showing BCM 2835), the previous allsky installed on the 17th of October ran fine.

Jonk2, 2022-01-10:
The pi outside ('live' allsky, showing BCM 2711) is running a very recent version, installed on the 2nd of January this year.
It's only because I saw a new version (0.8.3) with 'new features' a couple of days ago I decided to give it a go on the test pi....and here we are!

linuxkidd, 2022-01-10:
Interesting... I found the issue that spawned the PR ( Issue #509 ) .. but it was Sept 24th.

So, It's easy to workaround... but we need a proper fix since it's likely to come up again.  That's where things get tricky since none of us have a BCM2835 Pi to test with.  Lemme noodle on this a bit to develop a proper test for this.

Would you like a 'dirty' workaround just to get the code compiled?

Jonk2, 2022-01-10:
> Interesting... I found the issue that spawned the PR ( Issue #509 ) .. but it was Sept 24th.
>
> So, It's easy to workaround... but we need a proper fix since it's likely to come up again. That's where [...]

I'm happy to test and report back what I can if you want me to test on the bcm2835 version, but I'm not Linux inclined, I just follow instructions.

If you don't think it's worth a dirty workaround, don't worry - I can wait until you find a proper fix, test it on the bcm2835 but I won't upgrade the live allsky until it's worth doing after tests proven on the test setup.

linuxkidd, 2022-01-10:
Oohhh... @Jonk2 .. can you provide:
```
uname -a
```

There seems to be some discussion that this only occurs with the 64bit kernel, and it actually IS a reporting error.  The chip is  hard coded upstream for some reason.

Thanks!

Jonk2, 2022-01-10:
`Linux allsky 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux`

Jonk2, 2022-01-10:
> There we go... this is the 64bit raspbian, not 32bit. Thus, aarch64 is 'correct' for platform, but including the armv8 libs is not since the CPU arch is not v8. I'm seeing how we can better detect this.

That was exactly what I thought...... (not really).
Hope you can fix it!

Jonk2, 2022-01-10:
Hmm...ok thanks - I'll think about it, both pis are 8gb RAM so I always thought 32Bit OSes wouldn't be able to use over 4gb (unless that's just a windows limitation). I know allsky probably won't use anywhere near that but still...

The outside pi is newer, so the inside pi has had the 64 bit os on it for over a year. 

Amyway I digress - if you want me to test anything, please let me know, again with instructions to follow.

linuxkidd, 2022-01-10:
From the [official blog announcing the 8gb Pi availability](https://www.raspberrypi.com/news/8gb-raspberry-pi-4-on-sale-now-at-75/):
```
Our default operating system image uses a 32-bit LPAE kernel and a 32-bit userland. This allows multiple processes to share all 8GB of memory, subject to the restriction that no single process can use more than 3GB. 
```

EricClaeys, 2022-01-20:
@Jonk2 Has this Issue been resolved?  It sounds like you fixed it by installing the 32-bit version of the Raspberry OS.  If this Issue is resolved would you please close it?  Thanks.

Jonk2, 2022-01-21:
No, I didn't reinstall the OS so the issue is still active.

linuxkidd, 2022-01-22:
Well, this is very odd... 
- I booted a fresh install of [Raspberry Pi OS 64bit Lite - Bullseye](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64-lite.zip)
- ran `apt update && apt -y full-upgrade`
- rebooted
- ran the install process from the main page
- The compile went without issue.

It does have the CPU id issue:
```
# tail -n 4 /proc/cpuinfo
Hardware        : BCM2835
Revision        : c03112
Serial          : 10000000ec0ef79e
Model           : Raspberry Pi 4 Model B Rev 1.2
```

And does identify as aarch64
```
# uname -a
Linux allsky-dev 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

So, I'm now going to test with the older [64bit Buster](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip) version.

linuxkidd, 2022-01-25:
Well... I must have had my next update in a 'Preview' state and never saved it.. :(

I tested with a clean install of [64bit buster](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip) release:

- `apt update && apt -y full-upgrade`
- `apt -y install git`
- Ran the install instructions from the main page
- 100% success.

I'm kind of at a loss on why your compile ( or rather, linking ) is failing @Jonk2 ... My best guess is potentially corrupted libraries or possibly libraries in a partially upgraded / mixed version state.

I would recommend trying with a fresh install of your OS version / architecture of choice and see how it goes.

linuxkidd, 2022-02-02:
Hi @Jonk2 -- Wondering if you've had an opportunity to test the image reload / re-install?  Either the 32bit or 64bit should work fine with a fresh install based on my testing.

Thanks!

Jonk2, 2022-02-03:
Hi @linuxkidd no not yet - I'm going to install a fresh x64 on a new USB drive then immediately clone the latest allsky. I'll report back once I have done this.

Jonk2, 2022-02-04:
Hi @linuxkidd I've just installed a fresh x64 Raspberry PI OS on a blank USB drive and run the allsky clone / install.
It's all compiled fine with no errors.

I'm not in a position to re-install the x64 OS on the test PI that showed the original issue, and I would eventually want to update the live PI outside with the x64 OS...maybe I don't need to, but it shouldn't be restrictive.

I've attached a txt file of the console process so you can see everything that a fresh installation needed to do.

Where do I go from here, do I have any other options or are you able to 'fix' the installer to take whatever my issue is into consideration?

Do you want me to try again on the 'suspect' OS, now that allsky has been updated a couple of times since?

Are there any files outside of allsky that you want me to check or copy fresh versions over?

[Fresh RPi x64 and allsky installation.txt](https://github.com/thomasjacquin/allsky/files/8006492/Fresh.RPi.x64.and.allsky.installation.txt)

EricClaeys, 2022-02-04:
@Jonk2 It look slike the installation went perfectly.  Are there an problems using it?

Jonk2, 2022-02-05:
@EricClaeys so far, it's running in terms of an image showing on the admin page.

I attached the log however (this is all completely fresh, all I've changed is debug to 4). Notice when the software first starts, an 'ERROR: *** no cameras available ***" message is logged, then it goes on to detecting a camera and trying exposures and ultimately succeeding.

Thought you all might need to know, if this is an issue to look into.

Other than this test, is there anything I can do with my original suspect x64 OS? Any other files to check?

I appreciate that this probably isn't an allsky software issue, but if this can be resolved, surely it will make the software more robust, as I'm fairly certain I won't be the only one with a problem like this?!

Thanks

[allsky.log](https://github.com/thomasjacquin/allsky/files/8009028/allsky.log)

EricClaeys, 2022-02-06:
@Jonk2 this is Bullseye, correct?  And other than the "no cameras available" message, everything is working fine, right?

Someone else who I think also had Bullseye also had that error message that didn't appear to have any negative impact. I think it's a difference in Bullseye.  Would you please run `lsusb` and let me know what you get?
Also, could you add a `set -x` after line 87 in allsky.sh, then restart allsky and send the log file?  I think the ERROR message is coming from line 88 or 89 and adding `set -x` will confirm that.

In the meantime you can ignore that message.

Did you set your latitude and longitude?  It was daytime at 10 pm.

Jonk2, 2022-02-06:
@EricClaeys It's the latest 64bit version available from the raspberry pi imager. 
Yes on the face of it apart from the error message I didn't notice anything else immediately - as I say, I haven't changed anything apart from logging level = 4, as it was purely to test the compiling error I had on this particular PI with a 64bit PI-OS, presumably the same as the fresh install I'm testing for @linuxkidd.
I'll try your suggestion and get back to you when I can, hopefully today sometime.

Jonk2, 2022-02-06:
Ok, the lsusb shows this:

`Bus 002 Device 002: ID 03c3:178a ZWO ASI178MC`
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 001 Device 003: ID 0930:6544 Toshiba Corp. TransMemory-Mini / Kingston DataTraveler 2.0 Stick`
`Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub`
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

I've added `set -x` as requested...

[image]

New log file attached. (Spoiler alert!) the ERROR message still appears.

[allsky.log](https://github.com/thomasjacquin/allsky/files/8011112/allsky.log)

EricClaeys, 2022-02-06:
@Jonk2 Thanks.  I expected the ERROR to still be there; I am simply trying to determine where it is coming from.   My initial hunch was incorrect.
Would you please move the `set -x` line to be after the first line?
Thanks much.

Jonk2, 2022-02-06:
Ok - new log attached (cut to show log from changed file time).

[allsky.log](https://github.com/thomasjacquin/allsky/files/8011344/allsky.log)

EricClaeys, 2022-02-07:
@Jonk2 Perfect.  Thanks.  You can remove the `set -x`.  I know where the message is coming from (the command to check if you have a RPiHQ camera).  I will fix it when I get a chance.

If everything else is working related to this Issue, would you please close it?  Thanks

EricClaeys, 2022-02-11:
@Jonk2. If everything other than the incorrect ERROR meaasge is working related to this Issue, would you please close it? Thanks

Jonk2, 2022-02-14:
The issue of not compiling on the 64bit OS still remains, if file corruption or different versions of required files is the problem, is there a list of files required by allsky that I can check and overwrite with fresh versions?
I can't really reinstall the OS, I have other things running on that particular PI.

EricClaeys, 2022-02-25:
@linuxkidd Michael - any ideas?

@Jonk2 - FYI

EricClaeys, 2022-06-30:
@linuxkidd Michael - any ideas?

linuxkidd, 2022-07-17:
@Jonk2   Sorry it's taken me a while to get back to this..  
- Is this still an issue with a fresh 64bit OS install?  
- Back on Feb 4th, you reported:
> I've just installed a fresh x64 Raspberry PI OS on a blank USB drive and run the allsky clone / install.
> It's all compiled fine with no errors.

Does this not also work with a fresh flash of the 64bit OS image to sd card?

Jonk2, 2022-07-19:
@linuxkidd 

I don't know, I haven't tried as I don't use SD cards, sorry.

EricClaeys, 2022-07-20:
@Jonk2    It shouldn't matter if you use an SD card, USB drive, or something else.  Allsky doesn't know.
Were you able to get Allsky compiled and working on a 64-bit OS ?

EricClaeys, 2022-10-06:
@Jonk2    Jon, Were you able to get Allsky compiled and working on a 64-bit OS ?  I'm pretty sure other people have been able to get it to work.

closed 2022-10-31 by EricClaeys