AllskyTeam/allsky #907: Adding a time stamp to each image possible?

Wischi1, 2022-01-10:
Hi,
I must admit, that I´m a rookie, so forgive me if I ask for something simple.
I would like to know, if it is possible to add a time stamp to the image name e.g. image"hour""minute""second"?
Seconds are not mandotory, but hour and minute. Reason is that I have an internet page build by wordpress and there are only a few webcam plugins available. One which seems to work needs a time stamp in order to show a new image after refreshing.
Thanks in adavance
Sven

EricClaeys, 2022-01-10:
@Wischi1 Please see the Wiki page on overlays.  It explains what you can have overlaid on top of the image, including the date/time.
If this answers your question, please close the Issue.
And going forward,  if you have a question or want to request a feature, please use the Discussions link at the top of the page so we can keep the Issues list for actual bugs.
Thanks, and welcome to the Allsky community!

Wischi1, 2022-01-10:
Hi Eric,thanks for the info. Next time I will use discussion for raising questions. 
Maybe I was not clear enough, I mean is there chance, to add the time to the file name, not to the picture. This I already have.
Sven

EricClaeys, 2022-01-10:
@Wisch1 Which filename are you referring to?  The ones in allsky/images/DATE/*.jpg have the date and time in their names.  The newest version of Allsky, released a couple days ago, initially saves files with these names (it used to save a file called "image.jpg" which was then processed and copied to allsky/images/DATE/).

EricClaeys, 2022-01-12:
@Wischi1   ?

Wischi1, 2022-01-12:
Hey Eric, thanks for Information. I will try the new version and see, if it fits to what I expect. 
But in the meantime I installed a Wordpress plugin called webcamviewer and this plugin already solved my problem. It refreshes every minute the allsky image.jpg without having a time stamp.
Cheers Sven

EricClaeys, 2022-01-20:
@Wischi1 A plugin shouldn't be needed to do what you want, unless I don't understand what you're trying to accomplish.  But, if it's working, don't mess with it.
I'll close the Issue.

closed 2022-01-20 by EricClaeys