AllskyTeam/allsky #910: AllSky Administation Portal - No pictures visible

jimjackii, 2022-01-11:
Hi,

with the new AllSky Administation Portal for Allsky V0.8.3 there are a lot of issues with showing pictures.

There is no live image.

All days are listed under Images, with images / keograms / start trails etc.
But none of the images can be displayed.

I checked it with the Firefox debugger with F12 / Network.
A "404" always comes as a return code for all images.

These problems did not exist with the old version of the Administration Portal (V0.8.1).

Torakin9366, 2022-01-11:
Hallo Steffen, ich habe auch Probleme mit der neuen Software, deine Bilder werden jetzt im tmp Ordner angezeigt , zumindest das aktuelle Livebild.

jimjackii, 2022-01-11:
I found a solution for the pictures.

If I create a symbolic link "images" to the storage location of the images / keograms / startrails etc., the images can be displayed again.

The problem with the config files still persists and the live image is also not visible.

---

I also noticed that the entry for the AllSky website in the "config.js" file
imageName: "/current/tmp/image.jpg"
not working.

There I also created a symbolic link "image.jpg" to /home/pi/allsky/tmp/image.jpg "and
in config.js
imageName: "image.jpg"
registered.

jimjackii, 2022-01-11:
I could solve the problem with the config files with a symbolic link

/ var / www / html / current "->" / home / pi / allsky "fix"

fix it first.

AndreasLMeg, 2022-01-11:
> Hi,
> 
> with the new AllSky Administation Panel for Allsky V0.8.3 there are a lot of issues with showing pictures.
> 
> There is no live image.
> 
> All days are listed under "Images" with images / keograms / start trails etc. But none of the images can be displayed.
> 
> I checked it with the Firefox debugger with F12 / Network. A "404" always comes as a return code for all images.
> 
> Furthermore, no configuration files are loaded for editing under "Editor".
> 
> These problems did not exist with the old version of the Administration Panel (V0.8.1).

I made a new installation, but everything works for me - that's strange
1. running allsky stopped
2. directory allsky removed
3. content of  /etc/raspap/ removed
4.  git clone --recursive https://github.com/thomasjacquin/allsky.git
5. cd allsky; ./install.sh - reboot
6. cd allsky; sudo gui/install.sh - reboot
7. some settings (Mode mean, text, location, exposure, gain,...)

And this is the result:
[image]

EricClaeys, 2022-01-12:
@jimjackii As a test, can you restore imageName in config.js to `imageName: "/current/tmp/image.jpg"` and remove or rename your symbolic link, and in the browser get to the failed image load and do F12 and see what file is giving the 404?  It should list the URL.
You shouldn't need to do symbolic links - that's why we set up the lighttpd web server with an alias "/current/" to "/home/pi/allsky/".

Check if these lines are in /etc/lighttpd/lighttpd.conf:
~~~
alias.url		    = ("/current/" => "/home/pi/allsky/")
alias.url		    += ("/images/" => "/home/pi/allsky/images/")
~~~
They would be lines 27 and 28.

jimjackii, 2022-01-12:
@EricClaeys 
I've installed the actual version but the problem with no liveview-image and the image under /allsky persitsts.

I think there is a problem with generating the file "home/pi/allsky/tmp/image.jpg.
There is only the notification image from allsky start but no actual image from the camera.

I've attached the allsky.log and config.sh.

http://jimjackii.de/files/allsky/20220112/allsky.log
http://jimjackii.de/files/allsky/20220112/config.sh

The entries in allsky.log with "### --- SH --- ###" are entries from my own script files.

jimjackii, 2022-01-12:
@EricClaeys 
With your lighttpd settings I get a "404" error again.

jimjackii, 2022-01-12:
@EricClaeys 
I checked the log file again - I couldn't find error messages due to write errors in the tmp directory.

The file "image.jpg" is still not updated.
At the moment I'm doing this in my own script.

EricClaeys, 2022-01-12:
@jimjackii Are you saying allsky/tmp/image.jpg is never created?

jimjackii, 2022-01-12:
@EricClaeys 
Yes, only the fist notification image is created.

Torakin9366, 2022-01-12:
Und dieses Problem habe ich auch, hab es ja in einem anderen Beitrag schon beschrieben.
Wie wäre es denn, wenn ihr auch mal das macht was wir machen. Also eine neue SD karte nehmen, Raspbian (nicht Bullseye nehmen) installieren und dann auf die Seite gehen wo beschrieben wird, wie man Allsky installiert, und installiert es und schaut ob bei euch dann auch Probleme auftreten. Dann hat man zumindest einen Anfangspunkt. Was mir noch aufgefallen ist, wenn man sich auf der Github Seite die Allsky-Master runterläd, diese entpackt, umbenennt in allsky und sie dann ins original Verzeichnis legt, kommt am Ender der Installation ein Fehlermeldung das 2 Fehler aufgetreten sind.

EricClaeys, 2022-01-13:
@jimjackii this could be related to your permissions denied message on tmp.

jkoenig72, 2022-01-13:
> I could solve the problem with the config files and the live image with a symbolic link
> 
> "/var/www/html/current" -> "/home/pi/allsky"
> 
> fix it first.

I will try that and report back if it also works for me. Sounds reasonable.

jimjackii, 2022-01-14:
@EricClaeys 
I only mounted tmpfs with the user parameter, the directory has 777, so there are no longer any problems with access rights.
However, no "image.jpg" is generated.

As already written, I have built the creation of the "image.jpg" into my own script.
I call this at the end of saveImage.sh

http://jimjackii.de/files/allsky/20220114/saveImage.sh
http://jimjackii.de/files/allsky/20220114/_saveImage_script.sh

EricClaeys, 2022-01-14:
@jimjackii To help debug this, would you please add `set -x` to line 2 of scripts/saveImage.sh, then let it process one picture and then attach /var/log/allsky.log?
After that, remove the line you added.  That line will output each command it's running in saveImage.sh and should tell us why image.jpg isn't being created.
Thanks

jimjackii, 2022-01-14:
@EricClaeys 
Here the allsky.log with set +x in saveImage.sh.

http://jimjackii.de/files/allsky/20220114/allsky__with_set_x.log

EricClaeys, 2022-01-14:
@jimjackii That helped find the bug.  At night or when daytime images are being saved then image.jpg is created.  However it is NOT created if daytime images are not being saved, and that is a bug.
If you are comfortable editing scripts, could you move line 145 (starts with "mv") in saveImage.sh to after line 147 ("fi")?
Line 145 creates tmp/image.jpg.  And let me know if that fixes it?  It should be obvious during the day, as soon as you make the change images should appear.

jimjackii, 2022-01-14:
@EricClaeys 
Yes, that solves the problem, but the line must be after the upload part, otherwise there will be no image to upload to the external FTP storage.

The line must after line 196 ("fi").

EricClaeys, 2022-01-14:
@jimjackii Did it not work after line 147?
The old line 146 (IMAGE_TO_USE=$FINAL_FILE) sets IMAGE_TO_USE to the file in allsky/images/DATE, so moving the line to after 196 will move the images/DATE file, which isn't good.

jimjackii, 2022-01-14:
@EricClaeys 
No, there is the problem with upload.sh.

Error in allsky.log -> "*** upload.sh: ERROR: File to upload '/home/pi/allsky/tmp/image-20220114121302.jpg' not found!"

When you make an mv the actual image file "image-DATE.jpg" moves to "image.jpg" and now the file "image-DATE.jpg" isn't there anymore which needs by "upload.sh" to do a upload to the FTP space.

I think it's beter to do a cp -a instead a mv - then it can do after line 147.

jimjackii, 2022-01-20:
@EricClaeys @AndreasLMeg 
Hi, I think I can close this thread?

AndreasLMeg, 2022-01-20:
for me this bug is solved. I am currently not doing any research on this

EricClaeys, 2022-01-20:
Closing this issue since we implemented the change @jimjackii suggested and it appears to be working.

I'm still working on  version that does a "mv" in different places rather than "cp" to save an extra copy on the SD card.  When I have it working I'll do a PR for it.

closed 2022-01-20 by EricClaeys