AllskyTeam/allsky #911: Installation der neuen Software 0.8.3

Torakin9366, 2022-01-11:
Hier sind meine Probleme. Ich weiss garnicht wie oft ich gestern ein sauberes Bild von Raspbian installiert habe um Fehler bei der Allsky Software zu vermeiden. 
Installation läuft super ohne Probleme aber dann. Geh ich auf die WebUI und stelle dort im Editor und bei den Kameraeinstellungen alles ein und speichere dann alles ab, Startet Allsky neu und ich erhalte das Benachrichtigungsbild
"Allsky Software ist starting up" und dann  wars das es wird kein aktuelles Livebild mehr angezeigt.
Danach schau ich mir im config Ordner die config.sh und die settings_ZWO.json an, stelle ich fest, das was ich vorher in der WebUI eingestellt hatte, wurde in den Beiden Dateien nicht übernommen, Also hab ich es dort auch nochmal eingestellt. 
Nun das nächste Problem, im tmp Ordner wird das aktuelle live-Bild (img.jpg), wenn du Nachts aufnimmst, angezeigt und es wird eine Sequenz aufgenommen. Aber sobald die Nachtaufnahme zu Ende ist, werden alle 30 sek (je nachdem wie die Zeit eingestellt ist) die Bilder im Ordner als image-20220111100449.jpg abgespeichert. Warum, dann ist die Sd-Karte nach 2 Tagen voll und es ist dann kein Platz mehr für weitere Tage, es sei denn man stellt es so ein, das nach jedem Tag die Sequenz gelöscht wird.
Das selbe passiert auch wenn ich zuerst in der config.sh und settings_ZWO.json die Einstellungen vornehme und dann die WebUI besuche, es wird nichts übernommen und erhalte dann wieder den grünen Bildschirm mit den Hinweis.                                            
So das wars erstmal, ich hoffe ihr könnt mir und auch andere helfen.

jimjackii, 2022-01-11:
Hi,

I can also observe the filling of the tmp directory, "home / pi / allsky / tmp" with images "image-xxx.jpg".

At the moment the day mode is running with 60s images, but without auto exposure and auto gain.

The tmp directory was mounted as tmpfs with 200MB.

With regards, Steffen

AndreasLMeg, 2022-01-11:
@Torakin9366 
I made a new installation, but everything works for me - that's strange
Please send me /var/log/allsky.log and config/config.sh

AndreasLMeg, 2022-01-11:
> Hi,
> 
> I can also observe the filling of the tmp directory, "home / pi / allsky / tmp" with images "image-xxx.jpg".
> 
> At the moment the day mode is running with 60s images, but without auto exposure and auto gain.
> 
> The tmp directory was mounted as tmpfs with 200MB.
> 
> With regards, Steffen

This is what I can see:

```
pi@allsky:~/allsky/tmp $ ls -la
insgesamt 1792
drwxr-xr-x  2 pi pi    4096 Jän 11 19:10 .
drwxr-xr-x 14 pi pi    4096 Jän 11 17:24 ..
-rw-r--r--  1 pi pi     874 Jän 11 18:44 capture_args.txt
-rw-r--r--  1 pi pi       0 Jän 11 19:10 image-20220111191018.jpg~
-rw-r--r--  1 pi pi 1822296 Jän 11 19:10 image.jpg
```

please try ls -la ~/allsky/tmp/ and send me the output.

Torakin9366, 2022-01-11:
[allsky.log.zip](https://github.com/thomasjacquin/allsky/files/7849124/allsky.log.zip)
[config.sh.zip](https://github.com/thomasjacquin/allsky/files/7849125/config.sh.zip)

AndreasLMeg, 2022-01-11:
@Torakin9366 Was ist um 10:24 passiert ?
```

Jan 11 10:24:58 allsky allsky.sh[2985]:      ***** Stopping AllSky *****
Jan 11 10:25:10 allsky allsky.sh[831]:      ***** Starting AllSky *****
Jan 11 10:25:10 allsky allsky.sh[831]: ZWO Camera not found...
Jan 11 10:25:10 allsky allsky.sh[831]:   and no USB entry found for it.
Jan 11 10:25:10 allsky allsky.sh[831]:   Exiting.
Jan 11 10:25:10 allsky allsky.sh[831]:   If you have the 'uhubctl' command installed, add it to config.sh.
Jan 11 10:25:10 allsky allsky.sh[831]:   In the meantime, try running it to reset the USB bus.
Jan 11 12:19:08 allsky allsky.sh[854]:      ***** Starting AllSky *****
```

AndreasLMeg, 2022-01-11:
@EricClaeys:  cp: auf '/home/pi/allsky/tmp/image.jpg/notification-image.jpg' looks very strange, maybe a bug in copy_notification.sh ?

```
Jan 11 10:24:51 allsky allsky.sh[2985]:   > Sleeping 10.0 sec from histogram exposure
Jan 11 10:24:51 allsky allsky.sh[2985]:   > Saving DAY image 'image-20220111102451.jpg'
Jan 11 10:24:55 allsky allsky.sh[2985]: cp: auf '/home/pi/allsky/tmp/image.jpg/notification-image.jpg' konnte nicht zugegriffen werden: Ist kein Verzeichnis
Jan 11 10:24:55 allsky allsky.sh[2985]: mv: auf '/home/pi/allsky/tmp/image.jpg/image.jpg' konnte nicht zugegriffen werden: Ist kein Verzeichnis
Jan 11 10:24:55 allsky allsky.sh[2985]: copy_notification_image.sh: Uploading NotRunning.jpg
Jan 11 10:24:55 allsky allsky.sh[2985]: *** upload.sh: ERROR: File to upload '/home/pi/allsky/tmp/image.jpg/image.jpg' not found!
Jan 11 10:24:58 allsky allsky.sh[2985]:      ***** Stopping AllSky *****
```

AndreasLMeg, 2022-01-11:
@Torakin9366 have you changed this ?
```
# This avoids copying the image to the website.
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="/home/pi/allsky/tmp/image.jpg"
```

original:
```
# This avoids copying the image to the website.
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="current/tmp"
```

AndreasLMeg, 2022-01-11:
@EricClaeys - image look overexposed - why is gain so high, and getting higher.... ?

```
Jan 11 05:51:21 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:51:21   @ 50.0 sec
Jan 11 05:51:21 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055030.jpg' (above threshold: MEAN=97.283400, threshold = 90)
Jan 11 05:52:11 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055121.jpg'
Jan 11 05:52:11 allsky allsky.sh[1704]:   > Auto Gain value: 54
Jan 11 05:52:11 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure
Jan 11 05:52:11 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:52:11   @ 50.0 sec
Jan 11 05:52:12 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055121.jpg' (above threshold: MEAN=97.433900, threshold = 90)
Jan 11 05:53:02 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055211.jpg'
Jan 11 05:53:02 allsky allsky.sh[1704]:   > Auto Gain value: 55
Jan 11 05:53:02 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure
Jan 11 05:53:02 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:53:02   @ 50.0 sec
Jan 11 05:53:03 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055211.jpg' (above threshold: MEAN=96.958300, threshold = 90)
Jan 11 05:53:53 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055302.jpg'
Jan 11 05:53:53 allsky allsky.sh[1704]:   > Auto Gain value: 56
Jan 11 05:53:53 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure
Jan 11 05:53:53 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:53:53   @ 50.0 sec
Jan 11 05:53:54 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055302.jpg' (above threshold: MEAN=97.300400, threshold = 90)
Jan 11 05:54:44 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055353.jpg'
Jan 11 05:54:44 allsky allsky.sh[1704]:   > Auto Gain value: 57
Jan 11 05:54:44 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure
Jan 11 05:54:44 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:54:44   @ 50.0 sec
Jan 11 05:54:45 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055353.jpg' (above threshold: MEAN=98.158600, threshold = 90)
Jan 11 05:55:35 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055444.jpg'
Jan 11 05:55:35 allsky allsky.sh[1704]:   > Auto Gain value: 58
Jan 11 05:55:35 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure
Jan 11 05:55:35 allsky allsky.sh[1704]: STARTING EXPOSURE at: 2022-01-11 05:55:35   @ 50.0 sec
Jan 11 05:55:35 allsky allsky.sh[1704]: removeBadImages.sh: File is bad: removed 'image-baben-20220111055444.jpg' (above threshold: MEAN=98.116500, threshold = 90)
Jan 11 05:56:25 allsky allsky.sh[1704]:   > Saving NIGHT image 'image-baben-20220111055535.jpg'
Jan 11 05:56:25 allsky allsky.sh[1704]:   > Auto Gain value: 59
Jan 11 05:56:25 allsky allsky.sh[1704]:   > Sleeping 10.00 ms from manual exposure

```

EricClaeys, 2022-01-12:
@Torakin9366, @jimjackii,
We just posted a fix for the "tmp" folder filling up with images. It only happened when you were not saving daytime images.  The updated file is `scripts/saveImage.sh`.

@Torakin9366,
The "removeBadImages.sh: File is bad: ..." message are because you're using a manual 50 second nighttime exposure, which at 5:55 AM is probably too long and the image is almost pure white - I assume it's starting to get light then.  My only suggestion is to use auto exposure at night.

I'm pretty sure `IMG_DIR="/home/pi/allsky/tmp/image.jpg"` won't work because the Live View web server can't see files in /home/pi.  We get around that by having "current/tmp" be an alias for /home/pi/allsky/tmp.
I suggest setting `IMG_DIR="current/tmp"` and we'll work it from there.

When you go to "Camera Settings" in the WebUI, what's the configuration file name that appears at the top, above the settings?

jimjackii, 2022-01-12:
@EricClaeys 
As a test, I installed the deletion of the image files in the tmp folder in a separate script.

I will now install and test the current version.

Torakin9366, 2022-01-12:
[allsky.log.zip](https://github.com/thomasjacquin/allsky/files/7854471/allsky.log.zip)
[config.sh.zip](https://github.com/thomasjacquin/allsky/files/7854472/config.sh.zip)
[config.zip](https://github.com/thomasjacquin/allsky/files/7854483/config.zip)

So ich habe eben noch mal alles Neu installiert, Raspbian und die Allsky Software, mehr ist nicht drauf auf den Raspberry.
Der erste Abschnitt im Log Eintrag, ist der als ich die Datei geöffnet habe.
Der zweite ist, da hab ich die Datei erneut geöffnet und der letzte Abschnitt ist der, als ich auf dem Web-Browser des Raspberry die WebUI geöffnet habe. Das Startbild war wieder das grüne. Hab dann Kameraeinstellungen gemacht und abgespeichert. Das Benachrichtigungs Bild ist immer noch grün. Mehr habe ich bis jetzt noch gemacht. Wie ihr sehen könnt, hat das Programm 2 Dateien im config Ordner nicht installiert. Wenn ich jetzt aber den Allsky Ordner löschen würde und die Software nochmals installiere, sind die 2 fehlenden Dateien plötzlich wieder dort. Wie gesagt, der Rasp ist noch so wie ich ihn neu installiert habe und es wurden keine weiteren Anwendungen ausgeführt.

Torakin9366, 2022-01-12:
Hallo zusammen, ihr könnt den Beitrag schliessen. Kamera läuft jetzt. Habe mir nochmals die config.sh angeschaut und dort noch einpaar Einstellungen geändert. Der grüne Bildschirm erscheint weil die Bilder zu hell sind, dies habe ich abgeschaltet. Der rote Bildschirm erschein, wenn man die Benachrichtigungsbilder ausschaltet aber in der config.sh vergessen hat die schlechten Bilder löschen auf "false" zu stellen. ich glaub, ich habe euch genug Nerven gekostet bedanke mich aber trotzdem für eure Hilfe und hoffe einigen vielleicht geholfen zu haben.
Gruss Torsten

Torakin9366, 2022-01-19:
Hallo, mir ist bei der Installation der Software noch etwas aufgefallen. warum werden 2  verschiedene config Ordner erstellt?
Bei einer Installation sieht der Inhalt des config Ordners so aus ( siehe config ) und 5 min Später sieht sie bei einer Neuinstallation so aus ( siehe config-1 ) 
[config-1.zip](https://github.com/thomasjacquin/allsky/files/7895263/config-1.zip)
[config.zip](https://github.com/thomasjacquin/allsky/files/7895264/config.zip)
Wenn ich auf der WebUI Kamera Einstellungen vornehme und ich die Variante des config Ordners habe, werden die Einstellungen übernommen. Wenn ich aber die Variante mit config-1 habe muss, ich die Einstellungen in der jeweiligen Kamera einstellen. 
Habt ihr nicht die Möglichkeit die Alte Software 0.7.9 zum Download bereitzustellen? Denn diese hat tadellos funktioniert.
Schau mal auf meiner Webseite https://www.t-haucke.com/allsky/ und sag mir mal warum ein Bild hell aufgenommen wird und das andere dunkler. Auto-Exposure ist auf on eingestellt.
Danke
Gruss Torsten

jimjackii, 2022-01-19:
@Torakin9366 
It looks as if a backup folder "config-1" is created during a new installation and all important configuration files are stored in it.
With a new installation, it is easy to forget the files under "/etc/raspap".

When I reinstall, I always back up all the associated folders and reinstall from scratch.

The settings that you can make in the WebUI are written to the config files under "/etc/raspap".

Torakin9366, 2022-01-19:
Danke für Eure Hilfe

closed 2022-01-19 by Torakin9366

EricClaeys, 2022-01-19:
@Torakin9366  Thank you for reporting the issues you were having.  It's helping us improve Allsky for everyone.

The "2 different config folders" you mentioned, are actually the same folder (allsky/config) but the contents change when you install the WebUI.  The two "settings_xxx.json" files that are there when you initially install Allsky are the settings for the camera - gain, exposure, delay, etc.  Crazy people that don't install the WebUI have to edit those files manually to change settings.

When you install the WebUI, it **moves** those files to /etc/raspap, and then uses the files in /etc/raspsp - you'll see that location at the top of the "Camera Settings" page in the WebUI.

When I first started working on Allsky, installing the WebUI **copied** those two files instead of **moving** them, and it confused the heck out of me for a very long time.  I did what you did - changed one file and it had no effect.  I eventually changed the WebUI script to remove the files in allsky/config.
One of the things on our "to do" list is to combine all the config files (config.sh, ftp-settings.sh, settings_xxxx.json) into one big file and not to move it.

@jimjackii, The installation scripts are getting better about backing up and restoring things, but probably still have a ways to go.  For example, reinstalling the WebUI used to blow away the Allsky website if it was installed - say goodbye to all your saved keograms, startrails, etc.  Now it moves the website files to a temporary location, then reinstalls the WebUI, then restores the website files.
The goal of course, is to be able to update (even a full update) without the user needing to back up anything since it'll be automatic.

@Torakin9366, @jimjackii, Another thing on our "to do" list is to improve the auto-exposure in daytime and nighttime.  The version for RPiHQ cameras is pretty good and getting better.  In the next few months we'll port that algorithm to the ZWO cameras so hopefully the flickering at night will go away.  You are correct that using a manual exposure is one way to deal with it, but then you tend to get a lot of overexposed images at the transition from day to night and night to day.