AllskyTeam/allsky #914: Having major issues with this software…with Buster

ShedAstro, 2022-01-11:
Hi all,
I first installed the latest bulls eye, and would not work at all, so went back to buster a week ago, installed it all along with GUI, and all went fine, but it just now pleases itself if it starts up or not, or very often I get the error in red letters on the screen, I reckon it works about 3 times out of ten, the rest of the time it just says on the screen “all sky software starting up” but never does, no images recorded nothing…even when it does work and I have to change a setting it then never restarts….very disappointing 

link to log files, it’s long but three are some errors in there so may help
https://www.dropbox.com/s/y0hcq8ir0xqwier/allsky.log?dl=0

ShedAstro, 2022-01-11:
The latest version….

ShedAstro, 2022-01-11:
I have the log, and there is nothing on the wiki page about uploading  them, so have no idea why you mean….🤔🤔

ShedAstro, 2022-01-11:
This shows the errors I am getting all the time…

an 11 14:10:14 allsky allsky.sh[613]: XXXXXX == in IntHandle(), got signal 15
Jan 11 14:10:14 allsky allsky.sh[613]: XXXXXX == in IntHandle(), got signal 15
Jan 11 14:10:17 allsky allsky.sh[613]:      ***** Stopping AllSky *****
Jan 11 14:10:18 allsky allsky.sh[1488]:      ***** Starting AllSky *****
Jan 11 14:10:18 allsky allsky.sh[1488]: CAMERA: RPiHQ
Jan 11 14:10:24 allsky allsky.sh[1488]:  ********************************************
Jan 11 14:10:24 allsky allsky.sh[1488]:  *** Allsky Camera Software v0.8.3 | 2021 ***
Jan 11 14:10:24 allsky allsky.sh[1488]:  ********************************************
Jan 11 14:10:24 allsky allsky.sh[1488]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Jan 11 14:10:24 allsky allsky.sh[1488]: Add -h or --help for available options
Jan 11 14:10:24 allsky allsky.sh[1488]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Jan 11 14:10:24 allsky allsky.sh[1488]: Contributors:
Jan 11 14:10:24 allsky allsky.sh[1488]: -Knut Olav Klo
Jan 11 14:10:24 allsky allsky.sh[1488]: -Daniel Johnsen
Jan 11 14:10:24 allsky allsky.sh[1488]: -Robert Wagner
Jan 11 14:10:24 allsky allsky.sh[1488]: -Michael J. Kidd - <linuxkidd@gmail.com>
Jan 11 14:10:24 allsky allsky.sh[1488]: -Rob Musquetier
Jan 11 14:10:24 allsky allsky.sh[1488]: -Eric Claeys
Jan 11 14:10:24 allsky allsky.sh[1488]: -Andreas Lindinger
Jan 11 14:10:24 allsky allsky.sh[1488]:  Camera: Raspberry Pi HQ camera
Jan 11 14:10:24 allsky allsky.sh[1488]:   - Resolution: 4096x3040
Jan 11 14:10:24 allsky allsky.sh[1488]:   - Pixel Size: 1.55microns
Jan 11 14:10:24 allsky allsky.sh[1488]:   - Supported Bins: 1x, 2x and 3x
Jan 11 14:10:24 allsky allsky.sh[1488]: Capture Settings:
Jan 11 14:10:24 allsky allsky.sh[1488]:  Command: raspistill
Jan 11 14:10:24 allsky allsky.sh[1488]:  Image Type: RGB24
Jan 11 14:10:24 allsky allsky.sh[1488]:  Resolution (before any binning): 4096x3040
Jan 11 14:10:24 allsky allsky.sh[1488]:  Quality: 95
Jan 11 14:10:24 allsky allsky.sh[1488]:  Daytime capture: 1 (yes)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Exposure (night): 20000ms
Jan 11 14:10:24 allsky allsky.sh[1488]:  Auto Exposure (night): 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Gain (night): 1.00
Jan 11 14:10:24 allsky allsky.sh[1488]:  Auto Gain (night): 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Brightness (day): 50
Jan 11 14:10:24 allsky allsky.sh[1488]:  Brightness (night): 50
Jan 11 14:10:24 allsky allsky.sh[1488]:  Saturation: 50.0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Auto White Balance: 1 (yes)
Jan 11 14:10:24 allsky allsky.sh[1488]:  WB Red: 2.50
Jan 11 14:10:24 allsky allsky.sh[1488]:  WB Blue: 2.00
Jan 11 14:10:24 allsky allsky.sh[1488]:  Binning (day): 2
Jan 11 14:10:24 allsky allsky.sh[1488]:  Binning (night): 2
Jan 11 14:10:24 allsky allsky.sh[1488]:  Delay (day): 15000ms
Jan 11 14:10:24 allsky allsky.sh[1488]:  Delay (night): 10ms
Jan 11 14:10:24 allsky allsky.sh[1488]:  Text Overlay: Backyard Allsky  52.61N - 1.51W
Jan 11 14:10:24 allsky allsky.sh[1488]:  Text Position: 15px left, 75px top
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Name:  0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Color: 255, 255, 255
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Background Color: 0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Small Font Color: 0, 0, 255
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Line Type: 16
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Size: 7.0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Font Line: 1
Jan 11 14:10:24 allsky allsky.sh[1488]:  Outline Font : 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Rotation: 0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Flip Image: 0
Jan 11 14:10:24 allsky allsky.sh[1488]:  Filename: image.jpg
Jan 11 14:10:24 allsky allsky.sh[1488]:  Filename Save Directory: /home/pi/allsky/tmp
Jan 11 14:10:24 allsky allsky.sh[1488]:  Latitude: 60.7N
Jan 11 14:10:24 allsky allsky.sh[1488]:  Longitude: 135.05W
Jan 11 14:10:24 allsky allsky.sh[1488]:  Sun Elevation: -6
Jan 11 14:10:24 allsky allsky.sh[1488]:  Locale: en_US.UTF-8
Jan 11 14:10:24 allsky allsky.sh[1488]:  Notification Images: 1 (yes)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Show Time: 1 (yes) (format: %d/%m/%Y %H:%M:%S)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Show Exposure: 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Show Gain: 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Show Brightness: 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Show Focus Metric: 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]:  Preview: 0 (no)
[... 2 lines cut ...]
Jan 11 14:10:24 allsky allsky.sh[1488]:  On TTY: No
Jan 11 14:10:24 allsky allsky.sh[1488]:  Mode Mean: 0 (no)
Jan 11 14:10:24 allsky allsky.sh[1488]: ==========
Jan 11 14:10:24 allsky allsky.sh[1488]: === Starting nighttime capture ===
Jan 11 14:10:24 allsky allsky.sh[1488]: ==========
Jan 11 14:10:24 allsky allsky.sh[1488]: Stop the allsky service to end this process.
Jan 11 14:10:24 allsky allsky.sh[1488]: STARTING EXPOSURE at: 2022-01-11 14:10:24   @ 20.0 sec
Jan 11 14:10:46 allsky allsky.sh[1488]: Sleeping 0.0 seconds...
Jan 11 14:10:46 allsky allsky.sh[1488]: STARTING EXPOSURE at: 2022-01-11 14:10:46   @ 20.0 sec
Jan 11 14:10:47 allsky allsky.sh[1488]: removeBadImages.sh: File is bad: removed 'image-20220111141024.jpg' (above threshold: MEAN=99.901500, threshold = 90)
Jan 11 14:11:08 allsky allsky.sh[1488]: Sleeping 0.0 seconds...
[... cut ...]
Jan 11 14:12:14 allsky allsky.sh[1488]: Sleeping 0.0 seconds...
Jan 11 14:12:14 allsky allsky.sh[1488]: STARTING EXPOSURE at: 2022-01-11 14:12:14   @ 20.0 sec
Jan 11 14:12:15 allsky allsky.sh[1488]: removeBadImages.sh: File is bad: removed 'image-20220111141152.jpg' (above threshold: MEAN=99.901500, threshold = 90)
Jan 11 14:12:22 allsky allsky.sh[1488]: XXXXXX == in IntHandle(), got signal 15
Jan 11 14:12:22 allsky allsky.sh[1488]: XXXXXX == in IntHandle(), got signal 15
Jan 11 14:12:25 allsky allsky.sh[1488]:      ***** Stopping AllSky *****

ShedAstro, 2022-01-11:
This is another error 

Jan 11 13:25:33 allsky allsky.sh[4929]: === Starting daytime capture ===
Jan 11 13:25:33 allsky allsky.sh[4929]: ==========
Jan 11 13:25:33 allsky allsky.sh[4929]: Stop the allsky service to end this process.
Jan 11 13:25:33 allsky allsky.sh[4929]: STARTING EXPOSURE at: 2022-01-11 13:25:33   @ 32.00 ms (0.03 sec)
Jan 11 13:25:44 allsky allsky.sh[4929]: /home/pi/allsky/allsky.sh: line 225:  5011 Killed                  "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Jan 11 13:25:44 allsky allsky.sh[4929]: 'capture_RPiHQ' exited with RETCODE=137
Jan 11 13:25:44 allsky allsky.sh[4929]: ***
Jan 11 13:25:44 allsky allsky.sh[4929]: *** After fixing, restart the allsky service. ***
Jan 11 13:25:44 allsky allsky.sh[4929]: ***
Jan 11 13:26:16 allsky allsky.sh[5050]:      ***** Starting AllSky *****

AndreasLMeg, 2022-01-11:
@EricClaeys: mean calculation wrong ?
```
Jan 11 17:26:36 allsky allsky.sh[3333]: + MEAN=0.00217263
Jan 11 17:26:36 allsky allsky.sh[3333]: + egrep -q 'Huffman|Bogus|Corrupt|Invalid|Trunc|Missing|insufficient image data|no decode delegate|no images defined' /home/pi/allsky/tmp/badError.txt
Jan 11 17:26:36 allsky allsky.sh[3333]: ++ echo '0.00217263 * 100'
Jan 11 17:26:36 allsky allsky.sh[3333]: ++ bc
Jan 11 17:26:36 allsky allsky.sh[3333]: + MEAN=.21726300
Jan 11 17:26:36 allsky allsky.sh[3333]: + MSG=
Jan 11 17:26:36 allsky allsky.sh[3333]: + '[' 90 '!=' 0 ']'
Jan 11 17:26:36 allsky allsky.sh[3333]: ++ echo '.21726300 > 90'
Jan 11 17:26:36 allsky allsky.sh[3333]: ++ bc
Jan 11 17:26:36 allsky allsky.sh[3333]: + x=0
Jan 11 17:26:36 allsky allsky.sh[3333]: + '[' 0 -eq 1 ']'
Jan 11 17:26:36 allsky allsky.sh[3333]: + '[' false = true ']'
Jan 11 17:26:36 allsky allsky.sh[3333]: + '[' '' = '' -a 1 '!=' 0 ']'
Jan 11 17:26:36 allsky allsky.sh[3333]: ++ echo '.21726300 < 1'

```

ShedAstro, 2022-01-11:
Looks that way…if that is what you are asking…?

AndreasLMeg, 2022-01-11:
no the calculation is ok, but in my case the low value 1 is to high.
e.g. LOW=0.001 works fine

```
Jan 11 18:05:12 allsky allsky.sh[3333]: + MEAN=0.00779175
Jan 11 18:05:12 allsky allsky.sh[3333]: + egrep -q 'Huffman|Bogus|Corrupt|Invalid|Trunc|Missing|insufficient image data|no decode delegate|no images defined' /home/pi/allsky/tmp/badError.txt
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ echo '0.00779175 * 100'
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ bc
Jan 11 18:05:12 allsky allsky.sh[3333]: + MEAN=.77917500
Jan 11 18:05:12 allsky allsky.sh[3333]: + MSG=
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' 90 '!=' 0 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ echo '.77917500 > 90'
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ bc
Jan 11 18:05:12 allsky allsky.sh[3333]: + x=0
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' 0 -eq 1 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' false = true ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + LOW=0.001
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' '' = '' -a 0.001 '!=' 0 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ echo '.77917500 < 0.001'
Jan 11 18:05:12 allsky allsky.sh[3333]: ++ bc
Jan 11 18:05:12 allsky allsky.sh[3333]: + x=0
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' 0 -eq 1 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' false = true -a '' = '' ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' false = true -a '' = '' -a '' '!=' '' ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' '' '!=' '' ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' 0 -eq 0 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' image-20220111180505.jpg = '' ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + rm -f /home/pi/allsky/tmp/removeBadImages.log
Jan 11 18:05:12 allsky allsky.sh[3333]: + rm -f /home/pi/allsky/tmp/badError.txt
Jan 11 18:05:12 allsky allsky.sh[3333]: + '[' 0 -eq 0 ']'
Jan 11 18:05:12 allsky allsky.sh[3333]: + exit 0

```

AndreasLMeg, 2022-01-11:
the same, but with HIGH is in your case. the mean is to high
above threshold: MEAN=99.901500, threshold = 90

picture is very overexposed

my opinion, this is not a bug.

ShedAstro, 2022-01-11:
> the same, but with HIGH is in your case. the mean is to high above threshold: MEAN=99.901500, threshold = 90
> 
> picture is very overexposed
> 
> my opinion, this is not a bug.

Makes no difference if I change it to 100 still the same, does not work, had it working for about 3 mins, then stopped again, with the same error, this is the version loaded yesterday so it’s very latest, very very buggy and intermittent…even re booting Rpi zero makes no difference…☹️

AndreasLMeg, 2022-01-11:
let's make it step by step:

1. set the correct location (you ar using the default values)
Jan 11 14:10:24 allsky allsky.sh[1488]: Latitude: 60.7N
Jan 11 14:10:24 allsky allsky.sh[1488]: Longitude: 135.05W

2. raspi pi time is correct ?

after this send me the logging (start + 1 capture)

ShedAstro, 2022-01-11:
This is all I get on screen, but it never saves or shows any images and get the errors above in the log all the time, and this screen does not change
[image]

AndreasLMeg, 2022-01-11:
> Makes no difference if I change it to 100 still the same, does not work, had it working for about 3 mins, then stopped again, with the same error, this is the version loaded yesterday so it’s very latest, very very buggy and intermittent…even re booting Rpi zero makes no difference…

we have not yet recognized any errors - except perhaps imprecise instructions or incomprehensible messages

ShedAstro, 2022-01-11:
Well very sorry for wasting your time, I am not a coder or that computer savvy, and sorry for that too that I don’t have the correct terminology…
I have got it to work but will only work when the rpiHQ camera is binned 2x2 anything else and it fails all the time….
Although you are probably mot interested….

AndreasLMeg, 2022-01-12:
@ShedAstro:
if I didn't care, I wouldn't have answered your questions.
And I almost started at level 0 a year ago. And that it doesn't work straight away is the intention with Raspberry Pi projects. It's supposed to be a learning computer.

So let's start from the begining:
1. Describe your system (Raspi, OS, Camera, ...)?
2. Is this a new installation or an update?
3. What exactly does not work?
4. The log file is usually very important for an error analysis (you can find it under /var/log/allsky.log)
5. Have you seen our wiki yet? (e.g.  https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues)

AndreasLMeg, 2022-01-12:
I close this issue, and you make a new....

EricClaeys, 2022-01-12:
@ShedAstro
We're trying to help.  Please hang in their with us.  Let's look at things one at a time.

In your first log, it's taking nighttime pictures but it's only 2 pm, so a 20 second exposure is almost pure white, hence the "removeBadImages.sh: File is bad: ..." message - you have `REMOVE_BAD_IMAGES="true"` which is the recommendation.  You could set it to "false" but that doesn't fix the issue that your Pi considers 2 pm to be nighttime.

When Allsky starts it displays the "Allsky is starting up" message, and leaves it there until the first image appears.  If no image appears because they are being deleted for being too bright, that "Allsky is starting up" message will stay a long time - probable until 7 or 8 pm when a 20 second exposure gives a decent picture.

So, we need to determine why your Pi thinks nighttime is at 2 pm.
[This Wiki page](https://github.com/thomasjacquin/allsky/wiki/Angle-and-sunrise---sunset) describes the "Angle" setting.
Please run this to determine when your Pi thinks daytime and nighttime are, based on your latitude, longitude, and Angle:
`( cd ~/allsky; source config/config.sh; ./sunwait list angle $(jq -r ".angle,.latitude,.longitude" ${CAMERA_SETTINGS}) )`

According to the log file you uploaded to Dropbox, your Angle is -9, latitude is 52.57N, and Longitude is 4.7E.  **Are those correct?**

ShedAstro, 2022-01-12:
> @ShedAstro We're trying to help. Please hang in their with us. Let's look at things one at a time.
>
> In your first log, it's taking nighttime pictures but it's only 2 pm, so a 20 second exposure is [...]

So where is the thread gone why closed….
I have now set the correct long and lat, and it seems to work, but it will only work with 2x2 bin set, if I use 1x1 then I get red error message on screen and asks to check the log…
Also what time is the camera supposed to start and stop taking images at night, and can it be set to just come on at night as I don’t want in the daytime…
Thanks

ShedAstro, 2022-01-12:
> I close this issue, and you make a new....

What did you close and not give me time to get up in the morning and answer, you are not helpful at all, and have no patience whatsoever…

ShedAstro, 2022-01-12:
> @ShedAstro: if I didn't care, I wouldn't have answered your questions. And I almost started at level 0 a year ago. And that it doesn't work straight away is the intention with Raspberry Pi projects. [...]

If you had actually read the posts I made all these questions are answered, so what not read them…..before making judgements you complete idiot…..😮😡

AndreasLMeg, 2022-01-12:
@ShedAstro: Sorry, I thought this issue is actually a discussion and not a bug.

closed 2022-01-19 by ShedAstro

linuxkidd, 2022-01-21:
NOTE @ShedAstro  -- let's please be respectful of each other.   We're all volunteers working on a project, and trying to help everyone with any issues that arise.  It's easy to make a mistake - we're all human, especially when switching context while working across different issues.