AllskyTeam/allsky #915: WebUI can’t load image.jpg

DevilJack, 2022-01-12:
Today, I did a fresh build of the Allsky camera on a RPi 4. At first, it didn’t appear to be working but once I changed IMG_DIR="/home/pi/allsky/tmp", it worked and I was getting the current image in IMG_DIR/image.jpg and time stamped copies saved to the image directory. Now, it’s only Live View in the WebUI that’s not working. It displays a black square instead of an image. So, I’m looking for WebUI configuration setting, like a global variable, that needs to be set.

Any suggestions would be much appreciated! Thanks -JG

AndreasLMeg, 2022-01-12:
@EricClaeys: problem with update of allsky or WebUI (old data and configs) ? browser cache ?

EricClaeys, 2022-01-12:
@DevilJack `IMG_DIR="current/tmp"` didn't work?
Please check that these lines:
~~~
alias.url		    = ("/current/" => "/home/pi/allsky/")
alias.url		    += ("/images/" => "/home/pi/allsky/images/")
~~~
are in /etc/lighttpd/lighttpd.conf.  If not, I suggest reinstalling the WebUI: `cd allsky;  sudo gui/install.sh`

You can also press F12 in the "Live View" page and on the right side, click on "Network" (you may need to first click the >>>).  Then refresh your browser.  It should show what file it's trying to open for the image - it'll likely be in red since it failed.
Let me know.

jimjackii, 2022-01-12:
@EricClaeys 
I had already written something in the other post - could it have rights problems if the tmp directory is mounted as tmpfs?

https://github.com/thomasjacquin/allsky/issues/910

EricClaeys, 2022-01-12:
@jimjackii that's possible but you would see error messages in the log file when the capture program tried to write the file.  An easy test is to look in allsky/tmp while the software is running and see if image.jpf is updat

DevilJack, 2022-01-12:
The image.jpg file is being writtten to the temp allsky/tmp and later renamed and copied to allsky/images. That appears to be working well. The directory permissions look good and WebUI can access all the image lies in Allsky/images but not the current image.jpg in allsky/tmp.

EricClaeys, 2022-01-12:
@DevilJack Would you please try the stuff I mentioned 11 hours ago in my last reply to you, and report back?  Thanks

DevilJack, 2022-01-12:
I checked the alias.url entries in the /etc/lighttpd/lighttpd.config file and they were correct. I also reran the WebUI install script and the results were the same.

DevilJack, 2022-01-12:
I’m in the process of doing a fresh install and seeing if that works.

DevilJack, 2022-01-12:
New install duplicated my issue with Live View.

DevilJack, 2022-01-12:
Any thoughts?

jkoenig72, 2022-01-13:
Hi,

just have the same issue. Web UI does not show the live image. Had to leave but will investigate later today. What I noted that in the website source it’s named current/liveview-image.jpg

Does this file exists?

Regards
Jörg

EricClaeys, 2022-01-14:
@jkoenig72, @DevilJack 
Version 0.8.3 uses the filename you specify in the WebUI.  For almost everyone that is image.jpg.
When you said "the website source" what exactly are you referring to?
With 0.8.3 it's important to follow the instructions on the Wiki page dealing with upgrading from prior versions - I believe it's the second Wiki page.  It mentions needing to update the "imageName" vatiable in /var/www/html/allsky/config.js if you have the Allsky Website installed.  The Live View page of the WebUI looks in the directory specified by the IMG_DIR variable in config.sh, which should be "current/tmp".

Conrad414, 2022-01-14:
I also have no liveview image in 8.3 (ok in 8.1) in the WebGui but only during the day - it is set to show but not record daytime images. Liveview images are shown at night, and the last one persists through the day. Day or night the new image appears briefly with date in current/tmp, but image.jpg is not changed during the day. A daytime reboot shows "Allsky starting" until night recording begins.

EricClaeys, 2022-01-27:
@DevilJack Are things working ok now?

Conrad414, 2022-01-27:
Good morning Eric,

My apologies, I have been out of circulation for a while.

I could not get your "saveImage" script to work properly for me so I have
slightly modified it as per the attached file, between lines 140 and 150.
It seemed to have something to do with the symbolic link names, which I
tested by trial and error.

This now works as expected - i.e. it shows an image preview with "daytime
capture" on and "daytime save" on or off, and with "daytime save" on, it
places both day and night images in the "images" directory, from where a
timelapse, startrails or keogram are generated at "end-of-night". Images in
the "tmp" directory are now properly erased.

I am a novice script writer and have not comprehensively tested this
modified script. I have not properly implemented daytime capture -  it
captures whether on or off. I have also not set up "upload to web", so not
tested that. I apologise in advance for any silly errors!

I have not yet collected a new log file, will do so in due course if you
would like to see it.

Best regards,

Conrad Walker - Johannesburg, South Africa.

EricClaeys, 2022-01-27:
@Conrad414 would you please attach your modified saveImage.sh and config.sh files?  Where are the sumbolic links?

Conrad414, 2022-01-27:
I have attached saveImage.sh and config.sh to this email (changed to .txt to send)

Any changes in config were through the web gui. The only file I changed was the saveImage.sh, lines 140 to 150. I'm afraid I don't know how to send you any symbolic links -:(.

[config_sh.txt](https://github.com/thomasjacquin/allsky/files/7949408/config_sh.txt)
[saveImage_sh.txt](https://github.com/thomasjacquin/allsky/files/7949410/saveImage_sh.txt)

EricClaeys, 2022-01-30:
@Conrad414 Try the newest saveImage.sh, just updated a few days ago.  Also grab darkCapture.sh and darkSubtract.sh (all three should be updated together).
What symbolic link names are you referring to?  Is your allsky/tmp directory a tmpfs file system (in memory) or a symbolic link to another fileystem?
The current saveImage.sh is working fine for other people, although I suspect they didn't do anything special with allsky/tmp.

Conrad414, 2022-02-01:
Hello Eric,

I have replaced the saveImage.sh, darkCapture.sh and darkSubtract.sh  files
with your new ones.

I only ran the system for a short time during daylight but there was still
no preview image, only the ALLSKY splash. In addition, images were being
saved in the allsky/tmp directory. I did not have daylight save enabled so
did not expect them in the image/date directory, that would presumably
happen after sunset.

I only refer to symbolic links as you asked about them in an earlier post.

My tmp directory is not a tmpfs, it is exactly as installed by the allsky
install.

I am not using the upload to web module - perhaps this makes a difference.

Anyway, the saveImage.sh I sent you is working fine for me at present, I
will at some point try a complete re-install of Allsky using your latest
scripts and report how that works.

We have had weeks of overcast skies but it has now cleared and I hope to
capture some good images!

Best regards,

Conrad

EricClaeys, 2022-05-27:
@Conrad414    Conrad, are you still having this problem?  If so, please try installing the latest software and see if that fixes your problem.

Conrad414, 2022-06-03:
Good morning Eric,

Apologies for not responding sooner.

I have not yet had an opportunity to install and test the newest version of
Allsky.

The older version with the changes I made is still working OK.

As soon as I get a chance to test the new version, I will do so.

Best regards,

Conrad Walker

EricClaeys, 2022-06-03:
@Conrad414   Conrad, no problem.  When you do the upgrade, please follow the instructions in the Wiki for upgrading the software.  It's # 9.  That way you can revert to your current version if needed, although I'd prefer to fix whatever's broken.

If you're not able to upgrade in the next week or two, I'd like to close the Issue since it was opened about 6 months ago.  At that point if you still have the problem a new Issue can be opened.

Conrad414, 2022-06-05:
Good morning Eric,

Thanks for keeping me posted. I am traveling overseas until mid July and
can't upgrade until then, so please close the query. I will get back to you
later with an update and if I encounter any problems after the upgrade.

Best regards,

Conrad

DevilJack, 2022-06-05:
I haven’t had time to work on it over the last several weeks.  I’m look to do a new build over the next week or so.

Please, close the current issue.

Thanks

JG

EricClaeys, 2022-06-05:
Closing Issue with user's agreement.
After he has time to do a reinstall, if he still has a problem he will submit a new Issue.

closed 2022-06-05 by EricClaeys

Conrad414, 2022-08-08:
Good morning Eric,

I have at last had a chance to install the latest version of Allsky and it
is now working as expected.

However, I am struggling with the auto exposure feature, especially during
daylight. Although on auto, it only seems to use the minimum value setting
as per manual. I am using the Rpi HQ camera with a 2.1mm lens.

I will continue exploring to see if I am doing something wrong.

Best regards,

Conrad

EricClaeys, 2022-08-08:
@Conrad414  there's a bug that it ignores the **Maximum** value and uses the **Manual Exposure** as the maximum.  Just set **Manual Exposure** to the maximum.
This is fixed in the next release.