AllskyTeam/allsky #916: AllSky V0.8.3 (new) - no timelapse

jimjackii, 2022-01-12:
Hi,

sorry for my new post.

Today, with the updated version V0.8.3, another problem occurred - no timelapse could be created.

Error logfile:
http://jimjackii.de/files/allsky/20220112/timelapseTMP.txt

In config.sh these the timelaps parameters
TIMELAPSEWIDTH=1296
TIMELAPSEHEIGHT=972
TIMELAPSE_BITRATE="3500k"
FPS=5
VCODEC="libx264"
FFLOG="warning"
KEEP_SEQUENCE="false"
TIMELAPSE_EXTRA_PARAMETERS=""
UPLOAD_VIDEO="true"

jimjackii, 2022-01-13:
Good Morning,

today I found this in the allsky.log.
http://jimjackii.de/files/allsky/20220113/screenshot_timelapse.jpg

EricClaeys, 2022-01-13:
@jimjackii can you do a `ls -ld ~/allsky/tmp` ?
Also try `touch ~/allsky/tmp/test` and see if that works.
Is your tmp directory a tmpfs?  Did you make pi the owner?

jimjackii, 2022-01-13:
@EricClaeys 
yes, the tmp directory is a tmpfs, created in fstab as described in the notes.

pi@allsky:~ $ ls -ld ~/allsky/tmp
drwxrwxrwt 3 root root 200 Jan 13 09:44 /home/pi/allsky/tmp
User root is ownr of the tmp directory

pi@allsky:~ $ touch ~/allsky/tmp/test
was ok. The test file was created.

jimjackii, 2022-01-13:
I've tested old pictures to generate manually the mp4 file. Also with an old timelapse.sh from V0.8.1.
None of them was successful!

That's a strange behavior...

ShedAstro, 2022-01-13:
I too used my all sky camera last night with the latest software and got all images, a keogram and star trail image, but no time lapse, all times lapse settings are default…🤔

ShedAstro, 2022-01-13:
If I try to open my log on the pi I get this 
[image]

ShedAstro, 2022-01-13:
> @jimjackii can you do a `ls -ld ~/allsky/tmp` ?
> Also try `touch ~/allsky/tmp/test` and see if that works.
> Is your tmp directory a tmpfs? Did you make pi the owner?

I also ran the two above and both fine yesterday file was created, all images are in TMP folder

ShedAstro, 2022-01-13:
I also have two sequence folders for last two days in TMP folder, full of the images…I assume this is correct…?
[image]

jimjackii, 2022-01-13:
@ShedAstro 
Yes, that's correct. These are not images but symbolic links to the regular files. They need no space.

Usually the sequence folders are deleted after the videos have been created without errors.

ShedAstro, 2022-01-13:
Ok, thanks, so why am I not getting a time lapse and not able to open my log file as shown in forum on git, either, also while looking in the TMP folder there are a couple of files just flashing up for a second or so then disappear.... Sent from Mail for Windows From: SteffenSent: 13 January 2022 10:25To: thomasjacquin/allskyCc: ShedAstro; MentionSubject: Re: [thomasjacquin/allsky] AllSky V0.8.3 (new) - no timelapse (Issue ***@***.***, that's correct. These are not images but symbolic links to the regular files. They need no space.Usually the sequence folders are deleted after the videos have been created without errors.

	
        
		Virus-free. www.avg.com

jimjackii, 2022-01-13:
@ShedAstro 
That with the appearing files in the temp folder is ok too. These are the temporarily created image files, they will also be deleted after the final image has been created.

ShedAstro, 2022-01-13:
Jan 13 00:00:06 allsky rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="348" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Jan 13 00:00:06 allsky systemd[1]: man-db.service: Succeeded.
Jan 13 00:00:06 allsky systemd[1]: Started Daily man-db regeneration.
Jan 13 00:00:11 allsky allsky.sh[26216]: XXXXXX == in IntHandle(), got signal 15
Jan 13 00:00:11 allsky systemd[1]: Stopping All Sky Camera...
Jan 13 00:00:11 allsky systemd[1]: allsky.service: Main process exited, code=killed, status=15/TERM
Jan 13 00:00:12 allsky allsky.sh[26216]: XXXXXX == in IntHandle(), got signal 15
Jan 13 00:00:15 allsky allsky.sh[26216]:      ***** Stopping AllSky *****
Jan 13 00:00:18 allsky systemd[1]: allsky.service: Succeeded.
Jan 13 00:00:18 allsky systemd[1]: Stopped All Sky Camera.
Jan 13 00:00:18 allsky systemd[1]: Started All Sky Camera.
Jan 13 00:00:18 allsky allsky.sh[1514]:      ***** Starting AllSky *****
Jan 13 00:00:18 allsky allsky.sh[1514]: CAMERA: RPiHQ
Jan 13 00:00:24 allsky allsky.sh[1514]:  ********************************************
Jan 13 00:00:24 allsky allsky.sh[1514]:  *** Allsky Camera Software v0.8.3 | 2021 ***
Jan 13 00:00:24 allsky allsky.sh[1514]:  ********************************************
Jan 13 00:00:24 allsky allsky.sh[1514]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Jan 13 00:00:24 allsky allsky.sh[1514]: Add -h or --help for available options
Jan 13 00:00:24 allsky allsky.sh[1514]: Author: Thomas Jacquin - ***@***.***>
Jan 13 00:00:24 allsky allsky.sh[1514]: Contributors:
Jan 13 00:00:24 allsky allsky.sh[1514]: -Knut Olav Klo
Jan 13 00:00:24 allsky allsky.sh[1514]: -Daniel Johnsen
Jan 13 00:00:24 allsky allsky.sh[1514]: -Robert Wagner
Jan 13 00:00:24 allsky allsky.sh[1514]: -Michael J. Kidd - ***@***.***>
Jan 13 00:00:24 allsky allsky.sh[1514]: -Rob Musquetier
Jan 13 00:00:24 allsky allsky.sh[1514]: -Eric Claeys
Jan 13 00:00:24 allsky allsky.sh[1514]: -Andreas Lindinger
Jan 13 00:00:24 allsky allsky.sh[1514]:  Camera: Raspberry Pi HQ camera
Jan 13 00:00:24 allsky allsky.sh[1514]:   - Resolution: 4096x3040
Jan 13 00:00:24 allsky allsky.sh[1514]:   - Pixel Size: 1.55microns
Jan 13 00:00:24 allsky allsky.sh[1514]:   - Supported Bins: 1x, 2x and 3x
Jan 13 00:00:24 allsky allsky.sh[1514]: Capture Settings:
Jan 13 00:00:24 allsky allsky.sh[1514]:  Command: raspistill
Jan 13 00:00:24 allsky allsky.sh[1514]:  Image Type: RGB24
Jan 13 00:00:24 allsky allsky.sh[1514]:  Resolution (before any binning): 4096x3040
Jan 13 00:00:24 allsky allsky.sh[1514]:  Quality: 95
Jan 13 00:00:24 allsky allsky.sh[1514]:  Daytime capture: 1 (yes)
Jan 13 00:00:24 allsky allsky.sh[1514]:  Exposure (night): 25000ms
Jan 13 00:00:24 allsky allsky.sh[1514]:  Auto Exposure (night): 0 (no)
Jan 13 00:00:24 allsky allsky.sh[1514]:  Gain (night): 3.00
Jan 13 00:00:24 allsky allsky.sh[1514]:  Auto Gain (night): 0 (no)
Jan 13 00:00:24 allsky allsky.sh[1514]:  Brightness (day): 55
Jan 13 00:00:24 allsky allsky.sh[1514]:  Brightness (night): 50
Jan 13 00:00:24 allsky allsky.sh[1514]:  Saturation: -25.0
Jan 13 00:00:24 allsky allsky.sh[1514]:  Auto White Balance: 0 (no)
Jan 13 00:00:24 allsky allsky.sh[1514]:  WB Red: 2.50
Jan 13 00:00:24 allsky allsky.sh[1514]:  WB Blue: 2.00
Jan 13 00:00:24 allsky allsky.sh[1514]:  Binning (day): 2
Jan 13 00:00:24 allsky allsky.sh[1514]:  Binning (night): 2
Jan 13 00:00:24 allsky allsky.sh[1514]:  Delay (day): 5000ms
Jan 13 00:00:24 allsky allsky.sh[1514]:  Delay (night): 1000ms
Jan 13 00:00:24 allsky allsky.sh[1514]:  Text Overlay: Backyard Allsky  52.61N - 1.51W
Jan 13 00:00:24 allsky allsky.sh[1514]:  Text Position: 15px left, 75px top
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Name:  0
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Color: 255, 255, 255
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Background Color: 0
Jan 13 00:00:24 allsky allsky.sh[1514]:  Small Font Color: 255, 255, 255
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Line Type: 16
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Size: 10.0
Jan 13 00:00:24 allsky allsky.sh[1514]:  Font Line: 1
Jan 13 00:00:24 allsky allsky.sh[1514]:  Outline Font : 0 (no)
Jan 13 00:00:24 allsky allsky.sh[1514]:  Rotation: 180
Jan 13 00:00:24 allsky allsky.sh[1514]:  Flip Image: 0
Jan 13 00:00:24 allsky allsky.sh[1514]:  Filename: image.jpg
Jan 13 00:00:24 allsky allsky.sh[1514]:  Filename Save Directory: /home/pi/allsky/tmp
Jan 13 00:00:24 allsky allsky.sh[1514]:  Latitude: 52.6N
Jan 13 00:00:24 allsky allsky.sh[1514]:  Longitude: 1.51W
Jan 13 00:00:24 allsky allsky.sh[1514]:  Sun Elevation: -6
Jan 13 00:00:24 allsky allsky.sh[1514]:  Locale: en_US.UTF-8
Jan 13 00:00:24 allsky allsky.sh[1514]:  Notification Images: 1 (yes)
[... 794 lines cut ...]
Jan 13 01:37:22 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:37:22   @ 25.0 sec
Jan 13 01:37:49 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:37:50 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:37:50   @ 25.0 sec
Jan 13 01:38:17 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:38:18 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:38:18   @ 25.0 sec
Jan 13 01:38:45 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:38:46 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:38:46   @ 25.0 sec
Jan 13 01:39:01 allsky CRON[13508]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
[... cut ...]
Jan 13 01:40:38 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:40:38   @ 25.0 sec
Jan 13 01:41:05 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:41:06 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:41:06   @ 25.0 sec
Jan 13 01:41:33 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:41:34 allsky allsky.sh[3602]: STARTING EXPOSURE at: 2022-01-13 01:41:34   @ 25.0 sec
Jan 13 01:42:01 allsky allsky.sh[3602]: Sleeping 1.0 seconds...
Jan 13 01:42:02 allsky alls

jimjackii, 2022-01-13:
@ShedAstro 
Do you mount the tmp directory as a tmpfs?

ShedAstro, 2022-01-13:
Not sure what that means, I am not a coder or very good on pi...sorryI can follow instructions though, I did not see anything about that in the instructions when I installed your software...?? Sent from Mail for Windows From: SteffenSent: 13 January 2022 11:24To: thomasjacquin/allskyCc: ShedAstro; MentionSubject: Re: [thomasjacquin/allsky] AllSky V0.8.3 (new) - no timelapse (Issue ***@***.*** you mount the tmp directory as a tmpfs?

	
        
		Virus-free. www.avg.com

jimjackii, 2022-01-13:
@ShedAstro 
Do you follow these instructions?
https://github.com/thomasjacquin/allsky/wiki/Upgrade-from-0.8.2-or-prior-versions
and
https://github.com/thomasjacquin/allsky/wiki/Miscellaneous-Tips -> Reducing wear on your SDcard

Something else - your posts are kind of broken, obviously with HTML/CSS code.

ShedAstro, 2022-01-13:
I did not upgrade from old version, I installed latest version 2 days ago,  direct to rpi zero, and it all works just not time lapse, did you check the sys.log I sent as in there it shows the errors during the end of the night session...?? Not sure what you mean about posts broken...what posts....??? Sent from Mail for Windows From: SteffenSent: 13 January 2022 11:38To: thomasjacquin/allskyCc: ShedAstro; MentionSubject: Re: [thomasjacquin/allsky] AllSky V0.8.3 (new) - no timelapse (Issue ***@***.*** you follow these instructions?https://github.com/thomasjacquin/allsky/wiki/Upgrade-from-0.8.2-or-prior-versionsandhttps://github.com/thomasjacquin/allsky/wiki/Miscellaneous-Tips -> Reducing wear on your SDcardSomething else - your posts are kind of broken, obviously with HTML/CSS code.

	
        
		Virus-free. www.avg.com

ShedAstro, 2022-01-13:
I have checked and I have all the correct settings as shown in your first link... Sent from Mail for Windows From: SteffenSent: 13 January 2022 11:38To: thomasjacquin/allskyCc: ShedAstro; MentionSubject: Re: [thomasjacquin/allsky] AllSky V0.8.3 (new) - no timelapse (Issue ***@***.*** you follow these instructions?https://github.com/thomasjacquin/allsky/wiki/Upgrade-from-0.8.2-or-prior-versionsandhttps://github.com/thomasjacquin/allsky/wiki/Miscellaneous-Tips -> Reducing wear on your SDcardSomething else - your posts are kind of broken, obviously with HTML/CSS code.

	
        
		Virus-free. www.avg.com

jimjackii, 2022-01-13:
@ShedAstro 
Sorry - how do you post in this form here?
They're all broken with a lot of HTML / CSS code.

Please log in to Github directly via your web browser and have a look at it.
With so many unnecessary characters, the real question is difficult to find.

ShedAstro, 2022-01-13:
> @ShedAstro Sorry - how do you post in this form here? They're all broken with a lot of HTML / CSS code.
> 
> Please log in to Github directly via your web browser and have a look at it. With so many unnecessary characters, the real question is difficult to find.

really sorry was just replying to email, and thought they went direct to the sender, and not on here, will delete some, does this help
[log file.txt](https://github.com/thomasjacquin/allsky/files/7862536/log.file.txt)

jimjackii, 2022-01-13:
@EricClaeys 
I did more tests with the creation of the timelapse video.
I have now created the tmp directory as a symbolic link to the main tmp directory.

"/home/pi/allsky/tmp" -> "/tmp"

But I don't know at the moment whether this is a RAM memory directory? I don't think so.

This made it possible to create various timelaps videos, including old videos.

I therefore suspect a problem - if a tmpfs folder is used - with the created symbolic links to a single image files, which are in the sequence folders, if this is included as a storage tmp directory.

My tests showed that the termination when creating the timelaps videos was always different. the files created were of different sizes or the time until the termination was always different.

ShedAstro, 2022-01-13:
ok, so to summarise,
i installed the latest version just 2 days ago, all works as it should except i have to run the rpiHQ camera bin 2x2 as it wont work any other way.
the keogram and star trail images are created and the sequence folders with images in are there, all settings are as they should be,but as you see in the log, it throws an error when starting the end of night time lapse...
also i cant access the allsky.log as shown above, not sure why...
log posted above and hope that helps...
sorry for the useless posts i sent

jimjackii, 2022-01-13:
@ShedAstro 
No problem with the posts, I think.

Can you show the file "/home/pi/allsky/tmp/timelapseTMP.txt"?

ShedAstro, 2022-01-13:
> @EricClaeys I did more tests with the creation of the timelapse video. I have now created the tmp directory as a symbolic link to the main tmp directory.
>
> "/home/pi/allsky/tmp" -> "/tmp"
>
> But I [...]

so is this an error just related to us, or is there an issue with the main software...?

> @ShedAstro No problem with the posts, I think.
> 
> Can you show the file "/home/pi/allsky/tmp/timelapseTMP.txt"?

yes, did you see the log above i posted...?
here is the timelapseTMP.txt attached
[timelapseTMP.txt](https://github.com/thomasjacquin/allsky/files/7862590/timelapseTMP.txt)

jimjackii, 2022-01-13:
@ShedAstro 
Do you use a RasPi Zero?

Maybe there is a problem with the RAM memory. The RasPi Zero has only 512 MB.

The line
"Jan 13 07:36:37 allsky kernel: [62701.105420] Out of memory: Killed process 26146 (ffmpeg) ..."
indicates this.

---

your timelapseTMP.xt:
"[swscaler @ 0x18ee4b0] deprecated pixel format used, make sure you did set range correctly"
is the same error meassge linke me.

Something with the RAM seems to be the problem.
Maybe a new ffmpeg version has also been installed?

ShedAstro, 2022-01-13:
> @ShedAstro Do you use a RasPi Zero?
>
> Maybe there is a problem with the RAM memory. The RasPi Zero has only 512 MB.
>
> The line "Jan 13 07:36:37 allsky kernel: [62701.105420] Out of memory: Killed [...]

yes i just got the rpi zero...
sounds like an iisue with only 512mb RAM
i use the 12mp piHQ camera too...

ShedAstro, 2022-01-13:
i really hope this is not the case as i just spent 24 hours printing the enclosure for my camera, and i cant fit a full size rpi in there....

ShedAstro, 2022-01-13:
what size images  do you have your time lapse set to....?

jimjackii, 2022-01-13:
@ShedAstro 
Yes, with the RasPi Zero with 512MB it could be a problem with insufficient memory.

I have a RasPi 2B with 1GB of RAM. There seems to be another problem, which has to do with a tmp directory which is mounted as tmpfs.
Without a tmpfs directory, the timlapse videos are created for me.

---

I use these parameters for the timelapse videos
TIMELAPSEWIDTH=1296
TIMELAPSEHEIGHT=972
TIMELAPSE_BITRATE="3500k"
FPS=5

ShedAstro, 2022-01-13:
also do you use raspbian buster or the latest bullseye, as i use buster....??

ShedAstro, 2022-01-13:
> @ShedAstro Yes, with the RasPi Zero with 512MB it could be a problem with insufficient memory.
>
> I have a RasPi 2B with 1GB of RAM. There seems to be another problem, which has to do with a tmp [...]

i have no ide about the tmpfs directory i am not well up with linux and would not know what to do there.....

jimjackii, 2022-01-13:
@ShedAstro 
If you use the standard installation, no tmpfs directory will be used with you.

Then there is a problem with the RAM memory.
The message "... out of memory ..." from ffmpeg when creating the video indicates this.

---

I use also a HQ camera.

ShedAstro, 2022-01-13:
i have standard installation....just installed as shown on git...
well, not happy now then.....surely loads of people would have this issue, as loads use the zero and older pi's...??

ShedAstro, 2022-01-13:
does your HQ camera run on bin 1x1 as mine will only work on bin 2x2...??

jimjackii, 2022-01-13:
@ShedAstro 
Yes my system runs with 1x1.

ShedAstro, 2022-01-13:
> @ShedAstro Yes my system runs with 1x1.

i just get a red error screen  if i try that, could that be memory issue too...???

ShedAstro, 2022-01-13:
> @ShedAstro Yes my system runs with 1x1.

ok, all sorted....
had to increase the SWAP file to 1gb and all good, just created it manually on pi took 35 mins but worked fine....

jimjackii, 2022-01-13:
@ShedAstro 
That sounds good.

jimjackii, 2022-01-13:
@EricClaeys 
I have now added a "user" parameter when mounting the tmpfs.
This means that the user "pi" is the owner of the tmp directory.

tmpfs /home/pi/allsky/tmp tmpfs size=50M,user,noatime,lazytime,nodev,nosuid,mode=1777

ShedAstro, 2022-01-13:
> @EricClaeys I have now added a "user" parameter when mounting the tmpfs. This means that the user "pi" is the owner of the tmp directory.
> 
> tmpfs /home/pi/allsky/tmp tmpfs size=50M,user,noatime,lazytime,nodev,nosuid,mode=1777
> 
> Earlier I tested again with tmpfs, but only with 50MB. And still with the old swap setting of 100MB swap. A timelapse was also generated error-free.
> 
> To be on the safe side, I have now increased my RasPi 2B to 1GB additionally.
> 
> Strangely enough, however, there were never any messages about low memory in the log files. Now I'll watch it.

Can I ask why you use a resolution of 1296 x 972 as opposed to any other size, also why  only 5fps as mins is on 25 and worked well, albeit the Timelapse took 35 mins to make…?

ShedAstro, 2022-01-13:
Also mine works in 1x1 bin now with the new swap file size….👍🏼

jimjackii, 2022-01-14:
@ShedAstro 
The resolution is enough for me and the frame rate = 5 gives me the opportunity to comfortably look for meteors when watching the video.

EricClaeys, 2022-01-14:
@jimjackii, @ShedAstro 
It is VERY important when reporting problems to follow the instructions in the Wiki page on reporting problems.  It describes which log file to use and how to insure it has all the necessary information and nothing more.  And attaching it instead of copy / paste it is important. 

The Wiki page on troubleshooting timelapse issues should resolve your timelapse problems.  If not, let me know.

The RPiHQ camera works fine bin 1x1.  If it only works 2x2 for you, there is a problem somewhere and you should follow the troubleshooting Wiki page.

There should be no need to create a symbolic link to allsky/tmp.  If that directory is a tmpfs and you have problems, stopped making it a tmpfs until all other problems are resolved.

Thanks.

ShedAstro, 2022-01-14:
> @jimjackii, @ShedAstro It is VERY important when reporting problems to follow the instructions in the Wiki page on reporting problems. It describes which log file to use and how to insure it has all [...]

First of all if you had actually read my posted you will see that I already apologised for the cut and paste as I thought I was replying to someone directly via email and not posting on here directly, also if again you read my posts you will see that I can’t access the log they you need hence needing to post the sys.log, I showed this above and the error I got when trying to access the proper log, so all in all.JUST READ THE POSTS FIRST BEFORE COMPLAINING
MY SYSTEM IS FULLY WORKING NOW AND NO THANKS TO YOU, AND I NEVER MENTIONED ANYTHING ABOUT SYMBOLIC LINKS OR TMPFS DIRECTORIES, SO AGAIN READ MY POSTS BEFORE MOANING AT ME, THIS IS ALL TOTALLY NEW TO ME….

EricClaeys, 2022-01-14:
@ShedAstro Sorry if I didn't come across very well.  I should have directed the tmpfs and symbolic link portion of my reply to @jimjackii only.
/var/log/allsky.log probably isn’t readable by the "pi" user so you'll need to use `sudo` to access it.  I had forgotten that.  Soon after I got my Pi I made the file readable by pi: `sudo chmod 664 /var/log/allsky.log`.
BTW, I did read your messages but am doing it on my phone and read about a dozen related messages from different people and am having trouble keeping things straight.  Again, I appologize for not coming across very well.

ShedAstro, 2022-01-14:
> @ShedAstro Sorry if I didn't come across very well. I should have directed the tmpfs and symbolic link portion of my reply to @jimjackii only.
> /var/log/allsky.log probably isn’t readable by the "pi" [...]

Ok, no,worries, like I say this is all new to me…never used GitHub before..👍🏼
One question though, is there a way to stop it taking too many daylight images before the dark rolls in, I have too many light images at the start and end of my nighttime Timelapse, daytime capture is set to false, how does it determine what time to start capture for night, as it seems to start a bit too early and carry on in the AM a bit too long too…??
Thanks

EricClaeys, 2022-01-14:
@ShedAstro I suggest you read the Wiki page on "Angle and sunrise/sunset Explained".  It should answer your question.  Reading the other "Explained" pages would be a good idea and help you understand Allsky.
If you find something that doesn't make sense let me know.

ShedAstro, 2022-01-15:
> sudo chmod 664 /var/log/allsky.log

This did not work and the log was readable a few days ago and I still just get this with any log files I try and open…☹️

[image]

EricClaeys, 2022-01-15:
@ShedAstro are you running Mousepad on a PC or the Pi?
Try using `nano` to open the file.  Based on the error message it looks like it's not a permissions problem but a formatting problem.
Have you tried different encodings in Mousepad?

ShedAstro, 2022-01-15:
> @ShedAstro are you running Mousepad on a PC or the Pi?
> Try using `nano` to open the file. Based on the error message it looks like it's not a permissions problem but a formatting problem.
> Have you tried different encodings in Mousepad?

Ok, nano has worked, but the file is so long, and takes forever to get down to last nights log, is there a faster way to view, or clear it so I can try again tonight and just have the log for one night in there…?

EricClaeys, 2022-01-16:
@ShedAstro you can run this to truncate the log:
~~~
sudo truncate -s 0
~~~

EricClaeys, 2022-01-19:
@jimjackii Has the timelapse problem you originally reportrd been resolved?  If so, would you ease close this Issue?
Other problems described in this thread should probably have their own Issues if the are still problems.
Thanks

jimjackii, 2022-01-19:
@EricClaeys 
Oh sorry.
Yes,thet problem was solved for me.

closed 2022-01-19 by jimjackii