AllskyTeam/allsky #919: Error in allsky log

maserowik, 2022-01-12:
[image]

maserowik, 2022-01-12:
@EricClaeys I set the log to 4 and will attach the logs files.  

here is the ls ~/allsky/darks output

-0.jpg   13.jpg  20.jpg  29.jpg  36.jpg  43.jpg  50.jpg  59.jpg  -7.jpg
0.jpg    14.jpg  21.jpg  -2.jpg  37.jpg  44.jpg  51.jpg  -5.jpg  7.jpg
-10.jpg  15.jpg  22.jpg  2.jpg   38.jpg  45.jpg  52.jpg  5.jpg   -8.jpg
10.jpg   16.jpg  23.jpg  30.jpg  39.jpg  46.jpg  53.jpg  60.jpg  8.jpg
-11.jpg  17.jpg  24.jpg  31.jpg  -3.jpg  47.jpg  54.jpg  61.jpg  -9.jpg
11.jpg   18.jpg  25.jpg  32.jpg  3.jpg   48.jpg  55.jpg  62.jpg  9.jpg
-12.jpg  19.jpg  26.jpg  33.jpg  40.jpg  49.jpg  56.jpg  63.jpg
12.jpg   -1.jpg  27.jpg  34.jpg  41.jpg  -4.jpg  57.jpg  -6.jpg
-13.jpg  1.jpg   28.jpg  35.jpg  42.jpg  4.jpg   58.jpg  6.jpg

maserowik, 2022-01-12:
@EricClaeys I also post a new error in the all-sky github.

EricClaeys, 2022-01-13:
@maserowik I know what the problem is.  I will work on a solution.  Thanks for the info.

EricClaeys, 2022-01-14:
@maserowik this is almost for sure due to the image.jpg file being overwritten as it's being resized.  This is fixed in 0.8.3.
If you upgrade make sur you follow the instructions in the Wiki page on upgrading from prior releases.  It's the 2nd page.

maserowik, 2022-01-18:
@EricClaeys just did a fresh pull and no error in the allsky log file.  Closing the issue

EricClaeys, 2022-01-19:
@maserowik The bug is still there.  It only appears if the sensor temperature is 8 or 9 C.  We pass the temp as a 2-digit number with a leading 0 if needed, e.g., 09.  Unfortunately the script interpretes a number with a leading 0 as being in base 8 (an octal number that contains the characters 0-7).  That means 8 and 9 are not valid octal characters whic is basically what the error message said - the value (9) is too great for the base (8).

Keep bugging me until I fix it.

maserowik, 2022-01-21:
@EricClaeys 
I see you pushed a new update to the file.  I just did a 
cat /var/log/allsky.log | grep error and the log file is clear.

Going to close this issue.
Thanks
Mike

closed 2022-01-21 by maserowik