AllskyTeam/allsky #920: Raspberry PI zero out of memory error, no time lapse

ShedAstro, 2022-01-13:
I used my last night and it worked well, except it will not create a time lapse due to an out of memory error, this is the error i get:

**Jan 13 07:36:37 allsky kernel: [62701.105420] Out of memory: Killed process 26146 (ffmpeg)** 

i have the pi zero which i know only has 512mb of ram, but see no other people with this issue...??
also using latest buster and NOT bullseye....

Any ideas...

maserowik, 2022-01-13:
have you tried to increase you swap file  ? 
https://pimylifeup.com/raspberry-pi-swap-file/

ShedAstro, 2022-01-13:
> have you tried to increase you swap file ? https://pimylifeup.com/raspberry-pi-swap-file/
> 
> check out the wiki page https://github.com/thomasjacquin/allsky/wiki/Troubleshooting:-timelapse

ok, so have followed the guide to uping the swap file and now running htop alongside generating the time lapse, but how long should it normally take as its been running now for 35 mins and still going, swap file not gone above 400mb as of yet....

ShedAstro, 2022-01-13:
It’s finished and took 42 mins, and worked well. 👍🏼
Are there any good settings to use for image size, Timelapse image size, frames per second, for the piHQ camera, or are they all best left at default, I am binning the camera 2x2 as no need for such a big image, but more concerned about what size the video Timelapse file should be set at…

EricClaeys, 2022-01-16:
Closing Issue since problem was resolved.

closed 2022-01-16 by EricClaeys