AllskyTeam/allsky #923: new issue after update

maserowik, 2022-01-14:
website was working fine a few minutes now when I  to n3bsq.com/allksy getting the following
[image]

in my config.sh I have the command set to true
[image]

ftp site has the data.json file in it
[image]

here is the lighttpd error log
```
pi@Pi-AllSky:/var/log/lighttpd $ cat error.log
2022-01-14 09:06:23: (server.c.2059) server stopped by UID = 0 PID = 1
2022-01-14 09:06:39: (server.c.1464) server started (lighttpd/1.4.53)
2022-01-14 09:06:39: (server.c.1493) WARNING: unknown config-key: compress.cache-dir (ignored)
2022-01-14 09:06:39: (server.c.1493) WARNING: unknown config-key: compress.filetype (ignored)
2022-01-14 09:06:53: (server.c.2059) server stopped by UID = 0 PID = 1
2022-01-14 09:06:53: (server.c.1464) server started (lighttpd/1.4.53)
2022-01-14 09:06:53: (server.c.1493) WARNING: unknown config-key: compress.cache-dir (ignored)
2022-01-14 09:06:53: (server.c.1493) WARNING: unknown config-key: compress.filetype (ignored)
2022-01-14 09:52:21: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  opendir(/home/pi/allsky/images): failed to open dir: No such file or directory in /var/www/html/includes/days.php on line 37
```

Mike

rce1086, 2022-01-14:
Please note that the error message ends in “_DATE”, whereas the line in the config file that you show ends in “_DATA”… someone’s made a typo, by the looks of it…

EricClaeys, 2022-01-14:
@maserowik would you please attach your config.sh file, and the data.json and config.js files from your FTP server?
What version of Allsky-website is your FTP server running?

The lighttpd error log is from your Pi, not the FTP server.  The last entry is troubling - in the WebUI can you click on "Images" to view images for a day, and see if it works?

@rce1086 Thanks for pointing out the typo.  It should be _DATA.

maserowik, 2022-01-14:
@EricClaeys 

[config.js.txt](https://github.com/thomasjacquin/allsky/files/7872469/config.js.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/7872470/config.sh.txt)
[data.json.txt](https://github.com/thomasjacquin/allsky/files/7872471/data.json.txt)
 

how do I find the allsky website verison number?

I can view the images fine on the WebGui

I can set you with with remote ssh if you like to access my setup.

Mike

EricClaeys, 2022-01-16:
@maserowik try putting the data.json file in the same directory as controller.js.
The Allsky Website doesn't really have a version number.  To see if you have the latest files look for the newest file you have, then compare to what's in GitHub.  There's probably a better way to do it with Git though.

EricClaeys, 2022-01-20:
@maserowik   Any update?
FYI, yesterday I added a version number to the website and set it and the allsky-portal version to 0.8.3 to match Allsky.

maserowik, 2022-01-21:
@EricClaeys 
This issue is solved Closing ticket

closed 2022-01-21 by maserowik

EricClaeys, 2022-01-22:
@maserowik Mike, what did you do to fix it?

maserowik, 2022-01-22:
Did a reinstall of all software. Allsky/Allsky web and allsky hi.   Maybe I had something wrong on my end but it looks like it is working for now

Michael A. Serowik