AllskyTeam/allsky #924: Software Installation Instructions Needs An Update

darkmetalman, 2022-01-14:
When trying to execute the step 2:
git clone --recursive https://github.com/thomasjacquin/allsky.git

It will ask for a user and password, then fail (even after providing a valid one) because GitHub now needs a token in order to sign in

Regards!

EricClaeys, 2022-01-16:
@ckuethe, @linuxkidd, @AndreasLMeg, have you guys seen this?  I just did a full install yesterday and wasn't asked for a password.

linuxkidd, 2022-01-16:
I've only seen it when using `ssh` access to github, not with `https` access..  

@darkmetalman 
- Can you confirm for sure it happened with `https` git clone?
- If so, can you provide:
    -  a console copy/paste of the session.
    - output of `git --version`
- Also, please confirm your `.gitconfig` does not have something like:
```
[url "ssh://git@github.com/"]
  insteadOf = https://github.com/
```

Thanks!

darkmetalman, 2022-01-17:
Hello!!

I tried again this morning, but this time with a fresh install of Raspberry Pi OS Lite and everything worked fine, I don't know what caused the issue.

Thanks for your help!!

linuxkidd, 2022-01-17:
Great to hear!  I'll go ahead and close out this issue.  Let us know if you run into any other problems. :)

closed 2022-01-17 by linuxkidd