AllskyTeam/allsky #927: Cant install software

SM4IVE, 2022-01-15:
I had this camera going in 5 years now i decided to updaste the sw  but it cant finish the instalation se below, some words in swedish:-)

* Compile allsky software

make[1]:open katalogen ”/home/pi/allsky/src”
2022-01-15 05:57:18 Building capture program...
capture.cpp:1:28: fatal error: opencv2/core.hpp: Filen or katalogen does not exist
 #include <opencv2/core.hpp>
                            ^
compilation terminated.
Makefile:121: receptet för målet ”capture” misslyckades
make[1]: *** [capture] Fel 1
make[1]: leaving katalogen ”/home/pi/allsky/src”
Makefile:24: receptet för målet ”all” misslyckades
make: *** [all] Fel 2
Compile failed!
pi@allsky:~/allsky $ 

what to do?
Camera: ASI185MC
Computer: Raspberry Pi 3

AndreasLMeg, 2022-01-15:
@SM4IVE I think I had the same problem with an "old" Rasperry OS. Solved with an new(er) OS. (latest buster or new bullseye with some issues)
https://github.com/thomasjacquin/allsky/issues/790

EricClaeys, 2022-01-15:
@linuxkidd Michael, would you please investigate?  Looks like opencv2 isn't installed?  src/Makefile has "opencv" and "opencv4" in it, but not "opencv2", if that matters.
Thanks.

SM4IVE, 2022-01-15:
Andreas iave PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
So if i need to change to a newer sw i have to take down the camera from its location i dont think i can flash via wifi.

EricClaeys, 2022-01-15:
@SM4IVE I have an email with a reply by you regarding having Stretch installed, but that message isn't here.  Did you delete it?

Torakin9366, 2022-01-15:
Ich glaube, diese Fehlermeldung kommt wenn man das Master zip zum installieren nimmt, besser geht es wenn man die Seite aus der Beschreibung nimmt

linuxkidd, 2022-01-15:
> Ich glaube, diese Fehlermeldung kommt wenn man das Master zip zum installieren nimmt, besser geht es wenn man die Seite aus der Beschreibung nimmt

--- translation:
> I think this error message comes when you take the master zip to install, it works better if you take the page from the description

linuxkidd, 2022-01-15:
@SM4IVE Can you provide the output of:
```
cat /etc/os-release
```

Note that if the `VERSION_ID` value is < 10, then this would be the reason for this compile error.  In order to maintain compatibility with the most recent Raspberry Pi OS releases, we've had to drop support for Debian Stretch / Version 9.

SM4IVE, 2022-01-16:
I had some errors during update to Buster,
Now getting this error when trying to update.

apt: relocation error: /usr/lib/arm-linux-gnueabihf/libapt-private.so.0.0: symbol _ZN3URIcvNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEB5cxx11Ev version APTPKG_5.0 not defined in file libapt-pkg.so.5.0 with link time reference
pi@allsky:~ $ 
 How to fix this? i am not a Linux guru :-)

linuxkidd, 2022-01-16:
eek... that sounds like one of the files for the 'apt' tool got corrupted.  Possibly it got missed during the upgrade, or is corrupted due to a failing SD card.

I'd recommend copying off your current pictures ( and settings ), then start fresh with a new Raspberry Pi OS install on the SD card.  Sorry this is taking more effort than planned.

linuxkidd, 2022-01-19:
Hi @SM4IVE ... Any luck with the re-install?

SM4IVE, 2022-01-19:
Not yet been busySkickat från min Galaxy

EricClaeys, 2022-02-10:
@SM4IVE ... Any luck with the re-install?

EricClaeys, 2022-05-17:
Closing issue.  No response from user in 3 months.

closed 2022-05-17 by EricClaeys