AllskyTeam/allsky #932: No Timelapse, star trails or keogram produced….

ShedAstro, 2022-01-16:
My system worked fine for 1 night, since then and with absolutely no changes it’s failed since, I have already changed swap file size and other changes as recommended, and they solved previous issues, also now if I try and create the files manually, the terminal window closes after about 3 mins and does not finish the process of creating either the star trail or time lapse, but nothing in the log…it’s as if the system never even tried to complete them, it’s set to do so in the editor…
All images are present and correct, and system is set to create just time lapse and star trail, but neither work…

EricClaeys, 2022-01-17:
@ShedAstro Did it create startrails and timelapse the one night it worked?
What commands are you running to create the files manually?
When created manually. It won't write anything to /var/log/allsky.log.
How familiar are you with Linux?

ShedAstro, 2022-01-17:
> @ShedAstro Did it create startrails and timelapse the one night it worked?
> What commands are you running to create the files manually?
> When created manually. It won't write anything to /var/log/allsky.log.
> How familiar are you with Linux?

Yes it created everything the first night but nothing since and I have changed nothing…
To crate manually I run

cd allsky
scripts/generateForDay -t DATE

ShedAstro, 2022-01-17:
Thought it might have been a power issue as I had to add a heater ring in the dome as it was misting up all the time, so swapped it out for a bigger one, but still the same, so have been transfers g the images to my windows PC and using the time lapse generator app in there to create them….but still can’t get it to work on the rpi zero, I just don’t think the little zero is up to it anymore, the software seems to powerful to run on it when creating the end of night files….☹️

ShedAstro, 2022-01-19:
> @ShedAstro Other people with Pi Zero have things working, after increasing swap space.
> 
> Please attach you log file per the instructions on the "Reporting problems" Wiki page. We'll get this resolved.

Unfortunately I can’t access the log, as I keep saying on here I this error every time I try to access any of them, it did work fine for the first night and yes after upping the swap file size to 2gb, but then every night after, it’s failed, and yest I have changed nothing…..

ShedAstro, 2022-01-19:
If I can get past the error above I can give you all the log you would need, but no matter what I try the error is the same. And this worked to start with too….

ShedAstro, 2022-01-19:
Ok, have sorted the log issue, and have followed the instructions and set the debug level to 4, so after tonight when it’s does not record and Timelapse, I will have a log to send you…
So hopefully this will help to sort issue out, as I don’t want to have to swap to a pi4

ShedAstro, 2022-01-19:
> @ShedAstro Glad you sorted the log file out. What was the issue?

I have no idea, but if I run “sudo pcmanfm” it allows me to open it…in a different editor…don’t ask why as I have no clue, the other issue is still there I can’t open if I just try to open from the file manager…

EricClaeys, 2022-01-20:
@ShedAstro I check the other day, and I did this a long time ago so I could read and update the file as Pi.
~~~
chgrp pi /var/log/allsky.log
chmod 664 /var/log/allsky.log
~~~

Without doing this you have to use "sudo" to edit the file and possibly also view it.

Any update on startrails or keograms being produced?

ShedAstro, 2022-01-20:
> @ShedAstro I check the other day, and I did this a long time ago so I could read and update the file as Pi.
>
> ```
> chgrp pi /var/log/allsky.log
> chmod 664 /var/log/allsky.log
> ```
>
> Without doing this [...]

ok, it started a star trail and added about 5 images, but no timelapse, i have keogram turned off, but all images for the night are present as usual, log attached
[allskylog.txt](https://github.com/thomasjacquin/allsky/files/7904242/allskylog.txt)

EricClaeys, 2022-01-20:
@ShedAstro I see in the log where it started the timelapse, but don't see any error message or anything about uploading it.  Is it possible it just took a long time (the log ends about 30 minutes after it started the timelapse)?  It can take over a half hour depending on how many pictures you have.

Try `cd allsky; scripts/generateForDay -t 20220119` to generate last night's timelapse and see if it produces the file or an error.

ShedAstro, 2022-01-20:
No, it’s been running all day, and working fine, this is my issue, it’s an odd one. So if I try and run the Timelapse manually, I log into the PI from VNC, run the command from terminal, and it comes up, thats it’s creating the Timelapse, then after between 1 and 5 mins, the VNC connection goes off, then back up 2 seconds later with the PI  desktop and no terminal…it’s closed the terminal, and does this every time, so I assume this is also happening internally during the creation in the mornings, now the Timelapse started last night and there are around 20 images in it from what I can tell… I thought it may be a lower issue, so upped the power supply on the pi, to a 4 amp one.. I downloaded the log this morning from the PI at around 10:30..🤔🤔

 Sent from my iPad

EricClaeys, 2022-01-21:
@ShedAstro 
**Was the startrails created?**  Per the log, it looks like it was.  **What's `UPLOAD_STARTRAILS` set to in config.sh?**

Here's something to try manually for the timelapse.  The "nohup" command is from the old UNIX and modem days - it says if the modem hangs up ("hup") when a command is running, keep running the command instead of killing it - hence the "no".  Now days of course "modem" can be replaced with "VNC" but the concept is the same. 
~~~
cd allsky
nohup scripts/generateForDay -t 20220119 > /tmp/timelapse.txt 2>&1
~~~

The `> /tmp/timelapse.txt 2>&1` tells the shell to send out output and error output to a file called **/tmp/timelapse.txt**.  This is needed in case the program tries to write any output after the VNC connection goes away and there's nothing for the command to write to.  Assuming you lose your VNC connection again, check that file and **allsky/tmp/timelapseTMP.txt** (this file will have one line for every image, as well as one line about deprecated pixel format you can ignore.  Anything more than that indicates an error).

Let me know how this goes.

**How much RAM memory and swap do you have**?

ShedAstro, 2022-01-21:
Ok first of all I have actually stated three times that the star trail image was there yesterday but only consisted of around 20 images, then it must have stopped, also I have said numerous times I have the pi zero so 512mb ram and have created a 2gb swap file which did solve my very first issues when I first built the camera, but it only worked for 1 day.
If the issue is with VNC that will only help using your instructions if I am to do it manually every day, I need the auto Timelapse to work, something is not right…

ShedAstro, 2022-01-21:
Ok so just checked from last night, and the Timelapse worked, but there was just 1 image in the star trail image…so that failed….this is really odd….I have changed nothing. So why has it gone 5 days with nothing working and then suddenly the Timelapse works, for no reason…🤔🤔 I bet it won’t work again now for a week or more, but I would really like the star trail image to work

EricClaeys, 2022-01-21:
@ShedAstro Take a look at [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/Startrails-Explained).  I suspect that's your startrails problem.

I don't know why timelapse would fail 5 days then work.  One possibility is there still isn’t enough swap space and some days there are a lot of other processes using memory so it runs out, whereas other days not as much memory is being used. 

My suggestion on using "nohup" was to help determine what was causing the timelapse problem, since we have no idea what is causing it because we have only been able to get one log file and it didn't show any problems.

ShedAstro, 2022-01-21:
> @ShedAstro Take a look at [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/Startrails-Explained). I suspect that's your startrails problem.
>
> I don't know why timelapse would fail 5 days [...]

Where on that wiki page, there is just a long menu on the right side…???

bbillp, 2022-01-21:
Why not use a full size Pi instead of the Pi Zero to eliminate poor performance.   It seems there are many and various complaints from Pi-Zero users.

EricClaeys, 2022-01-30:
@ShedAstro Has your problem been resolved?
The Wiki page I gave you the link to has two "Click here" links to bring up an Overview and Troubleshooting information.

Startrails can also only use one image if the `BRIGHTNESS_THRESHOLD` value in config.sh needs adjusting, as described on the Wiki page.

ShedAstro, 2022-01-30:
NO, issue not solved at all, I have given up trying to get help on here, as nothing works….☹️

 Sent from my iPad

linuxkidd, 2022-02-01:
Hi @ShedAstro -- We've certainly not given up on helping you get this resolved. 

If you can't open the log file locally, it may be beneficial to us -- you can drag + drop the file into the comment block for this issue and it'll attach the file for us to review.  

We'd love to get this working for you.

ShedAstro, 2022-02-02:
> Hi @ShedAstro -- We've certainly not given up on helping you get this resolved.
>
> If you can't open the log file locally, it may be beneficial to us -- you can drag + drop the file into the comment [...]

Thanks for the offer of help.
The strange thing is, that is works around 3 days out of 7, and I get both star trail and Timelapse, (I disabled keogram) the rest of the time it’s hit and miss to whether I get anything or one or the other, so as it works some of the time, I guess it’s all set up correctly, and think it maybe down to the PI zero and minimum memory, I have upped the swap file to 2gb, and still the same, so I am at a loss as to why it is so intermittent.
I could post the log here but it does not post it all as there are only a certain amount of lines it allows, and as you see in the error above I can’t get to it on the Pi, but will try again, bear with me…👍🏼

ShedAstro, 2022-02-02:
I have also found that I get the same issues when trying to use the web browser on the PI, it’s just shuts it all down for a second and then comes back up with all windows closed, so it’s not just the allsky software that is causing the issue….it seems that the little PC just can’t cope, with more than one thing at a time being done….I have swap set at 4gb now and seems worse TBH

EricClaeys, 2022-02-02:
@ShedAstro Are you using the Pi for anything other than Allsky?  Everything that runs on it takes memory ans swap, so you can try increasing swap to 6 GB.

As I mentioned earlier, you may need to adjust the startrails threshold.  It determines how many pictures are included.  When you say startrails "don't work" do you mean no file is produced, or a file is produced but it only contains 1 picture?

linuxkidd, 2022-02-02:
@ShedAstro I suspect the issue is that the Pi is running the 'Desktop' version of Raspberry Pi OS.  The graphical environment consumes a massive amount of CPU and memory resources.  For a computing platform as low powered as the Pi, this conflicting overhead will easily lead to problems for other memory / cpu intensive workloads, like startrails, keogram and especially timelapse generation.

I would recommend to backup your config ( if you need to ), then re-image the card using the [Raspberry Pi OS 'Lite'](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit) version (3rd download option on the linked page).

This Lite version does not have a GUI at all, which leaves all of the resources available to base level operating system functions, and Allsky's needs.

ShedAstro, 2022-02-02:
> @ShedAstro I suspect the issue is that the Pi is running the 'Desktop' version of Raspberry Pi OS. The graphical environment consumes a massive amount of CPU and memory resources. For a computing platform as low powered as the Pi, this conflicting overhead will easily lead to problems for other memory / cpu intensive workloads, like startrails, keogram and especially timelapse generation.
> 
> I would recommend to backup your config ( if you need to ), then re-image the card using the [Raspberry Pi OS 'Lite'](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit) version (3rd download option on the linked page). Once that's done, follow the installation instructions for Allsky and see how it goes.
> 
> This Lite version does not have a GUI at all, which leaves all of the resources available to base level operating system functions, and Allsky's needs.
> 
> @EricClaeys Assuming the above corrects the issues seen here ( and I really think it will ), we probably need to update the installation documentation to specify the Lite version.

Ok, I can try that, so when it’s installed how do I then get my config back up on the new card, and also which file do I back up to keep all my settings….
And how do I get into the Phi to update etc, through SSH…?
And should I up the swap file on the new install…if so what too…?
I have never used a Pi without desktop before so that will be new to me…

EricClaeys, 2022-02-03:
@ShedAstro I have been reading your posts but am not sure if you have been reading, or acting upon, mine.  I mentioned twice that the issue you are having with startrails may be due to its 
 threshold setting.  I have not received any feedback that you have tried that, and if so, how it worked.  If you do not know how to implement a suggestion, please say so.  @linuxkidd and I have been working on Linux and UNIX for years and can help.

Back up the configuration in allsky/config and the .json files in /etc/raspap.  If you have the Allsky website, in /var/www/html/allsky back up config.js and virtualsky.json.
I know it's a hassle - we plan to make it easier to back up in the future.

Given that your Pi is having multiple problems, I suggest we tackle one problem at a time.  The timelapse problem is likely going to be the easiest so let's work on that, ok?

ShedAstro, 2022-02-03:
The start trail same as the time lapse works 3 times out of 7, so the fact that it can work fine shows that it’s not a setting that’s causing issues, and yes, I did try that setting you mention and it made no difference….so I put it back to the default…. It’s funny as it seems the more I up the swap file the worse it is getting…..🤔🤔

 Sent from my iPad

ShedAstro, 2022-02-04:
I have tried the different settings and also now used the lite version of the OS, and it’s no better, it made nothing the first two days, and last night it did produce a Timelapse but no star trails….so back to the drawing board…☹️☹️

EricClaeys, 2022-02-05:
@ShedAstro Any chance you could give us a login on your Pi so we can look around?  I did this with another user and found and fixed a problem quickly.

ShedAstro, 2022-02-05:
> @ShedAstro Any chance you could give us a login on your Pi so we can look around? I did this with another user and found and fixed a problem quickly.

Yes, but I have no idea how, or what you need….🤔

ShedAstro, 2022-02-06:
> @ShedAstro is your Pi accessable via the Internet?

Only on my local network…🤔

ShedAstro, 2022-02-06:
Tried over ssh to generate manually the star trails and the Timelapse, and half way through either the connection is lost and then come straight back up with terminal closed down and no files produced….so the Same as over VNV using the desktop environment ☹️☹️

EricClaeys, 2022-02-07:
@ShedAstro Try:
~~~
# Stop Allsky
nohup allsky/scripts/generateForDay.sh -s 20220206 > /tmp/startrails.txt 2>&1
# Restart Allsky
~~~

Look at the end of the log file to see what startrails said:
~~~
sudo tail -100 /var/log/allsky.log
~~~

ShedAstro, 2022-02-07:
Did that over ssh, and it hung after about 3 mins, same as it always does, and this is the last few lines of the lot again exactly as you said, and nothing there at all

50 WBB=2.00 GAIN=1.00 GAINDB=000 BIN=2 FLIP=0 BIT_DEPTH=08 FOCUS= 37
Feb  7 09:54:59 allsky allsky.sh[678]: STARTING EXPOSURE at: 2022-02-07 09:54:59   @ 32.00 ms (0.03 sec)
Feb  7 09:54:59 allsky allsky.sh[678]:  > Kill command: pgrep 'raspistill' | xargs kill -9 2> /dev/null
Feb  7 09:54:59 allsky allsky.sh[678]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220207095459.jpg' --thumb none --burst -st --timeout 2000 --nopreview --mode 2 --width 2028 --height 1520 --exposure auto --analoggain 1 --awb auto --rotation 180 --saturation -20 --brightness 57 --quality 95
Feb  7 09:55:04 allsky allsky.sh[678]:   > Focus: 36.870307
Feb  7 09:55:04 allsky allsky.sh[678]:   > Saving DAY image 'image-20220207095459.jpg'
Feb  7 09:55:04 allsky allsky.sh[678]: Sleeping 3.0 seconds...
Feb  7 09:55:04 allsky allsky.sh[678]: DEBUG: saveImage.sh DAY /home/pi/allsky/tmp/image-20220207095459.jpg EXPOSURE_US=32000 BRIGHTNESS=57 MEAN=-1.000 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=1 WBR=2.50 WBB=2.00 GAIN=1.00 GAINDB=000 BIN=2 FLIP=0 BIT_DEPTH=08 FOCUS= 37
pi@allsky:~ $

ShedAstro, 2022-02-09:
I guess you run out of ideas then….🤔🤔☹️

EricClaeys, 2022-02-09:
@ShedAstro not yet.  I have been traveling so have not been able to reply.
What was in /tmp/startrails.txt?

ShedAstro, 2022-02-09:
I posted it above….🤔🤔

 Sent from my iPad

EricClaeys, 2022-02-09:
@ShedAstro what you posted 3 days ago was the end of `/var/log/allsky.log`, not `/tmp/startrails.txt`

ShedAstro, 2022-02-09:
This is what you posted to me, and this is what I ran and posted here, so I am dong exactly what you asked…..🤔😂

Look at the end of the log file to see what startrails said:

sudo tail -100 /var/log/allsky.log

EricClaeys, 2022-02-10:
@ShedAstro Sorry for the confusion.  The `sudo tail -100 /var/log/allsky.log` command displays the last 100 lines of the main Allsky log file.  You already included that output so there's no reason to include it again.

`nohup allsky/scripts/generateForDay.sh -s 20220206 > /tmp/startrails.txt 2>&1` is the command to test if you can generate a startrails.  It sends its output to a file called `/tmp/startrails` which is a different directory than /home/pi/allsky/tmp.
I suggest stopping Allsky before running that command, then restart after the command finishes.
I'm hoping the contents of `/tmp/startrails.txt` will give a clue as to the problem, so if you could run that I'd appreciate it.

ShedAstro, 2022-02-10:
I ran that all above,  but because it hangs after a couple of mins, and never finished, it does not create that file, it does not exist….
It does this over ssh, after 2 mins or so the ssh screen shows a disconnection error, then pops back up with a new screen but empty, it does the same as over VNC, just hangs so that file never even gets created….

EricClaeys, 2022-02-10:
@ShedAstro Try this instead:
`nohup allsky/scripts/generateForDay.sh -s 20220206 > /home/pi/startrails.txt 2>&1`
An empty `/home/pi/startrails.txt` file should be created after you are done typing, even if the system hangs after that.  It's possible the file in /tmp got removed after a reboot.  The file in /home/pi won't get removed.

ShedAstro, 2022-02-10:
I can assure you that file does not exist, only only a Timelapse.txt file….

 Sent from my iPad

EricClaeys, 2022-02-11:
@ShedAstro The `timelapseTMP.txt` file is in /home/pi/tmp, whereas the file created by the command I suggested is in /home/pi.  If `/home/pi/startrails.txt` does not exist after running the command there is something very seriously wrong with your Pi that has nothing to do with Allsky.

When you run that command or any other that has "> some_file" on the command line the first thing the shell does, before executing the command, is create the file.

ShedAstro, 2022-02-11:
Ok, I just ran that command again, and it has now created the file you said in the correct Home/pi folder, BUT the file is completely empty, I open and it’s blank, nada…….it hung after about 30 seconds of running the command….
So what does this mean….

EricClaeys, 2022-02-11:
@ShedAstro it means we try something else until we find the problem and fix it if we can.
In config/config.sh, set `STARTRAILS_EXTRA_PARAMETERS="-v -v"` then try the exact same generateForDay.sh command you just tried.  This change will set the verbosity level to 2 (because there are 2 -v's) which should add a line to the output file for every image.
The Pi will still hang but we'll hopefully get some information on how far it got.

Please make sure you stop Allsky before running the test so the mi imal amount of memory is being used.

ShedAstro, 2022-02-11:
> @ShedAstro it means we try something else until we find the problem and fix it if we can. In config/config.sh, set `STARTRAILS_EXTRA_PARAMETERS="-v -v"` then try the exact same generateForDay.sh [...]

Ok, tried that and still exactly the same, the file is there and empty….it hung after 20 seconds or so

EricClaeys, 2022-02-11:
@ShedAstro Interesting.  I would have expected SOME output.  Try this:
~~~
cd;  allsky/startrails -v -v -d /home/pi/allsky/images/20220206 -e jpg -b 0.1 -o startrails_test.jpg
~~~
and watch for any output before the system drops the connection.  My output started about a second after entering the command.  If that works, there will be a startrails file in $HOME/startrails_test.jpg.

Something else to try:
* Create a directory called allsky/images/test
* Copy any 5 files from allsky/images/test/20220206 to allsky/images/test
* `cd;  allsky/startrails -v -v -d /home/pi/allsky/images/test -e jpg -b 0.1 -o startrails_test2.jpg`

That will only use 5 files so will have minimal impact on memory.  Again, do this when Allsky is off.

Also, please run `cat /etc/dphys-swapfile` and paste the output to your reply.
Thanks.

EricClaeys, 2022-05-27:
Closing Issue - no response from user in over 3 months.

closed 2022-05-27 by EricClaeys