AllskyTeam/allsky #940: question

crownedgrouse, 2022-01-17:
Hi,
is there a way to do this with allsky  ?  : 
[image]

EricClaeys, 2022-01-18:
@crownedgrouse What does this keogram show you?  I assume the blue is daylight.  How about the white, brown, and black stripes?
I'm not sure how many people would be interested in this.  Most people delete their files after 14 days due to space issues on their Pi.
As an alternative I could write instructions on how to do this manually.

crownedgrouse, 2022-01-18:
Hi,
This keogram shows sky of a place all along year.
Each 'pixel' is a picture. I'm not sure what is frequency of pictures. Several per hour I guess. Your input welcome.
I find this keogram poetic, artistic and pedagogic.
I suspect white stripes to be moon phases. Yellow ones I dunno to be honest.
This keogram shows also to kids length of daylight changes along year. Easy to show exinox , etc...
For pi disk space, a simple regular download of pictures elsewhere should do the trick?

EricClaeys, 2022-01-19:
@crownedgrouse See if you can contact Cees Bassa to find out what was done to create this.
If you use the Allsky website you should have lots of keogram files and it's easy to make a keogram of keograms.

loloderu, 2022-01-27:
You could launch the keogram utility givind it the sequence of all keograms of the year and with a --rotate 90 parameter

loloderu, 2022-01-28:
[image]
This what I obtained for a period from summer to winter

loloderu, 2022-01-28:
For this you need to modify a bit the keogram.cpp il order it accepts variable images with as input (your keograms)

EricClaeys, 2022-01-29:
@loloderu What exactly did you modify?  The program or arguments passed to it?

crownedgrouse, 2022-01-29:
@loloderu yes, can you share or propose a PR ? A documentation would be neat.

cbassa, 2022-01-29:
My camera ran 24/7 and I stored all images on a different computer to generate the daily keograms and then to generate the year long keogram from it. The individual images were captured using my python based `capture` tool (https://github.com/cbassa/asm/blob/master/capture) replacing `capture.cpp` and running within the rest of the `allsky` frame work (web client etc). My capture tool uses different exposure and gain logic and does not rely on the definition of night and day based on solar elevation. 

The daily keograms were created by taking the timestamps of each individual image and averaging those to obtain a vertical column in the keogram for every minute, hereby removing any time offsets introduced by delays in the camera and Raspberry Pi. The year long keogram just adds these keograms vertically and rescales them.

loloderu, 2022-01-29:
The generation of the image keogram.jpg is done wth the 2 commands below:
------------------------
find . -name "keogram-*.jpg" -exec cp -p {} seq \;  /* gather all keograms in the directory "seq"*/
~/keogall $ALLSKY_HOME/images/seq jpg keogall.jpg -rotate 90 /* build a meta keogram using an horizontal cut of each daily keogram*/
-----------------------
keogall.cpp is keogram.cpp with the below mod:
replace this code (from the previous unthreaded version of keogram.cpp)
<<
          if (accumulated.empty())
          {
             accumulated.create(imagedst.rows, files.gl_pathc, imagesrc.type());
          }
          // Copy middle column to destination
         imagedst.col(imagedst.cols / 2).copyTo(accumulated.col(f));
>>
by this code, that permit to manage different image with as input
<<
         int imrows;
         cv::Mat imcol;

          if (accumulated.empty())
          {
              imrows=imagedst.rows;
             accumulated.create(imrows, files.gl_pathc, imagesrc.type());
          }
          // Copy middle column to destination
          if(imagedst.rows>=imrows)
          {
              imagedst(cv::Range(0,imrows),cv::Range(imagedst.cols / 2,imagedst.cols / 2 + 1 )).copyTo(imcol);
          } else {
               imagedst.col(imagedst.cols / 2).copyTo(imcol(cv::Range(0,imagedst.rows),cv::Range(0,1)));
          }
          imcol.col(0).copyTo(accumulated.col(f));
>>

You can also replace the tm_hour annotation by a tm_mon annotation.

EricClaeys, 2022-01-29:
@loloderu The code changes above generate the yearly keogram?  Will they work for regular keograms? i.e., shuld we modify keogram.cpp?  Or do we need an "if" statement to check if we're creating a yearly keogram and if so, use the changes yo specified above?

@AndreasLMeg. @linuxkidd.  FYI

loloderu, 2022-01-29:
This code would work also for dayly keogram as it is more generic than the original as it accepts varying width images.
It only if you want month label tha you should additionally change the 3 lignes below
           if (t->tm_hour != prevHour)
                   snprintf(hour, 3, "%02d", t->tm_hour);
              prevHour = t->tm_hour;
by 
           if (t->tm_mon != prevHour)
                   snprintf(hour, 3, "%02d", t->tm_mon);
              prevHour = t->tm_mon;

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Add year-long keogram feature #940**

closed 2022-06-29 by EricClaeys