AllskyTeam/allsky #943: 8.3 Upload issue

hpservertech, 2022-01-18:
Updated to 8.3 from 8.2c.  Uploads are no longer working.  It is trying to upload images from the images/<date> folder but the images are all in the tmp folder.

What gets the files moved from the tmp folder to the date folder under the images directory now?

Here is the error:
Jan 18 10:31:54 allsky allsky.sh[25153]:   > Saving DAY image 'image-20220118103153.jpg'
Jan 18 10:31:55 allsky allsky.sh[25153]: *** upload.sh: ERROR: File to upload '/home/pi/allsky/images/20220118/image-20220118103153.jpg' not found!
Jan 18 10:31:59 allsky allsky.sh[25153]: STARTING EXPOSURE at: 2022-01-18 10:31:59   @ 409.54 ms (0.41 sec)
Jan 18 10:32:01 allsky allsky.sh[25153]:   > Sleeping 5.0 sec from histogram exposure
Jan 18 10:32:01 allsky allsky.sh[25153]:   > Saving DAY image 'image-20220118103159.jpg'
Jan 18 10:32:01 allsky allsky.sh[25153]: *** upload.sh: ERROR: File to upload '/home/pi/allsky/images/20220118/image-20220118103159.jpg' not found!
Jan 18 10:32:06 allsky allsky.sh[25153]: STARTING EXPOSURE at: 2022-01-18 10:32:06   @ 409.54 ms (0.41 sec)
Jan 18 10:32:07 allsky allsky.sh[25153]:   > Saving DAY image 'image-20220118103206.jpg'
Jan 18 10:32:07 allsky allsky.sh[25153]:   > Sleeping 5.0 sec from histogram exposure
Jan 18 10:32:07 allsky allsky.sh[25153]: *** upload.sh: ERROR: File to upload '/home/pi/allsky/images/20220118/image-20220118103206.jpg' not found!

hpservertech, 2022-01-18:
[config.txt](https://github.com/thomasjacquin/allsky/files/7890821/config.txt)
Here is the config.sh

hpservertech, 2022-01-18:
[image]

I made sure the save daytime images is true and restarted the service, still occurs.  I didn't copy over the config file, just made the necessary changes to the new one when I upgraded.

linuxkidd, 2022-01-19:
This issue should be resolved by PR #942 that was just merged.  Please update to the latest code and let us know if that resolves the issues.

Thanks!

linuxkidd, 2022-01-19:
Confirmed fixed by PR #942 by @hpservertech in comment there.

Thanks!  closing this issue.

closed 2022-01-19 by linuxkidd