AllskyTeam/allsky #948: Some is screw up on my end

maserowik, 2022-01-19:
So i am getting the following error on my log file.
[image]

any idea?

EricClaeys, 2022-01-19:
@maserowik

**Likely my bad, and I apologize.**
I made a fix to saveImage.sh a few days ago which caused another bug in certain situations (which I don't have since my settings are different).  The image-20220119hhmmss.jpg file in allsky/tmp was renamed to image.jpg before all the processing was done, so your resize of it failed since the original file no longer existed.

We fixed it late yesterday; please download the newest `saveImage.sh`.

Assuming that fixes the issue, you don't both IMG_RESIZE=true and RESIZE_UPLOADS=true if they are both resizing to the same size, 1304x976 in your case.  The second resize isn't doing anything.

maserowik, 2022-01-20:
@EricClaeys 
ok I set the resize_uploads to false.

I will d/l the saveImage.sh test and report the results.

thanks

maserowik, 2022-01-21:
@EricClaeys 
I did a fresh pull today and error went away.

Thanks
Mike

closed 2022-01-21 by maserowik