AllskyTeam/allsky #952: End of night not working with Bullseye

CleverJay, 2022-01-21:
As noted on the Allsky Facebook group, when using the Bullseye OS, Images are not moved to the Images folder from the tmp folder and therefore no time-lapse is created and the images do not appear in the Web GUI. Work around is to revert to Buster, but it would be nice if Allsky worked in Bullseye. Also have  had reports of the RPi HQ camera having issues under Bullseye.

ShedAstro, 2022-01-21:
Same issues, with bullseye, so swapped to buster and it worked for 1 night and then the same issues from then in with buster too…☹️☹️

EricClaeys, 2022-01-21:
@CleverJay   When did you last update the scripts/saveImage.sh file?  We fixed the issue you described 2 days ago.  It should have nothing to do with Bullseye versus Buster, but if you (or whoever had the problem) installed Bullseye and Allsky a few days ago they would have seen the problem.  If they gave up and restored Buster and re-installed Allsky they likely would have gotten the saveImage.sh update.
I have not heard of any issues with the RPi HQ camera under Bullseye.  Can you be more specific about "have had reports"?  Have those people let us know in GitHub about their issues?

@ShedAstro   Does your comment relate to the Issue @CleverJay just reported?

CleverJay, 2022-01-21:
@EricClaeys  replaced saveImage.sh and Allsky is now writing files to the images folder. Thank you. I'll give it 24hours to see if it creates a timelapse and then close this issue.

EricClaeys, 2022-01-22:
@CleverJay That's good to hear.

CleverJay, 2022-01-22:
The new saveImage.sh script now writes the files to the images folder. I'm not sure that it is deleting the images from the tmp folder though. Allsky did generate both a keogram and a startrail image, however it did not generate a time-lapse. Also running timelapse.sh 20220121  manually did not create a time-lapse. I am closing this Issue and will pursue the time-lapse problem in a separate issue. Thanks again to @EricClaeys .

closed 2022-01-22 by CleverJay